openv/openv #78: Keine Daten für Heizkreis 1!?

WikiSpaxe, 2017-10-26:
* Subject: Keine Daten für Heizkreis 1!?
* Wiki-Page: Viess-Data
* From: vimico <userid-31361792@openv.wikispaces>
* Date: Sat, 14 Apr 2012 16:20:58 UT
* Message-ID: <52881190-52881190$openv.wikispaces>

Hallo zusammen,

ich beabsichtige meine V150 KB1 fernzusteuern/überwachen.
Allerdings verwendet meine Anlage nur einen Heizkreis.
Entweder habe ich etwas übersehen, oder mit VIESS_DATA lassen sich nur Daten für Heizkreis 2 (M2) anzeigen/ändern.
Gibts ne Möglichkeit den HK1 (A1/M1) anstelle HK2 darzustellen?

Danke
vi

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Daten für Heizkreis 1!?
* Wiki-Page: Viess-Data
* From: TerminatorIII <userid-12797127@openv.wikispaces>
* Date: Mon, 16 Apr 2012 08:11:11 UT
* Message-ID: <52881190-52917290$openv.wikispaces>
* In-Reply-To: <52881190-52881190$openv.wikispaces>

Hallo,
Du hast wirklich etwas übersehen: Es ist der Quellcode dabei. Das heißt, Du kannst alles ändern, damit es zu deiner Steuerung passt.
Wenn Du nur den Grafen ändern willst, kannst Du es in der Datei vito_Config.xml unter Addr.
Gruß
Gruß

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Daten für Heizkreis 1!?
* Wiki-Page: Viess-Data
* From: vimico <userid-31361792@openv.wikispaces>
* Date: Tue, 17 Apr 2012 11:27:11 UT
* Message-ID: <52881190-52966164$openv.wikispaces>
* In-Reply-To: <52881190-52881190$openv.wikispaces>

Hallo,

danke für die Antwort.

Den Quellcode hatte ich nicht "übersehen" :-)
Ich wollte das Rad ja nicht unbedingt neu erfinden, wenn man es hätte umstellen können.

Deshalb, danke für die tolle Vorarbeit.

Gruß

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Daten für Heizkreis 1!?
* Wiki-Page: Viess-Data
* From: Heizelmann <userid-38742406@openv.wikispaces>
* Date: Sun, 24 Feb 2013 22:44:31 UT
* Message-ID: <52881190-60648722$openv.wikispaces>
* In-Reply-To: <52881190-52881190$openv.wikispaces>

Besten Dank an den Entwickler von viess_data. 

Habe die Software für meine Zwecke auf A1/M1 umgestellt. Weiterhin auch erweitert, dass RS232-Interface ordentlich mit Betriebsspannung versorgt wird. 

Funktioniert an V150 KB1 alles Prima. 
Habe lediglich ca. 2 bis 3 mal pro Stunde einen Lesefehler, welches nicht stört . Kann mir das jemand trotzdem erklären?

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Daten für Heizkreis 1!?
* Wiki-Page: Viess-Data
* From: Heizelmann <userid-38742406@openv.wikispaces>
* Date: Thu, 28 Feb 2013 08:56:42 UT
* Message-ID: <52881190-60755492$openv.wikispaces>
* In-Reply-To: <52881190-52881190$openv.wikispaces>

Kann jemand erklären warum es bei vies_data in unregelmäßigen Abständen (ca. 2-3 Mal pro Stuinde) zu Lesefehlern kommt. Meist betrifft es die Abfrage der Gerätekennng, es fehlen 1 oder 2 Byte.

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Daten für Heizkreis 1!?
* Wiki-Page: Viess-Data
* From: TerminatorIII <userid-12797127@openv.wikispaces>
* Date: Thu, 28 Feb 2013 10:03:57 UT
* Message-ID: <52881190-60755934$openv.wikispaces>
* In-Reply-To: <52881190-52881190$openv.wikispaces>

Hallo Heinzelmann,
soweit ich mich erinnern kann, frage ich das Gerät einmal beim Start ab. Ich stimme jedoch mit dir überein, dass es fehlerhafte "Telegramme" gibt. Daher frage ich auch alle 15s ab, um wenigstens einen gültigen Minutenwert zu haben, den ich in den Graphen eintrage. Meine Hardwareversuchsanordnung ist relativ simpel: Ich habe auf den gesamten Schnickschnack mit der optischen Schnittstelle verzichtet und habe drei Drähte (RX,TX,Masse) direkt auf die CPU-Platine meines Ofens gelötet. Die andere Seite ist mit einem FTDI-Chip verbunden, der als Serielle am PC läuft. Selbst in der Konstellation habe ich fehlerhafte Telegramme. 
Nun ja, da das Telegramm relativ blöde ist (keine Endekennung, keine Checksumme) muss ich irgenwo einen Timeout setzen, wenn kein Zeichen mehr kommt. Dann gibt es den Fehler. Da Windows seit .net ähnlich wie Linux arbeitet (schmeiss alle Daten in eine Queue, und Windows kümmert sich um die Übertragung oder ich bekomme einen Interrupt wenn alle Daten ausgelesen sind) gibt es viele Punkte wo man schrauben könnte. Der erste Ansatz wäre die Anzahl der erwarteten Bytes in den Threshold zu schreiben, so dass der Interrupt nur dann kommt. Nur dazu muss die komplette Empfangroutine umgeschreiben werden und was soll passieren wenn dem nicht so ist.
Kannst ja mal daran schrauben. Source ist ja vorhanden und Compiler gibt es kostenlos von MS.

Gruß

Chris

Chris

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Daten für Heizkreis 1!?
* Wiki-Page: Viess-Data
* From: Heizelmann <userid-38742406@openv.wikispaces>
* Date: Thu, 28 Feb 2013 11:28:40 UT
* Message-ID: <52881190-60756584$openv.wikispaces>
* In-Reply-To: <52881190-52881190$openv.wikispaces>

Hallo Terminatorlll,
danke für die Antwort. Wie schon weiter oben geschrieben habe ich die Software schon für meine V150 KB1 umgestellt. Es war für mich die einzige Alternative aus diesem Wiki, da ich a) auf Windows fixiert bin und b) die hier veröffentliche SW meinen Anforderungen nicht genügt bzw. ohne Source Code ist. Viess_data ist deshalb derzeit für mich die beste Grundlage.
Große Lob an Dich als Entwickler!

Ich verwende das einfache Selbstbau-RS232 Interface. Wenn ich Dich  richtig verstehe kann ich zumindest ein Hardwareproblem ausschliessen, oder?

Von .net verstehe ich nicht viel, bin aber mit allgemeiner Programmiererfahrung schon weit gekommen. Habe auf Heizkreis A1M1 umgestellt, die Stromversorgung über DTR sichergestellt und einiges mehr angepasst bzw. ergänzt. Werden Deinen Hinweis aufnehmen und mich versuchen in .net einzuarbeiten.

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Daten für Heizkreis 1!?
* Wiki-Page: Viess-Data
* From: TerminatorIII <userid-12797127@openv.wikispaces>
* Date: Thu, 28 Feb 2013 11:59:53 UT
* Message-ID: <52881190-60756766$openv.wikispaces>
* In-Reply-To: <52881190-52881190$openv.wikispaces>

Ich denke schon, dass Du ein Hardwareproblem ausschließen kannst. Du musst auch bedenken, dass Windows nicht unbedingt das Betriebssystem für ereignisgesteuerte Anwendungen ist. Mein Zielsystem ist ein kleiner PC (7Watt Leistungaufnahme mit XP). Der läuft seit Monaten mit Telegrammfehlern ohne Datenverluste im Minutenbereich. Wenn Du Fragen hast oder eine Lösung, kannst dich ja mal melden.

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Daten für Heizkreis 1!?
* Wiki-Page: Viess-Data
* From: Heizelmann <userid-38742406@openv.wikispaces>
* Date: Thu, 28 Feb 2013 13:44:50 UT
* Message-ID: <52881190-60757778$openv.wikispaces>
* In-Reply-To: <52881190-52881190$openv.wikispaces>

Gerne melde ich mich an dieser Stelle wieder. 
Ca. 2-3 mal die Stunde habe ich Datenfehler. Interessanterweise fast immer bei 00F8.

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Daten für Heizkreis 1!?
* Wiki-Page: Viess-Data
* From: vitoopen <23297437+vitoopen@users.noreply.github.com>
* Date: Thu, 28 Feb 2013 17:40:42 UT
* Message-ID: <52881190-60766048$openv.wikispaces>
* In-Reply-To: <52881190-52881190$openv.wikispaces>

Hallo Heizelmann
Wenn die Datenfehler so häufig bei 00F8 auftreten, solltest du mal versuchen, diese Adresse auszuklammern. Die Steuerung ändert ja nicht, so dass es nicht nötig ist, die Gerätekennung abzufragen. 
Warum ich das vorschlage? Es könnte sein, dass die Steuerung mit der Gerätekennungsabfrage intern stärker beschäftigt ist, als mit anderen Befehlen. Deshalb würde ich das mal versuchen.

Bei mir ist die Situation zwar anders. Ich verwende das 300er Protokoll und frage alle 1 Sek verschiedene Werte ab (total 18 unterschiedliche Datenpunkte). Nachdem die Steuerung zusätzlich zwei Heizkreise steuern muss ist die Anzahl Datenpunkte angestiegen. Weil dann die Steuerung spontan Fehler machte (z.B. Ausgang Heizkreispumpe gemäss Steuerung EIN, aber Digital Output ist AUS) habe ich dann das Abfrage Intervall auf 2 Sekunden erhöht - danach gab es keine Probleme mehr.
Übertragungsfehler habe ich - soweit ich das beurteilen kann - keine (300er Protokoll, total 3 Steuerungen, Optolink an Vitotronic 333 MW1, über LON 2 x Vitotronic 100 GC1). Ich erfasse aber weder CRC noch Parity Errors...

WikiSpaxe, 2017-10-26:
* Subject: Re: Keine Daten für Heizkreis 1!?
* Wiki-Page: Viess-Data
* From: mich-beck <userid-46393678@openv.wikispaces>
* Date: Sat, 28 Dec 2013 23:48:08 UT
* Message-ID: <52881190-68544782$openv.wikispaces>
* In-Reply-To: <52881190-52881190$openv.wikispaces>

Hallo Zusammen,

möchte das Thema nochmal aufwaermen.
Möchte meine Vitronic ebenfalls mit Viess-Data steuern. Klappt soweit auch super. Allerdings habe ich ebenfalls das Problem, dass ich den ersten Heizkreis (habe nur einen) nicht steuern kann. In Sachen Programmierung bin ich leider nicht sonderlich begabt :-)
Kann mir evtl jemand Hilfestellung oder gar eine Version fuer Heizkreis 1 geben?

Viele Gruesse
Micha

Knallix, 2020-01-11:
Hallo vimico ,

kannst du mir deine vito.xml schicken. Habe die gleiche Steuerung. Wie liest du diese aus ?