openv/openv #86: Vitocal 222G EVU Sperre und Heizpatrone

WikiSpaxe, 2017-10-26:
* Subject: Vitocal 222G EVU Sperre und Heizpatrone
* Wiki-Page: home
* From: callimero <userid-34867758@openv.wikispaces>
* Date: Wed, 22 Aug 2012 19:36:53 UT
* Message-ID: <60019775-55900079$openv.wikispaces>

Moins,

meine Vitocal ist soweit am Netz, Arduino mit Ethernet schickt Daten zu Cosm.

Adressen habe ich an sich alle, nur wie ist die für die Elektrozusatzheizung und die EVU Sperre?

Gruß,
Carsten

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 222G EVU Sperre und Heizpatrone
* Wiki-Page: home
* From: mimax <userid-21458015@openv.wikispaces>
* Date: Wed, 22 Aug 2012 21:10:23 UT
* Message-ID: <60019775-55901625$openv.wikispaces>
* In-Reply-To: <60019775-55900079$openv.wikispaces>

Hallo,
Hmm, genau diese beiden Infos/Adressen fehlen mir auch, leider. Habe ansonsten alle wichtigen Stati, Temperaturen Etc. 
Wo hast Du die Adressen für die 222 her? Irgendwo im Netz gefunden? Selbst ermittelt?

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 222G EVU Sperre und Heizpatrone
* Wiki-Page: home
* From: callimero <userid-34867758@openv.wikispaces>
* Date: Wed, 22 Aug 2012 22:20:48 UT
* Message-ID: <60019775-55902913$openv.wikispaces>
* In-Reply-To: <60019775-55900079$openv.wikispaces>

An sich stehen sie alle hier: http://openv.wikispaces.com/Adressen

unter Wärmepumpe.

EVU könnte sein: 0A81 (Bool). Konnte ich aber nicht probieren bisher.

Heizstab vielleicht Sekundärquelle?

Carsten

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 222G EVU Sperre und Heizpatrone
* Wiki-Page: home
* From: callimero <userid-34867758@openv.wikispaces>
* Date: Wed, 22 Aug 2012 22:59:17 UT
* Message-ID: <60019775-55903403$openv.wikispaces>
* In-Reply-To: <60019775-55900079$openv.wikispaces>

Neee, Heizstab muss was anderes sein. Ich frage mich was die "Sekundärquelle" ist? Hat bei mir 2800 Stunden, k.a. was das meint?

Carsten

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 222G EVU Sperre und Heizpatrone
* Wiki-Page: home
* From: mimax <userid-21458015@openv.wikispaces>
* Date: Thu, 23 Aug 2012 04:36:27 UT
* Message-ID: <60019775-55909151$openv.wikispaces>
* In-Reply-To: <60019775-55900079$openv.wikispaces>

Bist sicher, dass da 2800 Stunden stehen? Welche Adresse ist das?

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 222G EVU Sperre und Heizpatrone
* Wiki-Page: home
* From: callimero <userid-34867758@openv.wikispaces>
* Date: Thu, 23 Aug 2012 16:05:01 UT
* Message-ID: <60019775-55914933$openv.wikispaces>
* In-Reply-To: <60019775-55900079$openv.wikispaces>

0584, scheint sich aber nicht zu ändern. 2800 (ja ich habe schon durch 3600 geteilt) ist aber def. zu viel auch wenn man die Bautrocknung einrechnet....

EVU Sperre scheint auch nicht richtig zu sein.

Carsten

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 222G EVU Sperre und Heizpatrone
* Wiki-Page: home
* From: mimax <userid-21458015@openv.wikispaces>
* Date: Fri, 24 Aug 2012 08:58:13 UT
* Message-ID: <60019775-55928867$openv.wikispaces>
* In-Reply-To: <60019775-55900079$openv.wikispaces>

Nein, eigentlich passen keine der in den Tabellen hier bei openv genannten Register auf die 222-G. Das einzige, was näherungsweise korrekt ist, sind die Temperatur-Adressen, alles andere liegt in komplett anderen Bereichen.

Welche Status/Werte liest Du aus? Wie bist Du an die Adressen gekommen?

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 222G EVU Sperre und Heizpatrone
* Wiki-Page: home
* From: callimero <userid-34867758@openv.wikispaces>
* Date: Fri, 24 Aug 2012 11:02:28 UT
* Message-ID: <60019775-55929493$openv.wikispaces>
* In-Reply-To: <60019775-55900079$openv.wikispaces>

Die Adressen unter "Wärmepumpe" stimmen an sich.

Also ich meine Vorlauf/Rücklauf, Warmwassertemp, Soletemp, Status der Mischer und Pumpen.

Bis auf Heizpatrone und EVO Sperre bin ich an sich "befriedigt" damit.

Carsten

WikiSpaxe, 2017-10-26:
* Subject: Re: Vitocal 222G EVU Sperre und Heizpatrone
* Wiki-Page: home
* From: callimero <userid-34867758@openv.wikispaces>
* Date: Fri, 24 Aug 2012 11:03:18 UT
* Message-ID: <60019775-55929497$openv.wikispaces>
* In-Reply-To: <60019775-55900079$openv.wikispaces>

Ansonsten bleibt wohl nur ein Speicherauszug vor und während der EVU Sperre.

C!

callimero, 2017-11-20:
Ich will das nur mal abschließen wenn jemand über die Suche hier landet.

Nachdem sich meine SD Karte im Raspi verabschiedet hatte musste ich alles neu aufsetzen und habe es etwas modernisiert (altes Image Backup leider verschwunden...). Ich habe ja in Python ein eigenes Ausleseprogramm gebaut das mit die Daten in eine Mysql (jetzt MariaDB) schreibt und dann per PHP die Daten anzeigt.

Netterweise sind in den neuen Tabellen jetzt mehr Einträge für die Vitocal 222G (V200WO1,VBC700). Die funktionieren auch soweit ich getestet habe. Kann jetzt auch Pumpen und Verdichter Status auslesen etc.

Bin noch drann die EVU Sperre zu suchen, ich teste mal die Sammelmeldung...

VG,
Carsten