openv/openv #94: Pumpenstatus Solar Pumpe 2/SpeicherStatus Speicher2

WikiSpaxe, 2017-10-26:
* Subject: Pumpenstatus Solar Pumpe 2/SpeicherStatus Speicher2
* Wiki-Page: home
* From: ceteris_paribus <userid-22560513@openv.wikispaces>
* Date: Tue, 16 Oct 2012 13:52:23 UT
* Message-ID: <60524868-57358022$openv.wikispaces>

Hallo zusammen,

hat es schon jemand geschaft, eine andere pumpe als die des Spiecher 1 (Pumpe1) Adresse 0x6552 auszulesen und sinnvolle Prozente zu bekommen?

Ich habe eine vitotronic 200 und eine über KM-Bus angeschlossene Vitosolic200.

Hat jemand Hinweise auf Adressen ?
6553 usw hatte ich versucht ist 0-255 vom Wert. (bei 100% Drehzahl 255).

Gruß
CP

WikiSpaxe, 2017-10-26:
* Subject: Re: Pumpenstatus Solar Pumpe 2/SpeicherStatus Speicher2
* Wiki-Page: home
* From: Schwammerl <userid-44718776@openv.wikispaces>
* Date: Tue, 31 Dec 2013 14:48:25 UT
* Message-ID: <60524868-68555908$openv.wikispaces>
* In-Reply-To: <60524868-57358022$openv.wikispaces>

Hallo,
ich habe eine Vitotronic 200KO1B Adresse 209F und eine Vitosolic 200SD4 beide über KM-Bus verbunden.  
hast Du schon weitere Adresen rausgefunden? Bin auf der Suche der Adressen des Temperaturfühlers der des Umschaltventils das zwischen Öl und Solar umschaltet.
 Auf der Vitosolic 200 Steuerung ist das Ventil auf R6 angeschlossen und der Tempfühler auf S5.
Eine Adresse habe ich noch gefunden. Es ist 0x6515, das ist bei mir die Zirkulationspumpe WW.

WikiSpaxe, 2017-10-26:
* Subject: Re: Pumpenstatus Solar Pumpe 2/SpeicherStatus Speicher2
* Wiki-Page: home
* From: mickey_muc <userid-53322968@openv.wikispaces>
* Date: Sat, 11 Apr 2015 23:22:20 UT
* Message-ID: <60524868-81067734$openv.wikispaces>
* In-Reply-To: <60524868-57358022$openv.wikispaces>

Hallo WW,

bin Neuling hier und suche die Adressliste für meine Vitotronic 200KO1B mit Kennung
209F. Soweit ich das sehen kann habt Ihr auch so eine Steuerung. Bitte seid so nett und teilt mit mir Eure Erfahrungen und Adressen.

Habe selbst openhab seit 2 Jahren im Einsatz mit einem KNX System und möchte jetzt auch die Heizung einbinden. Erster Schritt war erfolgreich (heute). Ich habe den Optolink in Betrieb genommen und meiner Anlagenversion ausgelesen:

ID 209F
ExtID 0002
Protocol 300
Description Vitotronic 200 KO2B: Witterungsgeführte Kessel- u. Heizkreisregelung für gleitend abgesenkte Kesselwassertemperatur mit Regelung für 2 Heizkreise mit Mischer

Lieben Dank
Michael

dima-357, 2020-11-22:
Please tell me the request address V100
boiler pump in operation or not /
It is not in the list of addresses of Viessmann-adresses