openv/openv #98: Verwendung Lego VLL Rs232 Kabel

WikiSpaxe, 2017-10-26:
* Subject: Verwendung Lego VLL Rs232 Kabel
* Wiki-Page: home
* From: NavNie <userid-37541282@openv.wikispaces>
* Date: Mon,  5 Nov 2012 17:51:48 UT
* Message-ID: <60721878-57995968$openv.wikispaces>

Hallo Zusammen,
bin neu dabei und habe eine Viotronic 200 H01C. Im Spielzeugfundus haben wir ein Lego Spybotic Rs232 VLL Kabel. Habe bereits erste positive Tests beim initialisieren und Beenden gemacht. Der Abstand der Infrarot Led's passt ziemlich gut, werde hierfür ein einfachen Adapter aus Blech biegen. Bereite gerade eine Excelanwendung für das Datenloggen mit VBA vor, werde euch auf dem laufenden halten.
Gruss
NavNie

WikiSpaxe, 2017-10-26:
* Subject: Re: Verwendung Lego VLL Rs232 Kabel
* Wiki-Page: home
* From: NavNie <userid-37541282@openv.wikispaces>
* Date: Mon,  5 Nov 2012 18:53:52 UT
* Message-ID: <60721878-57999282$openv.wikispaces>
* In-Reply-To: <60721878-57995968$openv.wikispaces>

Hallo Zusammen,
anbei ein Link:http://de.wikipedia.org/wiki/VLL-Schnittstelle
Gruss
NavNie

WikiSpaxe, 2017-10-26:
* Subject: Re: Verwendung Lego VLL Rs232 Kabel
* Wiki-Page: home
* From: NavNie <userid-37541282@openv.wikispaces>
* Date: Tue,  4 Dec 2012 18:08:42 UT
* Message-ID: <60721878-58858834$openv.wikispaces>
* In-Reply-To: <60721878-57995968$openv.wikispaces>

Hallo Zusammen,
Kabel arbeitet einwandfrei mit v-control, getestet unter Win98 
Gruss
NavNie

kuebx, 2017-11-10:
Hallo,
ich mag mier keinen rs232 adapter bauen, das original Optolink RS232 scheint es nirgends zu kaufen geben, habe einen shop gefunden der die Spybotics Serial Cable vekauft für ca. 8 EUR + Versand.
Müsste dann auch an der Vitodens 200 passen oder ?