openv/openv #99: Sind hier besitzer einer Vitoligno 300P

WikiSpaxe, 2017-10-26:
* Subject: Sind hier besitzer einer Vitoligno 300P
* Wiki-Page: home
* From: AndreasFloth <userid-37313974@openv.wikispaces>
* Date: Mon,  5 Nov 2012 22:25:34 UT
* Message-ID: <60727718-58008976$openv.wikispaces>

Hallo zusammen, vielen Dank für Eure Arbeit an den Projekten.

Ich habe eine Viessmann Vitoligno 300P die Steuerung ist eine V200F01. Leider schaffe ich es nicht anständige Daten aus der Anlage auszulesen.
Hier einige Bildschirmkopien der Programme.

https://www.dropbox.com/sh/7it0xse93ghxaf0/6ULDNTclee/Heizung%20V200F01

Mit Viss-Data schaffe ich eine Verbindung nach eingriffen in die xml-Dateien und kann die Daten auslesen. 
Jedoch mit allen anderen Programmen geht gar nichts. 
Was jedoch komisch ist, mit v-Control kann ich die "Konfiguration" und "Zeitsteuerung" lesen und schreiben!!!!

Bei Viess-Data passen jedoch die meisten Messwerte nicht. z.B. ist die Kesseltemp. die Puffer-Temp. oben. 

Gibt es hier weitere Pelleter hier im Wiki (Forum)?  Ich bin langsam am verzweifeln.... 

Zur Technik:
Vitoligno 300P 
Steuerung V200F01
Steuerung Antwortet mit VBC550P
Optolink Adapter von Dirk http://openv.wikispaces.com/LAN-Ethernet+%26+USB+Kombiadapter
WinXP und Win7 sowie HP ThinClient mit WinXP Pre.

WikiSpaxe, 2017-10-26:
* Subject: Re: Sind hier besitzer einer Vitoligno 300P
* Wiki-Page: home
* From: AndreasFloth <userid-37313974@openv.wikispaces>
* Date: Mon,  5 Nov 2012 22:28:45 UT
* Message-ID: <60727718-58009116$openv.wikispaces>
* In-Reply-To: <60727718-58008976$openv.wikispaces>

zu schnell auf senden geklickt.... 

Es wäre schön, wenn mit hier einer helfen könnte. 

Danke 

Andreas

WikiSpaxe, 2017-10-26:
* Subject: Re: Sind hier besitzer einer Vitoligno 300P
* Wiki-Page: home
* From: ksmx <userid-38314920@openv.wikispaces>
* Date: Sun, 30 Dec 2012 22:47:11 UT
* Message-ID: <60727718-59408618$openv.wikispaces>
* In-Reply-To: <60727718-58008976$openv.wikispaces>

Habe ebenfalls eine VBC550P (V200F01, Vitoligno 300P). Folgende Adressen ließen sich finden:

Puffer Solltemperatur    6600
Brenner Betriebsstunden 1. Stufe   08A7
 Brenner Starts  088A
Brennstoffverbrauch  08B4 (oder 08B0)
Gefördertes Material Pellet  08B0 (oder 08B4)
Flammtemperatur  0B14
Kesseltemperatur  0B12
Lambdasonde  0B18
Position Primärlaufklappe  0B1A
Posotion Sekundärlaufklappe  0B1B
Drehzahl Gebläse Ist  0B1C
Drehzahl Gebläse Soll  0B1E
Art der Fördertechnik -> Schnecke + Gewebesilo  00FE
Vorlauftemperatur  2900
Raumtemperatur Soll Normalbetrieb A1M1  2306
Raumtemperatur Soll Reduzierter Betrieb  2307
Speichertemperatur  0814
Warmwassertemperatur Soll  6300
Solar Speichertemperatur  081A
Solar Kollektortemperatur  6564
Solar Waermemenge  6560
Solar Betriebsstunden  6568

Ich habe mich bisher nur mit dem Lesen von Werten beschaeftigt. Vielleicht findet sich jemand, der weiss, wie man diese Adressen im Wiki direkt vermerkt. 

Gruss ksmx

WikiSpaxe, 2017-10-26:
* Subject: Re: Sind hier besitzer einer Vitoligno 300P
* Wiki-Page: home
* From: AndreasFloth <userid-37313974@openv.wikispaces>
* Date: Wed,  2 Jan 2013 14:43:54 UT
* Message-ID: <60727718-59425668$openv.wikispaces>
* In-Reply-To: <60727718-58008976$openv.wikispaces>

Hallo ksmx, danke für die Adressen, sind einige dabei, die mir neu sind. 

Hier noch einige bekannte adressen.
AussenTemp. 0800
KesselTemp A202
WarmWasser Ist 0806 oder 0814
WW Soll Aktuell 6500
Vorlauf M1 0808
Puffer Oben 0810
Puffer Unten 0804 oder 0812
BrennerStarts 088A
HK-Pumpe M1 2323

Wo ich noch am suchen bin, ist die Adressse für die "Kesselleistung" von 0-100%

Darf ich fragen, mit welchen System Sie die Daten sammeln und darstellen? Ich verwende eine angepasste version von ViessData

Grüße Andreas

WikiSpaxe, 2017-10-26:
* Subject: Re: Sind hier besitzer einer Vitoligno 300P
* Wiki-Page: home
* From: AndreasFloth <userid-37313974@openv.wikispaces>
* Date: Sat,  5 Jan 2013 18:30:56 UT
* Message-ID: <60727718-59477768$openv.wikispaces>
* In-Reply-To: <60727718-58008976$openv.wikispaces>

Hallo,

die Adresse für die Leistung ist A305 und ergibt einen zwischen 0 und 200!
60=30% 
200=100%

Grüße Andreas

WikiSpaxe, 2017-10-26:
* Subject: Re: Sind hier besitzer einer Vitoligno 300P
* Wiki-Page: home
* From: vstyx <userid-38511402@openv.wikispaces>
* Date: Sat, 12 Jan 2013 14:14:55 UT
* Message-ID: <60727718-59634054$openv.wikispaces>
* In-Reply-To: <60727718-58008976$openv.wikispaces>

Hallo,
Danke für die Info.
Ich habe mit einem Raspberry PI VControl am laufen. Das ganze soll ein Datenlogger werden.
Ich suche noch die Adresse für den Brennerzustand (Ruhezustand, Anheizen, Regelbetrieb, Ausbrandphase) und für den Umschalter (Warmwasser oder Speicherladung).
Würde mich freuen wenn jemand diese gefunden hat.

Danke.
Stefan

PS: Ich habe für die Abfrage mir ein getNoSync gebaut. Damit kann man mit dem vclient alle Daten in weniger als 10 Sekunden  auslesen ;-). Wenn interesse an den xml bitte melden.

WikiSpaxe, 2017-10-26:
* Subject: Re: Sind hier besitzer einer Vitoligno 300P
* Wiki-Page: home
* From: AndreasFloth <userid-37313974@openv.wikispaces>
* Date: Sat, 12 Jan 2013 22:53:18 UT
* Message-ID: <60727718-59637928$openv.wikispaces>
* In-Reply-To: <60727718-58008976$openv.wikispaces>

Hallo Stefan,

versuche mal für den Betriebsmodus die
2500        2 wäre "Regelbetrieb" zumindest liefert dies meine Vitoligno von der Wohnzimmer Couch zurüch, wenn ich die daten mit der Vitrol 300A vergleiche.

Was Du mit Warmwasser und Speicherladung meinst, müsste die Speicherladepumpe sein. Adresse: 0845 oder 6513 kann ich leider nicht bestätigen, da der Kessel z.Z. im Regelbetrieb läuft bei 30% Leistung bei -9°C Außentemperatur läuft. 

Grüße Andreas

WikiSpaxe, 2017-10-26:
* Subject: Re: Sind hier besitzer einer Vitoligno 300P
* Wiki-Page: home
* From: vitoopen <23297437+vitoopen@users.noreply.github.com>
* Date: Tue, 15 Jan 2013 22:38:47 UT
* Message-ID: <60727718-59701148$openv.wikispaces>
* In-Reply-To: <60727718-58008976$openv.wikispaces>

Hallo and die Holz-Heizer
habe mal die Adressliste in Google-Docs für die Vitoligno erweitert (VBC550).
Den Link findet ihr oben in der Wiki-Seite "Adressen"
Darf ich euch bitten, allfällige Fehler oder zusätzliche Datenpunkte zu melden?
Gruss Hanspeter

WikiSpaxe, 2017-10-26:
* Subject: Re: Sind hier besitzer einer Vitoligno 300P
* Wiki-Page: home
* From: vitoopen <23297437+vitoopen@users.noreply.github.com>
* Date: Wed, 16 Jan 2013 00:02:45 UT
* Message-ID: <60727718-59703060$openv.wikispaces>
* In-Reply-To: <60727718-58008976$openv.wikispaces>

@vstyx
Hallo Stefan

probier mal 0x0962 (1 Byte) für den Brennerstatus.
Die Adresse fand ich in Vitosoft, allerdings sind nur die Werte 0/1 (Aus/Ein) dokumentiert. Vielleicht steht da noch mehr drin.

Dann gibt es noch an der Adresse 0xA340 [Airbus ;-)] ein Datenpunkt "Warmwasseranforderung an den internen Speicher" mit dem Zustand 1/0 aktiv/inaktiv. An dieser Adresse müssen 4 Bytes gelesen werden und der Status steht im 13. Bit (wobei ich jetzt nicht weiss, ob die Bits bei 0 oder 1 beginnen). Es müsste also irgend ein Bit in diesem 4 Byte Wert sein (das 14. Bit wäre Speicherladung, das 15. Leistungsreduzierung ist kritisch/unkritisch 1/0, das 20. Produktionsstatus).
Gruss Hanspeter

WikiSpaxe, 2017-10-26:
* Subject: Re: Sind hier besitzer einer Vitoligno 300P
* Wiki-Page: home
* From: vstyx <userid-38511402@openv.wikispaces>
* Date: Thu, 17 Jan 2013 18:09:21 UT
* Message-ID: <60727718-59750412$openv.wikispaces>
* In-Reply-To: <60727718-58008976$openv.wikispaces>

Danke an alle. Ihr seit super.
Im Moment habe ich das Problem, dass ich immer wieder fehlerhafte Messwerte bekomme. Scheint das Problem aus dem Thread: http://openv.wikispaces.com/message/view/vcontrold/51845098 zu sein. (Linux)
Werde versuchen dieses zuerst zu fixen und dann gehe ich wieder an die Adressen. 

Gruß von dem Pessholzheizer Stefan

WikiSpaxe, 2017-10-26:
* Subject: Re: Sind hier besitzer einer Vitoligno 300P
* Wiki-Page: home
* From: vstyx <userid-38511402@openv.wikispaces>
* Date: Sat, 19 Jan 2013 15:26:33 UT
* Message-ID: <60727718-59794440$openv.wikispaces>
* In-Reply-To: <60727718-58008976$openv.wikispaces>

Hallo Hanspeter,

danke, dass mit der 0x0962 ist Brennkammer Ein/Aus aber nicht in welchem Stadium sich die Brennkammer befindet. Mit dem Speicher kann ich erst wieder prüfen, wenn es draußen wärmer wird. Durchs Schichten im Puffer habe ich bei der aktuellen Vorlauftemperatur immer genug Hitze im oberen Puffer (Warmwasser).

Gruß
Stefan

WikiSpaxe, 2017-10-26:
* Subject: Re: Sind hier besitzer einer Vitoligno 300P
* Wiki-Page: home
* From: vitoopen <23297437+vitoopen@users.noreply.github.com>
* Date: Sat, 19 Jan 2013 15:35:06 UT
* Message-ID: <60727718-59794516$openv.wikispaces>
* In-Reply-To: <60727718-58008976$openv.wikispaces>

Hallo Stefan,
dann probier mal 0xA302
Der 1 Byte Wert kann nur gelesen werden und hat folgende Bedeutung:

ApplicationMode

Values

 0=AUTO
 1=HEAT
 2=MRNG_WRMUP
 3=COOL
 4=NIGHT_PURGE
 5=PRE_COOL
 6=OFF
 7=TEST
 8=EMERG_HEAT
 9=FAN_ONLY
 10=FREE_COOL
 11=ICE
 110=SLAVE_ACTIVE
 111=LOW_FIRE
 112=HIGH_FIRE
 100=FLOW_TEMP

geht das in die richtige Richtung?
Gruss, Hanspeter

WikiSpaxe, 2017-10-26:
* Subject: Re: Sind hier besitzer einer Vitoligno 300P
* Wiki-Page: home
* From: vstyx <userid-38511402@openv.wikispaces>
* Date: Sun, 20 Jan 2013 19:21:11 UT
* Message-ID: <60727718-59807928$openv.wikispaces>
* In-Reply-To: <60727718-58008976$openv.wikispaces>

Danke Hanspeter,

werde es testen und melde mich dann.

Gruß
Stefan
PS: Bin gerade einen Logger in Java am schreiben (Protokoll 300). Der vclient bringt mir zu viele Abbrüche. (Hoffe in Java ist es nicht so.)

Cupra85, 2025-09-26:
Hi, hast du eine komplette Liste deiner Datenpunkte mit Byte, dezimal Stellen und Einheit?

wschleifer, 2025-09-26:
Hallo,
ich hätte eine XLS (ohne Gewähr)
basiert auf einer Vorlage aus dem Forum hier und dem Wiki.

wschleifer, 2025-09-26:
meine grafana Auswertung...

[image]

[image]

[image]

Cupra85, 2025-09-26:
Sieht sehr gut aus!
Ich lasse es auf einem ESP32 C6 laufen und funktioniert super. 
Habe die originale Repo kopiert und das Sensor schema Fehler gefixt. Zudem hab ich die maximale Sensorzahl von original 20 auf 40 erhöht. Denke mal das originale Repo ist für kleine Wemos D1 mini gedacht.