openv/openv #371: Kommando für Einmalladung

Hardy74, 2017-10-31:
Hallo,

auf den Apps wie auf der Anlage selbst gibt es das Feature "Einmalladung". Das Warmwasser wird im Nachtbetrieb einmal aufgeheizt.

Kann man das irgendwie nachbilden?

Hardy74, 2018-01-02:
Kann keiner der Protokollpäpste die Adresse der Einmalladung einmal erschnüffeln? Verständlicher Weise ist Viessmann nicht gewillt, solcherlei Interna kund zu tun. Email von heute:

> Leider muß ich Ihnen mitteilen, dass es sich bei den angefragten Adressdaten um interne Daten handelt welche nicht raus gegeben werden.

speters, 2018-01-02:
Schau dir mal die 0x65f5 an, vielleicht kommt das hin.

Hardy74, 2018-01-03:
Hallo,
ehrlich gesagt bin ich unsicher, was Du mit den 4Bytes ab 0xf8 meinst. Der Steuerungstyp ist VScotH01. Auf Verdacht hänge ich mal die Ausgabe zu getDevType an, was ja 00f8 ist:
`[827] Wed Jan  3 13:14:24 2018 : Befehl: getDevType
[827] Wed Jan  3 13:14:24 2018 : Process 827 tries to aquire lock
[827] Wed Jan  3 13:14:24 2018 : Process 827 got lock
[827] Wed Jan  3 13:14:24 2018 : >FRAMER: open device /dev/ttyUSB0 ProtocolID 41
[827] Wed Jan  3 13:14:24 2018 : konfiguriere serielle Schnittstelle /dev/ttyUSB0
[827] Wed Jan  3 13:14:24 2018 : >SEND: 04
[827] Wed Jan  3 13:14:26 2018 : <RECV: len=1 05 (2000.0 ms)
[827] Wed Jan  3 13:14:26 2018 : <RECV: received 05
[827] Wed Jan  3 13:14:26 2018 : >FRAMER: closed
[827] Wed Jan  3 13:14:26 2018 : >SEND: 16
[827] Wed Jan  3 13:14:26 2018 : >SEND: 00
[827] Wed Jan  3 13:14:26 2018 : >SEND: 00
[827] Wed Jan  3 13:14:26 2018 : <RECV: len=1 06 (10.0 ms)
[827] Wed Jan  3 13:14:26 2018 : <RECV: received 06
[827] Wed Jan  3 13:14:26 2018 : >FRAMER: opened
[827] Wed Jan  3 13:14:26 2018 : >SEND: 41
[827] Wed Jan  3 13:14:26 2018 : >SEND: 05
[827] Wed Jan  3 13:14:26 2018 : >SEND: 00
[827] Wed Jan  3 13:14:26 2018 : >SEND: 01
[827] Wed Jan  3 13:14:26 2018 : >SEND: 00
[827] Wed Jan  3 13:14:26 2018 : >SEND: F8
[827] Wed Jan  3 13:14:26 2018 : >SEND: 08
[827] Wed Jan  3 13:14:26 2018 : >SEND: 06
[827] Wed Jan  3 13:14:26 2018 : <RECV: len=1 06 (30.0 ms)
[827] Wed Jan  3 13:14:26 2018 : <RECV: received 06
[827] Wed Jan  3 13:14:26 2018 : >FRAMER: Command send
[827] Wed Jan  3 13:14:26 2018 : <RECV: len=1 41 (0.0 ms)
[827] Wed Jan  3 13:14:26 2018 : <RECV: len=1 0D (0.0 ms)
[827] Wed Jan  3 13:14:26 2018 : <RECV: len=1 01 (10.0 ms)
[827] Wed Jan  3 13:14:26 2018 : <RECV: len=1 01 (0.0 ms)
[827] Wed Jan  3 13:14:26 2018 : <RECV: len=1 00 (0.0 ms)
[827] Wed Jan  3 13:14:26 2018 : <RECV: len=1 F8 (0.0 ms)
[827] Wed Jan  3 13:14:26 2018 : <RECV: len=1 08 (10.0 ms)
[827] Wed Jan  3 13:14:26 2018 : <RECV: len=1 20 (0.0 ms)
[827] Wed Jan  3 13:14:26 2018 : <RECV: received 41 0D 01 01 00 F8 08 20
[827] Wed Jan  3 13:14:26 2018 : <RECV: len=1 CB (0.0 ms)
[827] Wed Jan  3 13:14:26 2018 : <RECV: len=1 1F (0.0 ms)
[827] Wed Jan  3 13:14:26 2018 : <RECV: len=1 C9 (10.0 ms)
[827] Wed Jan  3 13:14:26 2018 : <RECV: len=1 00 (0.0 ms)
[827] Wed Jan  3 13:14:26 2018 : <RECV: len=1 00 (0.0 ms)
[827] Wed Jan  3 13:14:26 2018 : <RECV: len=1 01 (0.0 ms)
[827] Wed Jan  3 13:14:26 2018 : <RECV: len=1 14 (10.0 ms)
[827] Wed Jan  3 13:14:26 2018 : <RECV: len=1 F7 (0.0 ms)
[827] Wed Jan  3 13:14:26 2018 : <RECV: received CB 1F C9 00 00 01 14 F7
[827] Wed Jan  3 13:14:26 2018 : 20 CB 1F C9 00 00 01 14 -> UNKNOWN
[827] Wed Jan  3 13:14:26 2018 : UNKNOWN
[827] Wed Jan  3 13:14:26 2018 : Befehl: quit
[827] Wed Jan  3 13:14:26 2018 : >SEND: 04
[827] Wed Jan  3 13:14:26 2018 : <RECV: len=1 06 (0.0 ms)
[827] Wed Jan  3 13:14:26 2018 : <RECV: received 06
[827] Wed Jan  3 13:14:26 2018 : >FRAMER: closed
[827] Wed Jan  3 13:14:27 2018 : Process 827 released lock
[827] Wed Jan  3 13:14:27 2018 : Verbindung beendet (fd:2)
[827] Wed Jan  3 13:14:27 2018 : Child Prozess pid:827 beendet
`
Dann habe ich folgendes in die vito.xml eingetragen:
`<command name='getIsEinmalladung' protocmd='getaddr'>
	<addr>65f5</addr>
	<len>1</len>
	<unit>E_ST</unit>
	<description>Ermittle ob Einmalladung </description>
</command>

<command name='setEinmalladung' protocmd='setaddr'>
	<addr>65f5</addr>
	<len>1</len>
	<unit>E_ST</unit>
	<description>Setze Einmalladung</description>
</command>`

Die Ausgaben des get-Befehls folgen dem vorausgegangenen set-Befehl. Allerdings ist die Heizung momentan im Tagbetrieb, das Wasser ist heiß. Nach dem ersten set war auch auf der Anlage Einmalladung aktiv. Auf das Ausschalten via set reagierte die Anlage nicht, die Einmalladung blieb im Display aktiv, auch wenn das get nun AUS lieferte. Auf der Anlage habe ich dann die Einmalladung beendet. Sie ließ sich nun via set augenscheinlich nicht wieder einschalten, auch wenn das get EIN lieferte. 
Ich werde das weiter testen und hier berichten.

Vielen Dank für die Adresse!

Hardy74, 2018-01-03:
Die Formatierung ist ja grauenvoll... ich hänge das Wesentliche nochmal als Text an...

[einmal.txt](https://github.com/openv/openv/files/1600352/einmal.txt)

speters, 2018-01-03:
(Formatierung  oben geändert mit \`\`\` als Code-Begrenzer)

Du scheinst eine VScotHO1_200 (VSCotHO1 ab Softwareindex 200, bei dir Softwareindex 201) zu haben.

Welche "App" meinst du?
Wenn du per _Vitoconnect_ an deiner Steuerung dran bist, könntest da das serielle Protokoll belauschen, um zu sehen, was da zusätzlich zur o.g. Adresse noch so gemacht wird.

WLAN-Verbindung mitzuschneiden dürfte wg. der verschlüsselten Verbindung zur Viessman-Cloud schwieriger sein.
Da wäre ich auch vorsichtiger, weil Viessmann prinzipiell den Zugang sperren könnte, wenn die Unregelmäßigkeiten erkennen. Das wäre blöd, weil Seriennummer, MAC-Adresse, etc. dann ggf. in einer Blacklist landen könnten (muss nicht sein, aber lieber Vorsicht walten lassen).

Hardy74, 2018-01-03:
Ich habe keine App benutzt. Ich stand mit dem Laptop (Putty zum Raspi) vor der Heizung und hab die set und get Befehle abgesetzt und geschaut, was auf der Heizung passiert, respektive direkt auf der Heizung die Einmalladung deaktiviert. (Und leider direkt danach nicht daran gedacht, get zu senden...).

Danke für die Warnung, daß Viessmann einen sperrt.

Wie schnüffelst Du seriell? Auf den Sub-D's damals habe ich es mit einem Y-Kabel gemacht. Jetzt würde ich mal schauen, ob man das nicht auch mit dem Raspi realisieren kann!? Sprich aus der Viessmann-WLAN-Kiste via USB in den Raspi, und die Optolink-Strippe an einem zweiten USB-Port angeschlossen. Ein entsprechendes Programm auf dem Raspi schnüffelt den Verkehr dann mit.
Kurzes angoogeln soeben läßt mich jedoch ahnen, daß es so einfach dann doch nicht gehen wird...

speters, 2018-01-03:
Bzgl. der Warnung: Ich weiss nicht, _ob_ Viessmann bei Unregelmäßigkeiten sperrt, aber wäre vorsichtig mit dem, was ich der Viessmann-Cloud in einem Anfall von Forscherdrang unterjubele.

Den Datenverkehr kannst am einfachsten da mitschneiden, wo er noch gutes altes serielles Protokoll ist. Kommt man bei einem echten USB-Optolink-Kabel nur leider schlecht dran.

Ansonsten geht natürlich auch deine Raspi-Lösung, also USB-seriell-Konverter an USB der [Vitoconnect](https://github.com/openv/openv/wiki/Vitoconnect-100), seriell an Raspi Hardware-Serial (RX des einen an TX des anderen). Optolink an USB vom Raspi, dann auf diesem per `socat` o.ä. die beiden Schnittstellen verbinden und loggen.
Vitoconnect-seitig bestehen m.E. gute Chancen, dass jeder USB-seriell-Konverter funktioniert, der sich als `/dev/ttyUSB0` einbindet.

Hardy74, 2018-01-04:
Da die Heizung momentan in der Absenkung ist, habe ich aus der Ferne die Einmalladung via set eingeschaltet, get lieferte dann EIN. Nach einer Weile mit laufendem Brenner hatte das Warmwasser die Solltemperatur erreicht und get lieferte AUS.

Kurz: Die Adresse scheint zu funktionieren!