openv/openv #372: Compilerwarnungen, die scheinbar behoben werden sollten

l3u, 2017-11-04:
Hallo allerseits!

Ich habe ein [ebuild für vcontrold für Gentoo Linux](https://bugs.gentoo.org/574964) geschrieben. Wenn man das Paket baut, dann kommt folgende Warnung:
```
 * QA Notice: Package triggers severe warnings which indicate that it
 *            may exhibit random runtime failures.
 * common.c:65:2: warning: implicit declaration of function ‘vasprintf’ [-Wimplicit-function-declaration]
 * socket.c:61:2: warning: implicit declaration of function ‘asprintf’ [-Wimplicit-function-declaration]
 * vsim.c:40:2: warning: implicit declaration of function ‘vasprintf’ [-Wimplicit-function-declaration]

 * Please do not file a Gentoo bug and instead report the above QA
 * issues directly to the upstream developers of this software.
```
Ich hab mal in die entsprechenden Quelltexte reingeschaut, aber ich bin nicht ganz so fit mit C und hab ehrlich gesagt keine Ahnung, warum das passiert (stdio.h ist doch eingebunden?!).

Vielleicht weiß jemand anderes, was da los ist, und wie man es behebt?

Viele Grüße :-)

l3u, 2017-11-04:
Nach weiterer Recherche habe ich herausgefunden, was los ist: Die genannten Funktionen sind GNU-Extensions der Standardbibliothek. Deswegen muss man in den Quellen angeben, dass man diese GNU-Extensions nutzt. Folgender Patch behebt das Problem:
```
diff -ur vcontrold/common.c.orig vcontrold/common.c
--- vcontrold/common.c.orig	2017-11-04 09:33:01.309352203 +0100
+++ vcontrold/common.c	2017-11-04 09:52:44.019193686 +0100
@@ -2,6 +2,9 @@
  * Testprogramm fuer Vito- Abfrage
  *  */
 /* $Id: common.c 26 2008-03-20 20:56:09Z marcust $ */
+
+#define _GNU_SOURCE
+
 #include <stdlib.h>
 #include <stdio.h>
 #include <errno.h>
diff -ur vcontrold/socket.c.orig vcontrold/socket.c
--- vcontrold/socket.c.orig	2017-11-04 09:33:01.312352182 +0100
+++ vcontrold/socket.c	2017-11-04 09:52:35.549252116 +0100
@@ -1,6 +1,8 @@
 /* socket.c */
 /* $Id: socket.c 13 2008-03-02 13:13:41Z marcust $ */
 
+#define _GNU_SOURCE
+
 #include <sys/time.h>
 #include <time.h>
 #include <errno.h>
diff -ur vcontrold/vsim.c vcontrold/vsim.c
--- vcontrold/vsim.c.orig	2017-11-04 09:33:01.305352231 +0100
+++ vcontrold/vsim.c	2017-11-04 09:52:26.808312412 +0100
@@ -1,3 +1,5 @@
+#define _GNU_SOURCE
+
 #include <stdlib.h>
 #include <stdio.h>
 #include <ctype.h>
```

l3u, 2017-11-05:
Abgesehen davon: Wollt ihr nicht vielleicht den Code von vcontrold hierher importieren? Würde einiges leichter machen …

l3u, 2017-11-06:
Ich hab den Patch jetzt einfach mal zu einem Fork hinzugefügt: https://github.com/l3u/vcontrold2

speters, 2017-11-06:
VControld ist nur mit History aus dem SF.net SVN hier importiert. Deine Fork ist da als "cmake" Branch drin und du hast einen Zugangslink bekommen. Wäre toll, wenn du _da_ dran weitermachen würdest, damit die Historie der Dateien erhalten bleibt. Ausserdem liessen sich dann die ursprünglichen Autoren vielleicht besser bewegen, da mitzumachen.

PS:
Issues zu vcontrold besser im dortigen Issuetracker https://github.com/openv/vcontrold/issues

closed 2017-11-06 by speters