openv/openv #375: Modernisierung von vcontrold

l3u, 2017-11-05:
Hallo allerseits!

Ich denke, man sollte den Umzug den OpenV-Wikis zum Anlass nehmen, dem Code von vcontrold mal einen Komplettwaschgang zu verpassen. Ich bin jetzt kein C-Profi, aber ich hab einfach mal damit angefangen, basierend (geforkt von) Revision 107 auf sf.net. Der momentane Stand ist auf https://github.com/l3u/vcontrold2 zu finden (ich hab's mal "vcontrold2" genannt, nicht, dass einer denkt, das sei das offizielle Repository).

Die hauptsächliche momentane Änderung ist, dass ich das bisherige Buildsystem rausgeworfen, und durch cmake ersetzt habe. Vielleicht gibt es hier ja Leute, die auch der Meinung sind, dass das eine gute Idee wäre? Bzw. vielleicht meldet sich auch einer der eigentlichen Entwickler?

Ich würde mich jedenfalls freuen, wenn das Projekt weiterentwickelt wird, und vielleicht auch im Zuge dessen von den Vorteilen profitiert, die z. B. ein Repository hier auf github hat.

Viele Grüße :-)

speters, 2017-11-06:
Ich habe mal den vcontrold mit kompletter Historie aus dem SF.net SVN importiert unter https://github.com/openv/vcontrold und dir einen Invite-Link dazu geschickt. Vielleicht hast du ja Lust, dann in dem Fork weiterzuarbeiten, damit die Historie drin bleibt.

hmueller01, 2017-11-06:
Sehr schön. Endlich geht mal wieder was. Ich werde meinen P300 Patch die Tage mal einstellen, nachdem der seit über einem Jahr keine (vollständige) Beachtung gefunden hat. :-(

speters, 2017-11-06:
@hmueller01 Gerne, am besten Fork machen, Patch einpflegen und dann Pull-Request erstellen. So bleibt das am Übersichtlichsten.

hmueller01, 2017-11-06:
Yo, so hatte ich das vor. Mit github habe ich schon etwas Erfahrung. ;-)

l3u, 2017-11-06:
Ich denke mal, das hier hat sich mit https://github.com/openv/vcontrold erledigt :-)

closed 2017-11-06 by l3u