openv/openv #376: Openhab

thomytp, 2017-11-06:
Hallo,
Ich bin schwer begeistert von eurer jetzigen Arbeit, daher wollte ich nun unsere Heizung in unseren openhab einbinden.
Leider ist es doch etwas verwirrend, ..

Ich stelle mir das ganze so vor:

USB-Adapter mit einem neuen pi werden an der Heizung installiert
Der openhab pi greift auf diesen zu und bekommt die items aus der Heizung.

Wäre diese Installation sinnvoll? Hat jemand zu dem Thema Erfahrungen und gibt es eine Übersicht der erforderlichen Programmen am pi Heizung und am openhab pi?
Wie erfolgt ein Zugriff auf historische Daten? Muss der openhab diese Werte selbst mitschreiben oder ist ein Zugriff auf die dB des pi Heizung möglich?

Danke für eure Hilfe..

Gruß
Thomytp

Heizelmann, 2017-11-06:
Habe keine Erfahrung aber robuster, kompakter und wartungsfreundlicher  fände ich eine Lösung mit einem ESP8266 an der Heizung. [VitoWifi von bertmelis](https://github.com/bertmelis/VitoWifi) ist da ein erster Schritt. Man sollte diesen Ansatz fortsetzen und z.B. mit MQTT ausbauen.

Jojo-A, 2017-11-06:
Hallo zusammen,

ich bin zwar recht neu hier, möchte mich aber gern beteiligen.

Ich fänd eine Lösung mit ESP8266+MQTT ebenfalls sehr gut. Ich war auch dabei den gleichen Ansatz wie @Heinzelmann zu verfolgen. Ich würde das ganze dann in Home Assistant einbinden wollen.

Im Prinzip könnte man einfach z.B. PubSubClient einbinden und dann über entsprechende Nachrichten sowas wie ein MQTT-Optolink-Gateway bauen.

Von Viessmann hab ich leider auf Nachfrage hin keine sinnvolle Lösung angeboten bekommen. Die Vitoconnect 100 ist für eine Einbindung in ein Home Automation System nicht vorgesehen *kopfschüttel* und das Vitogate ist, wenn man es nur als Optolink-LON-Umsetzer betrachtet, keine Alernative, da viel zu teuer.

bertmelis, 2017-11-06:
For ESP8266 and MQTT I can recommend the [Homie-framework](https://github.com/marvinroger/homie-esp8266). That way, you won't have to think about reconnecting to MQTT or wifi. My MQTT-broker runs on a OpenWRT router. History is on a raspberry pi with Node-red, InfluxDB and Grafana. (the broker could also run on the RPI off course). I also run Openhab which listens to the MQTT topics.

This way I can get my Openhab running very lean (only KNX and MQTT binding). Persistence is done by the MQTT broker or by InfluxDB but it is not managed by Openhab.

Jojo-A, 2017-11-07:
@bertmelis Ok, I think I've got your point. But what will be the particular advantage of that? In my setup, I am going to have an ESP8266, that acts like a MQTT-Optolink-Gateway. My MQTT broker is directly integrated in my Home Assistent system. Is there anything "bad" about that?
To be honest: I don't really know, what Homie is for. I understood, that they try to establish some kind of "convention" how to organize the MQTT topics. But I don't know, if this is the only purpose and if this is worth the effort. Actually I am quite happy to use my MQTT topics the way _I_ want.
Please don't get me wrong, this is totally not meant as on "offense". It is surely just a lag of knowledge on my side :P .

thomytp, 2017-11-22:
So wie das aussieht ist das ganze noch recht experimental, schade zum basteln habe ich leider z.z. Keine echte Zeit. Ich hatte gehofft das es schon was fertigeres gibt...
Trotzdem danke..

bertmelis, 2017-11-22:
VitoWifi is still "experimental" but the firmware runs very stable. I only call it experimental because I personality haven't used writing yet, only reading. (and I do have an extremely DIY hardware setup) I have yet to implement all the different types of addresses. But as I personally don't use them, progress is rather slow in this area. I can only test on 1 heating...

But you're right when you say it is less a finished product as vcontrold/vclient. It does require more skills from the user.

s0170071, 2018-03-31:
+1 
for Vito->ESP8266->MQTT

@bertmelis:
 If you are looking for a 'platform' to operate the ESP8266 you may want to have a look at ESPEasy. They use plug-ins to extend their  capabilities. VitoWifi could be such a plugin.