openv/openv #377: VScotHO1 setBetriebsartPartyM1 funktioniert nicht

Hardy74, 2017-11-07:
Hallo,

zuerst kam das Setzen bei obiger Steuerung, 20CB, mit Fehlern zurück. Das Lesen dieses Blogs förderte zu Tage, daß in der vcontrold.xml die Zeile "<command>SEND 01 F4</command>" zu ändern ist zu "<command>SEND 00 02</command>". Das führt zu einer augenscheinlichen, einwandfreien Auführung:
pi@Raspi4:~ $ vclient -h localhost:3002 -c "setBetriebPartyM1 1"
setBetriebPartyM1 1:
OK

Allerdings liefert get die Ernüchterung, daß schlicht nichst passiert ist:
pi@Raspi4:~ $ vclient -h localhost:3002 -c getBetriebPartyM1
getBetriebPartyM1:
0

Das Log sieht wie folgt aus:
Nov  7 09:42:56 Raspi4 vito[815]: Client verbunden 127.0.0.1:57272 (FD:2)
Nov  7 09:42:56 Raspi4 vito[815]: Child Prozess mit pid:1383 gestartet
Nov  7 09:42:56 Raspi4 vito[1383]: Befehl: setBetriebPartyM1 1
Nov  7 09:42:56 Raspi4 vito[1383]: Process 1383 tries to aquire lock
Nov  7 09:42:56 Raspi4 vito[1383]: Process 1383 got lock
Nov  7 09:42:56 Raspi4 vito[1383]: >FRAMER: open device /dev/ttyUSB0 ProtocolID 41
Nov  7 09:42:56 Raspi4 vito[1383]: konfiguriere serielle Schnittstelle /dev/ttyUSB0
Nov  7 09:42:57 Raspi4 vito[1383]: >SEND: 04
Nov  7 09:42:59 Raspi4 vito[1383]: <RECV: len=1 05 (1990.0 ms)
Nov  7 09:42:59 Raspi4 vito[1383]: <RECV: received 05
Nov  7 09:42:59 Raspi4 vito[1383]: >FRAMER: closed
Nov  7 09:42:59 Raspi4 vito[1383]: >SEND: 16
Nov  7 09:42:59 Raspi4 vito[1383]: >SEND: 00
Nov  7 09:42:59 Raspi4 vito[1383]: >SEND: 00
Nov  7 09:42:59 Raspi4 vito[1383]: <RECV: len=1 06 (10.0 ms)
Nov  7 09:42:59 Raspi4 vito[1383]: <RECV: received 06
Nov  7 09:42:59 Raspi4 vito[1383]: >FRAMER: opened
Nov  7 09:42:59 Raspi4 vito[1383]: 1 -> 01
Nov  7 09:42:59 Raspi4 vito[1383]: >SEND: 41
Nov  7 09:42:59 Raspi4 vito[1383]: >SEND: 06
Nov  7 09:42:59 Raspi4 vito[1383]: >SEND: 00
Nov  7 09:42:59 Raspi4 vito[1383]: >SEND: 02
Nov  7 09:42:59 Raspi4 vito[1383]: >SEND: 23
Nov  7 09:42:59 Raspi4 vito[1383]: >SEND: 03
Nov  7 09:42:59 Raspi4 vito[1383]: >SEND: 01
Nov  7 09:42:59 Raspi4 vito[1383]: >SEND: 01
Nov  7 09:42:59 Raspi4 vito[1383]: >SEND: 30
Nov  7 09:42:59 Raspi4 vito[1383]: <RECV: len=1 06 (30.0 ms)
Nov  7 09:42:59 Raspi4 vito[1383]: <RECV: received 06
Nov  7 09:42:59 Raspi4 vito[1383]: >FRAMER: addr was still active FE06
Nov  7 09:42:59 Raspi4 vito[1383]: >FRAMER: Command send
Nov  7 09:42:59 Raspi4 vito[1383]: >FRAMER: no preset result
Nov  7 09:42:59 Raspi4 vito[1383]: <RECV: len=1 41 (0.0 ms)
Nov  7 09:42:59 Raspi4 vito[1383]: <RECV: received 41
Nov  7 09:42:59 Raspi4 vito[1383]: <RECV: len=1 05 (0.0 ms)
Nov  7 09:42:59 Raspi4 vito[1383]: <RECV: received 05
Nov  7 09:42:59 Raspi4 vito[1383]: <RECV: len=1 01 (10.0 ms)
Nov  7 09:42:59 Raspi4 vito[1383]: <RECV: len=1 02 (0.0 ms)
Nov  7 09:42:59 Raspi4 vito[1383]: <RECV: len=1 23 (0.0 ms)
Nov  7 09:42:59 Raspi4 vito[1383]: <RECV: len=1 03 (0.0 ms)
Nov  7 09:42:59 Raspi4 vito[1383]: <RECV: len=1 01 (10.0 ms)
Nov  7 09:42:59 Raspi4 vito[1383]: <RECV: len=1 2F (0.0 ms)
Nov  7 09:42:59 Raspi4 vito[1383]: <RECV: received 01 02 23 03 01 2F
Nov  7 09:42:59 Raspi4 vito[1383]: 00 -> OK
Nov  7 09:42:59 Raspi4 vito[1383]: OK
Nov  7 09:42:59 Raspi4 vito[1383]: Befehl: quit
Nov  7 09:42:59 Raspi4 vito[1383]: >SEND: 04
Nov  7 09:42:59 Raspi4 vito[1383]: <RECV: len=1 06 (0.0 ms)
Nov  7 09:42:59 Raspi4 vito[1383]: <RECV: received 06
Nov  7 09:42:59 Raspi4 vito[1383]: >FRAMER: closed
Nov  7 09:42:59 Raspi4 vito[1383]: Process 1383 released lock
Nov  7 09:42:59 Raspi4 vito[1383]: Verbindung beendet (fd:2)
Nov  7 09:42:59 Raspi4 vito[1383]: Child Prozess pid:1383 beendet
Nov  7 09:43:04 Raspi4 vito[815]: Client verbunden 127.0.0.1:57276 (FD:2)
Nov  7 09:43:04 Raspi4 vito[815]: Child Prozess mit pid:1385 gestartet
Nov  7 09:43:04 Raspi4 vito[1385]: Befehl: getBetriebPartyM1
Nov  7 09:43:04 Raspi4 vito[1385]: Process 1385 tries to aquire lock
Nov  7 09:43:04 Raspi4 vito[1385]: Process 1385 got lock
Nov  7 09:43:04 Raspi4 vito[1385]: >FRAMER: open device /dev/ttyUSB0 ProtocolID 41
Nov  7 09:43:04 Raspi4 vito[1385]: konfiguriere serielle Schnittstelle /dev/ttyUSB0
Nov  7 09:43:04 Raspi4 vito[1385]: >SEND: 04
Nov  7 09:43:05 Raspi4 vito[1385]: <RECV: len=1 05 (1480.0 ms)
Nov  7 09:43:05 Raspi4 vito[1385]: <RECV: received 05
Nov  7 09:43:05 Raspi4 vito[1385]: >FRAMER: closed
Nov  7 09:43:05 Raspi4 vito[1385]: >SEND: 16
Nov  7 09:43:05 Raspi4 vito[1385]: >SEND: 00
Nov  7 09:43:05 Raspi4 vito[1385]: >SEND: 00
Nov  7 09:43:05 Raspi4 vito[1385]: <RECV: len=1 06 (10.0 ms)
Nov  7 09:43:05 Raspi4 vito[1385]: <RECV: received 06
Nov  7 09:43:05 Raspi4 vito[1385]: >FRAMER: opened
Nov  7 09:43:05 Raspi4 vito[1385]: >SEND: 41
Nov  7 09:43:05 Raspi4 vito[1385]: >SEND: 05
Nov  7 09:43:05 Raspi4 vito[1385]: >SEND: 00
Nov  7 09:43:05 Raspi4 vito[1385]: >SEND: 01
Nov  7 09:43:05 Raspi4 vito[1385]: >SEND: 23
Nov  7 09:43:05 Raspi4 vito[1385]: >SEND: 03
Nov  7 09:43:05 Raspi4 vito[1385]: >SEND: 01
Nov  7 09:43:05 Raspi4 vito[1385]: >SEND: 2D
Nov  7 09:43:05 Raspi4 vito[1385]: <RECV: len=1 06 (30.0 ms)
Nov  7 09:43:05 Raspi4 vito[1385]: <RECV: received 06
Nov  7 09:43:05 Raspi4 vito[1385]: >FRAMER: addr was still active FE06
Nov  7 09:43:05 Raspi4 vito[1385]: >FRAMER: Command send
Nov  7 09:43:05 Raspi4 vito[1385]: >FRAMER: no preset result
Nov  7 09:43:05 Raspi4 vito[1385]: <RECV: len=1 41 (0.0 ms)
Nov  7 09:43:05 Raspi4 vito[1385]: <RECV: received 41
Nov  7 09:43:05 Raspi4 vito[1385]: <RECV: len=1 06 (0.0 ms)
Nov  7 09:43:05 Raspi4 vito[1385]: <RECV: received 06
Nov  7 09:43:05 Raspi4 vito[1385]: <RECV: len=1 01 (0.0 ms)
Nov  7 09:43:05 Raspi4 vito[1385]: <RECV: len=1 01 (10.0 ms)
Nov  7 09:43:05 Raspi4 vito[1385]: <RECV: len=1 23 (0.0 ms)
Nov  7 09:43:05 Raspi4 vito[1385]: <RECV: len=1 03 (0.0 ms)
Nov  7 09:43:05 Raspi4 vito[1385]: <RECV: len=1 01 (0.0 ms)
Nov  7 09:43:05 Raspi4 vito[1385]: <RECV: len=1 00 (10.0 ms)
Nov  7 09:43:05 Raspi4 vito[1385]: <RECV: len=1 2F (0.0 ms)
Nov  7 09:43:05 Raspi4 vito[1385]: <RECV: received 01 01 23 03 01 00 2F
Nov  7 09:43:05 Raspi4 vito[1385]: 00 -> 0
Nov  7 09:43:05 Raspi4 vito[1385]: 0
Nov  7 09:43:05 Raspi4 vito[1385]: Befehl: quit
Nov  7 09:43:05 Raspi4 vito[1385]: >SEND: 04
Nov  7 09:43:05 Raspi4 vito[1385]: <RECV: len=1 06 (0.0 ms)
Nov  7 09:43:05 Raspi4 vito[1385]: <RECV: received 06
Nov  7 09:43:05 Raspi4 vito[1385]: >FRAMER: closed
Nov  7 09:43:05 Raspi4 vito[1385]: Process 1385 released lock
Nov  7 09:43:05 Raspi4 vito[1385]: Verbindung beendet (fd:2)
Nov  7 09:43:05 Raspi4 vito[1385]: Child Prozess pid:1385 beendet

Hat jemand eine Idee, wie man den (oder wahrscheinlich die) Setzbefehle erfolgreich ausführen kann?

Danke!

hmueller01, 2017-11-07:
Hallo,

verwendest Du das P300 oder das KW Protokoll?
Mit dem P300 Protokoll kann man mit der Version r106/r107 der sf Quellen nichts setzen.
Entweder auf KW Protokoll wechseln oder den Patch einspielen
[https://sourceforge.net/p/vcontrold/patches/3/](https://sourceforge.net/p/vcontrold/patches/3/)
Allerdings ist der für die r106 gemacht und hat vmtl. ein paar Konflikte bei der r107 ...

Gruß Holger

Hardy74, 2017-11-07:
Hallo Holger,

ich nutze das 300er Protokoll in der Version, wie es am 30.10.17 abends auf dem svn lag. (Wie ermittel ich die Version?)
Spannender Weise kann ich die Warmwassertsolltemperatur setzen, wenn auch mit Fehlermeldungen (hier im Forum auch zitiert), der Wert wird aber übernommen.
Meine Frau hatte direkt an der, sich auf Absenkung befindlichen Heizung, zwischenzeitlich auf Partybetrieb gestellt, ich sitze in der Firma und siehe da, wie von Geisterhand konnte ich nun setPartybetriebM1 setzen und zurück setzen. Der get-Befehl lieferte zumindest entsprechend folgend 1 oder 0. Wie sich die Heizung nun tatsächliche verhält, kann ich aus der Ferne Stand jetzt nicht sagen.

Grüße
Hardy

hmueller01, 2017-11-07:
Hi,
ist das Release 107, welches komischerweise ein paar Zeilen meines Patches enthält. Habe die (damaligen) Entwickler schon persönlich angeschrieben. Leider keine Antwort ...
Ich bin gerade dabei einen neuen Patch für die jetzt hier nach github umgezogenen sourcen auf Basis des r107 zu machen. Muss ich erst noch testen. Etwas Geduld ...

Hardy74, 2017-11-08:
Hallo Holger,
werden denn dann, wenn Du fertig bist, alle SET Befehle funktionieren? Wenn nicht, muß ich mir ja ohnehin was anderes überlegen!?
Grüße
Hardy

hmueller01, 2017-11-08:
Das weiß ich nicht genau. Theoretisch schon. Ich wechsele bei mir z.B. die Betriebsart und die Solltemperatur. Das geht.

Hardy74, 2017-11-08:
Gerade die Betriebsart funktioniert nicht. Das Abfragen der Betriebsart, hier Party, führt nach dem Setzen auf 1, wieder zur 0. Das Setzen der WW SollTemp haut zwar einige Fehler raus, scheint aber von der Heizung übernommen worden zu sein. Die Abfrage liefert zumindest den neuen Wert.

vclient -h localhost:3002 -c "setBetriebPartyM1 1"
setBetriebPartyM1 1:
OK

vclient -h localhost:3002 -c getBetriebPartyM1
getBetriebPartyM1:
0

 vclient -h localhost:3002 -c "setTempWWsoll 56"
SRV ERR: >FRAMER: addr was still active FE06
>FRAMER: Command send
>FRAMER: unexpected length 2 02
Fehler recv, Abbruch
Fehler beim ausfuehren von setTempWWsoll 56
setTempWWsoll 56:
server error

getTempWWsoll:
56.000000 Grad Celsius

hmueller01, 2017-11-08:
Vieles hängt auch davon ab, wie man die vito.xml/vcontrol.xml bedatet hat.
Bei mir liefert das Kommando
```
vclient -h localhost:3002 -c "getPartyMode_M2"
getPartyMode_M2:
AUS
```
Will ich es einschalten muss ich
```
$ vclient -h localhost:3002 -c "setPartyMode_M2 EIN"
setPartyMode_M2 EIN:
OK
$ vclient -h localhost:3002 -c "getPartyMode_M2"
getPartyMode_M2:
EIN
```
verwenden.

Ich habe die Quellen gepatched und den pull request [#3](https://github.com/openv/vcontrold/pull/3) angelegt.
Kannst ja mal den Stand von [mir](https://github.com/hmueller01/vcontrold/tree/cmake) laden und schauen, ob es damit geht ...
Gruß Holger

Hardy74, 2017-11-09:
Hallo Holger,

folgendes habe ich eben gemacht:
- vcontrold gestopt
- das bestehende Verzeichnis in home vcontrold-code umbenannt in _vcontrold-code
- die Qullen via git clone https://github.com/openv/vcontrold.git vcontrold-code geholt und mit cmake übersetzt.
- make und sudo make install ausgeführt
- vcontrold gestartet

Es hat sich leider garnichts am Verhalten geändert.

Sind Deine Änderungen noch nicht eingeflossen, sprich, sollte ich sie mir von Deinem Verzeichnis holen und nochmal übersetzen?

hmueller01, 2017-11-09:
Ja, Code ist noch nicht aufgenommen. Werden das vmtl. auch anders machen. Siehe openv/vcontrold pull request #3.

Hardy74, 2017-11-09:
Ich habe nun Deine Quellen mal hergenommen, leider mit unverändertem Ergebnis.

Es gibt 2 vcontrold auf dem Raspi. Eine in /usr/sbin/, der Zeit nach vom ersten Schuß vorhin, die zweite in /usr/local/sbin/, der Zeit nach von der Übersetzung Deines Codes. Which meint, daß Deine vcontrold diejenige welche sei...

hmueller01, 2017-11-09:
Lösche am Besten beide und mache ein neues make install. Dann bist Du sicher. Bei mir hast du den cmake branch ausgecheckt, oder (hast also kein ./configure verwendet, sondern cmake .. im build Verzeichnis)?

Hardy74, 2017-11-09:
Spannender Weise liefert "git clone https://github.com/hmueller01/vcontrold.git" nicht den Inhalt mit CMakeLists.txt, sondern einen Ordner vcontrold und xml-32 sowie einer LIESMICH.txt.
cmake funktioniert also nicht. Deswegen habe ich vorhin auch autobuild, configure, make und make install verwendet.

Hardy74, 2017-11-09:
Über zip habe ich den branch bekommen und mittels cmake übersetzt, dann make und sudo make install.
Nun wird das USB Device nicht mehr geöffnet, lsof ergibt, daß das Device von niemenandem benutzt wird.
Ergo folgt auf
 vclient -h localhost:3002 -c getBetriebPartyM1
ein
[736] Thu Nov  9 14:27:49 2017 : TTY Net: Keine Verbingung zu localhost:3002
[736] Thu Nov  9 14:27:49 2017 : Keine Verbindung zu localhost

Reboot ändert daran nichts.

speters, 2017-11-09:
@Hardy74 bei deinem git clone hast du den master Branch ausgecheckt. Du musst in dem Verzeichnis noch auf den Branch wechseln, den du haben willst (git checkout branchname)

Hardy74, 2017-11-09:
Das habe ich mir gedacht, weswegen ich den Umweg über das Zip gemacht habe, mit dem vorhin zitierten Ergebnis.

hmueller01, 2017-11-09:
Was sagt denn /var/log/vcontrold (wenn das bei Dir angelegt wird)?
@speters Danke für die Hilfestellung. Wollte es Hardy74 auch schon schreiben.

Hardy74, 2017-11-09:
In /var/log existiert kein vcontrold.
Der vcontrold existiert zusammen mit vclient und vsim in dem Ordner, wo sie von cmake hingebaut worden sind. Offenbar hat da bei  etwas nicht so ganz hingehauen wie es sollte!?
Im syslog ist seit dem cmake bau auch nichts mehr von vito zu lesen.

hmueller01, 2017-11-09:
Was steht denn in Deiner logging section in /etc/vcontrold/vcontrold.xml?
Ich habe da <file>/var/log/vcontrold.log</file> stehen ...

Hardy74, 2017-11-09:
Da steht folgendes:
       <logging>
                <file>vcontrold.log</file>
                <syslog>y</syslog>
                <debug>y</debug>
        </logging>

Es gibt auf dem ganzen Raspi kein vcontrol.log. Ich denke, vcontrol wird nicht ausgeführt, da es am Ende des Builds nicht an die Stelle kopiert wurde, an die es gehört. Vermutlich weil die Rechte fehlten!? Ich werde es nochmal mit sudo cmake probieren!?

hmueller01, 2017-11-09:
sudo cmake ist nicht notwendig. cmake legt nur die Makefiles entsprechend an.
Schau Dir bitte noch einmal das Vorgehen unter 
[INSTALL.md](https://github.com/openv/vcontrold/blob/master/doc/INSTALL.md)
an. Nur das make install benötigt ein sudo (was hier noch nicht eindeutig beschrieben ist).
Meine vcontrold.xml config sieht so aus:
```
  <unix>
     <config>
        <serial>
                <tty>/dev/vitoir0</tty>
        </serial>
        <net>
                <port>3002</port>
                <allow ip='127.0.0.1'/>
                <allow ip='192.168.2.0/24'/>
        </net>
        <logging>
                <file>/var/log/vcontrold.log</file>
                <syslog>n</syslog>
                <debug>n</debug>
        </logging>
        <device ID="20CB"/>
     </config>
  </unix>

```
Leider kann man Deine Anwort nicht richtig lesen, da du den Text nicht als Code formatiert hast. Dann fallen die ganzen Klammern bei HTML weg und man kann nicht mehr viel verstehen. Wenn da nur vcontrold.log ohne Pfad seht, dann wird die log irgendwo angelegt. Vmtl. in dem Directory in dem Du vcontrold startest. Aber nur eine Vermutung. Viel Erfolg! Das log File ist wichtig, da Du nur so heraus bekommst woran es klemmt. Sonst poste auch noch mal den ganzen Vorgang exakt und die Ausgaben.

Hardy74, 2017-11-10:
Moin Holger,

was ich nun gemacht habe, hänge ich mal in Dateien an. Ich hatte mir gestern ja den Branch als Zip heruntergladen. Als KlickerBunti-TortoiseHg-Mercurial Benutzer ist mir Kommandozeilen svn/git nicht geläufig genug...
[cmake.txt](https://github.com/openv/openv/files/1460274/cmake.txt)

Weiterhin habe ich den logging Eintrag in der vcontrold.xml an Deinen angepaßt. Es scheint aber, als würde vcontrol nicht gestartet werden, obwohl der Status etwas anders sagt.
[status.txt](https://github.com/openv/openv/files/1460279/status.txt)
sudo lsof auf /dev/ttyUSB0 sagt auch nichts, sprich, das Device wird offensichtlich nicht verwendet.

Syslog:
[syslog.txt](https://github.com/openv/openv/files/1460287/syslog.txt)

Ich hoffe, so ist es nachvollziehbarer.

Grüße,
Hardy

hmueller01, 2017-11-10:
Ja, läuft nicht. Sieht man hier
> Active: active (**exited**) since Fri 2017-11-10 07:24:12 CET; 1min 7s ago

Vmtl. bringt auch `ps aux|grep vcontrold` nix raus.
Was kommt denn für eine Ausgabe bei 
`vcontrold -n` bzw. `vcontrold -n -g`

Hardy74, 2017-11-11:
pi@Raspi4:~ $ vcontrold -n
Konnte /var/log/vcontrold.log nicht oeffnen Permission deniedpi@Raspi4:~ $ sudo vcontrold -n
[18259] Sat Nov 11 12:50:29 2017 : TCP socket 3002 geoeffnet

Mit -n -g ist die Ausgabe zu lang, ein lenken in eine Datei mittels > funktioniert warum auch immer nicht.. Die Datei bleibt 0Byte groß, die Ausgabe wandert in die Console...

Hardy74, 2017-11-11:
pi@Raspi4:~ $ sudo ps aux|grep vcontrold
pi       18317  0.0  0.0   4776   548 pts/0    S+   12:58   0:00 grep --color=auto vcontrold

hmueller01, 2017-11-11:
Das Logging geht nach stderr nicht nach stdout. ">" lenkt nur stdout um.
Trotzdem verstehe ich nicht, warum das bei Dir nicht mehr läuft und warum die /var/log/vcontrold.log nicht angelegt wird. Beendet sich der deamon von selbst, wenn Du ihn mit vcontrold -n startest, oder drückst Du dann ^C? Wenn er sich von selbst beendet (was er nicht sollte, der er mit -n nicht in den Hintergrund geschickt wird), was sind die letzten Meldungen? Irgendwelche Fehler?

Wenn das alles nicht hilft, versuche noch mal die original Version von SF zu übersetzen und zu testen, ob es mit Deiner Konfiguration grundsätzlich noch geht.

Hardy74, 2017-11-11:
Ich beende ihn mit ^C, händisch mit sudo und ohne Parameter gestartet kann ich mit telnet und auch vclient ganz normal eine Verbindung aufbauen. Offenbar kneift es beim automastischen Starten des Daemons...
Was nach wie vor auch so nicht funktioniert ist das Setzen des Partybetriebes auf 1, Antwort ist wie gehabt ok, die Abfrage liefert dann aber wie gehabt 0.
Anbei die vcontrold.log...
[vcontrold.log](https://github.com/openv/openv/files/1463929/vcontrold.log)

Hardy74, 2017-11-17:
Hallo Holger,
bist Du an der Sache noch dran? Primär daran, daß das Setzen des Partybetriebes nicht funktioniert!?
Grüße
Hartwig

hmueller01, 2017-11-17:
Hallo Hartwig,

grundsätzlich schon. Siehe [pull request #3](https://github.com/openv/vcontrold/pull/3) und [issue #13](https://github.com/openv/vcontrold/issues/13).
Aber noch keine richtige Lösung gefunden ...

Dennoch, mit dem aktuellen master und der Verwendung einer unit (statt 0 und 1) sollte es gehen.
Hast Du deinen Party Mode so konfiguriert (auf die unit kommt es an)?
vito.xml:
```
    <command name='getPartyMode_M2' protocmd='getaddr' >
      <addr>3330</addr>
      <len>1</len>
      <unit>E_ST</unit>
      <description>Party mode</description>
    </command>
    <command name='setPartyMode_M2' protocmd='setaddr' >
      <addr>3330</addr>
      <len>1</len>
      <unit>E_ST</unit>
      <description>Setze Party mode</description>
    </command>
```
vcontrold.xml:
```
    <unit name='Status' >
      <abbrev>E_ST</abbrev>
      <type>enum</type>
      <enum bytes='00' text='AUS'/>
      <enum bytes='01' text='EIN'/>
      <enum text='UNKNOWN'/>
    </unit>
```

Hardy74, 2017-11-18:
Hallo Holger,
in der vito.xml stand als unit RT, was ich zu E_ST geändert habe. E_ST gab es in der vcontrold.xml nicht, ich habe Deinen kompletten Schnipsel eingefügt, das Ergebnis sieht nun optisch anders aus, verhält sich leider aber gleich:
pi@Raspi4:~ $ vclient -h localhost:3002 -c getBetriebPartyM1
`getBetriebPartyM1:`
AUS
pi@Raspi4:~ $ vclient -h localhost:3002 -c "setBetriebPartyM1 EIN"
setBetriebPartyM1 EIN:
OK
pi@Raspi4:~ $ vclient -h localhost:3002 -c getBetriebPartyM1
getBetriebPartyM1:
AUS
pi@Raspi4:~ $

Als Basis dient mir die vito_p300.xml, in der ja ein "ceteris paribus" einiges angepaßt hat!?
Vielleicht kannst Du mir ja mal Deine beiden XMLs schicken? Mit Chance liegt es daran!?

`              <command name='getBetriebPartyM1' protocmd='getaddr'>
			<addr>2303</addr>
			<len>1</len>
			<unit>E_ST</unit>
			<description>Betriebsart Party M1</description>
			<device ID="2053" />
		</command>

                <command name='setBetriebPartyM1' protocmd='setaddr'>
			<addr>2303</addr>
			<len>1</len>
			<unit>E_ST</unit>
			<description>Setze Betriebsart Party M1</description>
			<device ID="2053" />
		</command>`

In einem weiteren Versuch habe ich dann noch mal mit viel Unwissenheit die device-Zeile entfernt, was am Ergebnis allerdings nichts änderte.

Hardy74, 2017-11-18:
Dann nochmal den get Eintrag als Code, da klappte offensichtlich etwas nich...

`             <command name='getBetriebPartyM1' protocmd='getaddr'>
			<addr>2303</addr>
			<len>1</len>
			<unit>RT</unit>
			<description>Betriebsart Party M1</description>
			<device ID="2053" />
		</command>`

hmueller01, 2017-11-18:
Wie war bzw. ist denn RT in der vcontrold.xml definiert?
Wenn Du den Party Mode an der Anlage einschaltest, kommt dann ein EIN bzw. 01 zurück? Also die Frage ist, stimmt die Adresse 100%ig?
Wie gesagt, an meiner Anlage mit der dafür richtigen Adresse tut es eigentlich.

Hardy74, 2017-11-18:
`     <unit name='ReturnStatus'>
	   <abbrev>RT</abbrev>
	   <type>enum</type>
	   <enum bytes='00' text='0'/>
	   <enum bytes='01' text='1'/>
	   <enum text='NOT OK'/>
	</unit>`

Laut https://openv.wikispaces.com/Adressen ist die Adresse 2303 korrekt.

Ich habe eben 'Comfort' wie es heißt, direkt auf der Anlage eingeschaltet.
siehe da:
pi@Raspi4:~ $ vclient -h localhost:3002 -c getBetriebPartyM1
getBetriebPartyM1:
EIN

Dann habe ich es an der Anlage ausgeschaltet:
pi@Raspi4:~ $ vclient -h localhost:3002 -c getBetriebPartyM1
getBetriebPartyM1:
AUS

Dann habe ich es an der Anlage eingeschaltet und folgendes probiert
pi@Raspi4:~ $ vclient -h localhost:3002 -c "setBetriebPartyM1 AUS"
setBetriebPartyM1 AUS:
OK

Die Anlage wie auch das get folgte, der Partybetrieb war aus. Der nun logische Versuch des Ein- wie auch Ausschaltens verlief mehrfach erfolgreich. Die Anlage wie auch das get folgte korrekt.

Haken an der Sache:
Genau das Verhalten hatte ich aus der Ferne schon mal, als meine Frau zu Hause mal den Comfort einschaltete. Da funktionierte es auch schon mal. Demnach wahrscheinlich wohl, bis die Anlage das nächste Mal in die Absenkung geht (im Moment ist sie im Tagbetrieb). Ich würde insofern erwarten, daß es dann (heute Abend bzw. morgen) wieder nicht mehr funktioniert.
Wenn dem so ist (ich werde berichten), wäre wohl Viessmann Insiderwissen gefragt!??

Hardy74, 2017-11-19:
Wie erwartet, funktioniert das Einschalten des Partybetriebes jetzt wieder nicht mehr.
Das Einzige, was zwischenzeitlich passiert ist, die Heizung ist in die Absenkung gegangen und jetzt wieder in den Tagbetrieb. Kein Reboot des Raspis, nichts sonst.

Hardy74, 2017-11-23:
Hallo Holger,
hast Du eine Idee für das Verhalten oder läßt sich das nicht abstellen?

hmueller01, 2017-11-23:
Hallo Hartwig,

leider nein. Es hört sich so an als ob es aber nicht an vcontrold liegt. Ich dachte ja erst, dass es das bekannte Problem bei "unit off" ist. Wusste damals aber noch nicht, das Dein enum "RT" auf "0" und "1" hört (statt auf EIN/AUS o.ä.). Ist eigentlich gar nicht schlecht das so zu machen, da sich dass dann später durch andere Anwendungen gut auswerten lässt. Ich gehe bisher immer den Umweg über "unit off", um 0 und 1 zu bekommen ...

Bei meiner Anlage scheint das aber richtig zu funktionieren. Ist aber auch eine VScotHO1 (20CB) allerdings nur mit einem verwendeten Heizkreis auf M2.

Du setzt die Adresse 2303 für M1. Das müsste dann 3303 für den M2 sein. Die hatte ich in einer alten config auch mal. Irgendwo hatte ich dann die Adresse 3330 für M2 gefunden. Müsste die 2330 für den M1 sein. Kannst ja mal versuchen die Adresse zu lesen und den Party Mode ein und aus zu schalten, ob sich bei Deiner Anlage an der Adresse daraufhin was tut ...
Wenn ja versuch's zu schreiben und schaue ob dann geht.

Holger

Hardy74, 2017-11-23:
Hallo Holger,

der Tip war brandheiß! Nun funktioniert es!!! :-)

Ich habe diegleiche Steuerung wie Du, auch mit nur einem Heizkreis. Offenbar ist doch dann die Doku zu den Adressen hier im Forum falsch!? Und warum es mit der alten Adresse zumindest bis zum nächsten Betriebsartwechsel funktionierte, wenn man es einmal direkt an der Anlage eingeschaltet hatte, muß man auch nicht verstehen?
Der "Kumpel" von Party, der Sparbtrieb funktioniert auch. Die Adresse, 2302, scheint demnach korrekt zu sein.

Weißt Du zufällig, ob die Steuerung hinreichend intelligent ist, oder ob es kostspielige Probleme gibt, wenn man beispielsweise aus dem Partymodus direkt den Ecomodus einschaltet oder vice versa? Auf der Viessmann App war das meine ich nicht möglich, man mußte erst den einen ausschalten, um den anderen einzuschalten.

Es scheint mir auch zu viele xml's zu geben. Alleine mit dem vcontrol Download bekommt man ja 2. Schick wäre doch genau eine zu haben, die für alle Steuerungen gilt? Wenn ich die XML quer lese, kann man doch mit dem device-tag Befehle unausführbar machen, wenn sie für die spezifizierte Steuerung nicht verfügbar sind!?

Lauern das vielleicht noch mehr Adreßfehler? Wenn bei Dir alle Befehle funktionieren, kannt Du mir ja vielleicht mal Deine xml's zur Verfügung stellen!?

So weit erstmal... Nochmal vielen Dank für Deinen Tip!

hmueller01, 2017-11-23:
Prima. Das ist gut. Ja leider gibt es sehr viele Quellen mit Adressen. Ich hatte es damals aus einer google Tabelle und der [vito.rampro.de](vito.rampro.de) Datenbank. Leider geht der Zugriff mit dem alten User+Pass nicht mehr.

Ich stecke leider nicht so tief drin, ob man mit dem device tag das Kommando für verschiedene Anlagen steuern kann. Vielleicht kann da @vitoopen weiter helfen. Von der Idee wäre dann aber eine vcontrold.xml für alle Anlagen möglich ... Das wäre mal ein richtig gute Sache, da es momentan (auch von wiki hier) ein ziemliches Chaos ist ...

Ob man vom Party- direkt in den Ecomodus und umgekehrt schalten darf, weiß ich leider auch nicht. Die ganze Sache hier ist schon mit einem gewissen Risiko verbunden, da derartige Zugriffe ja von Viessmann so nicht für den Endkunden gedacht waren ...

speters, 2017-11-23:
Eine einzige Konfiguration für alle Anlagen ist wg. https://github.com/openv/vcontrold/issues/15 in vcontrold nicht möglich.
Ich bin aber gerade dran, das ein wenig aufzudröseln.

Hardy74, 2017-11-24:
Hallo Holger,

offenbar zu früh gefreut... der PartyBetrieb funktioniert nach wie vor. Was jetzt aber nicht mehr funktioniert, ist der Sparbetrieb. Er bleibt schlicht aus, dasselbe Verhalten wie bis gestern beim Partybetrieb. Hast Du an der Stelle vielleicht eine andere Adresse für den Sparbetrieb, mit der ich es probieren kann? Derzeit ist es die 2302.
Der einzige Unterschied zu gestern Abend ist, daß die Heizung momentan in der Absenkung ist, gestern Abend war sie im Tagbetrieb. Händisch direkt an der Heizung etwas tun kann ich gerade nicht, da ich aus der Ferne zugreife.

Ein wenig von diesem Thread abweichend, finde ich es auch merkwürdig, daß es offenbar keinen Befehl gibt, mit dem erfragen kann, ob die Heizung gerade in der Absenkung ist oder im Tagbetrieb (gleiches für WW und Zirku). Beholfen habe ich mir an der Stelle über den Umweg, zu prüfen, ob die aktuelle Zeit innerhalb oder außerhalb einer Schaltzeit liegt.
Ebenso frage ich mich, warum es keinen Befehl für die Einmalladung gibt!? Gibt es den nicht, oder hat den einfach noch keiner ausgemichelt!?

So weit erstmal...

Grüße
Hartwig

hmueller01, 2017-11-24:
Hallo,

stimmt, ich wollte meine [vito.xml](https://github.com/openv/openv/files/1500999/vito_complete.xml.txt) hier mal einstellen. Da habe ich für den M2 eine andere Adresse (3331). Heißt bei mir EcoMode. Für den M1 könnte das die 2331 sein. 

Gruß Holger

Hardy74, 2017-11-24:
Leider funktioniert es mit der 2331 auch nicht, das Ergebnis ist dasselbe. Eco/Spar bleibt aus.
Kann das vielleicht von Viessmann gewollt sein, daß wenn Absenkung kein Eco möglich ist? Das werde ich sonst mit der Viessmann App bzw. auch direkt auf der Heizung mal probieren. Zu Party kann man ja eine Temperatur setzen (was ich in deiner xml auch nicht finde), womit es Sinn machen würde, auch im Tagbetrieb den Partybetrieb zuzuschalten, weil man es zur Party lieber wärmer/kälter hätte als normal. Wenn man bei Eco aber keine Temperatur angeben kann, würde es auch keinen Sinn machen, diesen in der Absenkung einzuschalten!? Für längere Absenkungen gibt es ja den Ferienmodus.

Danke für die XML. Nur damit es bei den Einheiten nicht kneift, benötigt man doch auch zumindest noch den units-Teil der vcontrold.xml!?
Die Fleißarbeit wäre dann, zu Deinem Heizkreis 2, die Kommandos sprich Adressen für Heizkreis 1 anzupassen.

Wenn Du den units-Teil hier noch mal schickst, werde ich die xml's demnächst mal zusammenführen.

Hardy74, 2017-11-24:
Da meine Heizung um 12 in den Tagbetrieb ging, habe ich direkt nochmal das Setzen des Spar-/Ecobetriebes  probiert und siehe da, nun funktioniert es wieder, zumindest sagt der get-Befehl dann auch EIN.
Vielleicht kannst Du das an Deiner Heizung ja auch mal probieren. Wenn sie sich in der Absenkung verhält wie meine, könnte meine Theorie ja richtig sein...

hmueller01, 2017-11-24:
In der Absenkung kann ich kein EcoMode setzen. Das geht auch an der Anlage nicht. Macht ja irgendwie auch Sinn. Die Temperatur für den PartyModus zu setzen habe ich auch noch nicht gefunden.

Anbei noch meine [vcontrold.xml](https://github.com/openv/openv/files/1502509/vcontrold.xml.txt).

Hardy74, 2017-11-25:
So sieht das Setzen der Comforttemperaur in der Viessman-App aus: 
[image]

In der P300-vito.xml, die beim Download hier auch dabei war, stehen Befehle wie set/getTempPartyM1 (2308).  Ganzzahlig scheinen sie auch zu funktionieren.

hmueller01, 2017-11-25:
Danke. Leider kommt bei mir (3308 für M2) nichts sinnvolles raus (11°, obwohl in der Anlage 19° eingestellt ist).

PS: Der git master branch enthält jetzt meinen Fix für das set_addr Problem bei unit off [#13](https://github.com/openv/vcontrold/issues/13). Kannst es ja mal testen ...

Hardy74, 2017-11-26:
Werde ich demnächst mal testen. Was mich mehr umtreibt, ist die Geschichte mit der Einmalladung, sprich, letztlich: welche Adresse hat der Befehl? Für Schnüffelein fehlt mir eigentlich die Zeit, auch die um mir Kenntnisse über das Protokoll anzueignen...
Weiterhin wäre dringend auch mal Kosmetik an den XML's nötig, Party-Komfort-Mischmasch, diverse, für mich/meine Heizung irrelevante Befehle... etc... und nicht zuletzt das kuriose Verhalten, warum der vcontrold nicht mehr von alleine starten mag...
Mein primäres Ziel momentan: Die Heizung (auch für die Frau) Browser- sprich auch vom Handy aus bedienbar machen. Mit Eco- Comfort klappt das ja nun schonmal.

Hardy74, 2017-11-29:
Hallo Holger,

bekommt man auch Mails bei bereits geschlossenen Threads?
Falls nicht, nochmal hier mein Problem zu Deiner setaddr Geschichte:

Wirkt das Fix auch gegen folgendes Problem?

vclient -h localhost:3002 -c "setTempWWsoll 54"
SRV ERR: >FRAMER: error 15
Fehler recv, Abbruch
Fehler beim ausfuehren von setTempWWsoll 54
setTempWWsoll 54:
server error

Das Setzen der Raumtemperaturen funktioniert bei gleichen unit-Einstellungen (UTI) einwandfrei.
Adresse von setTempWWsoll: 6300, vielleicht stimmt die ja auch wieder mal nicht!? (VScotHO1, P300).

sigma415, 2024-12-13:
See solution here: https://github.com/openv/openv/issues/120#issuecomment-615083487
Set party mode duration (0x27F2) = 0 !