openv/openv #380: Sprache deutsch/englisch?

Jojo-A, 2017-11-09:
Hallo zusammen,

ich frage mich, ob es nicht sinnvoll wäre, das Repository hier (oder zumindest das Wiki) auf englisch umzustellen oder zumindest um entsprechende Seiten auf englisch zu ergänzen. Damit würde man evtl. ein breiteres Publikum ansprechen und auch projektübergreifende Kommunikation mit anderen Entwicklern wäre evtl. einfacher.
Ich würde zum Beispiel gern neue Issues erstellen, frage mich aber, in welcher Sprache ich das machen soll...
Wie das mit Git und diesen Sachen wie Forks und Commits und so funktioniert, weiß ich noch nicht so genau. Ich könnte aber, sofern gewünscht, mich an der Übersetzung ins Englische beteiligen.

Grüße,
Joachim

speters, 2017-11-09:
Das Wiki enthält momentan recht viele Sachen, die in projektspezifischen Repositories besser aufgehoben wären.
Wenn wir jetzt Inhalte durch Übersetzung quasi duplizieren, wird das noch unübersichtlicher.

Du kannst sehr gerne helfen, indem du z.B. schonmal die vcontrold-spezifischen Wikiseiten nach  https://github.com/openv/vcontrold bringst.
Dazu gerne einen Fork von https://github.com/openv/vcontrold erstellen und im _doc/_ Verzeichnis deines vcontrold-Repos die Wiki-Infos aufgearbeitet einpflegen. Hier kann die Hauptdoku gerne in Englisch gehalten werden. Danach per pull-request die Aufnahme ins openv/vcontrold Repo anfragen.

Jojo-A, 2017-11-09:
Oha, sowas in der Art hab ich befürchtet... ;) . Ich glaube es ist an der Zeit, daß ich mich mit diesem git-Kram mal intensiver befasse. Ich gucke mal, was ich machen kann.