openv/openv #382: Linux software

l3u, 2017-11-11:
Currently, we are liststing three linux projects which can talk to a Viessmann heating device:

- vcontrold
- vconnect
- vitalk

Fortuantely, vcontrold has come to new life (cf. https://github.com/openv/vcontrold ).

But what about the two ther ones? The last vconnect release seems to date back at 2012, the last commit to vitalk dates back to 2013.

If there's no good reason why there are multiple projects apparently doing the same thing, we definitely should pool our manpower to develop one project doing it (time's short, why would we re-invent the wheel?!). In this case, we have to try to contact the non-vcontrold developers and try to bring them into vcontrold development.

If there is a good raison d'être for vconnect and vitalk, we should examine if the projects are still alive. If not, we should remove the wiki reference to (now) dead upstream.

speters, 2017-11-24:
You're perfectly right about concentrating manpower.

``vitalk`` is clearly a dead end due to its implementation details and lack of user-configurability.
I moved the Wiki docs concerning this software to the openv/vitalk repo and added an EOL info.

``vconnect`` seems to predate FNobis's development efforts in ``vcontrold``.
I removed it from the Wiki and put it in a separate project where it can rest in peace and collect dust.

The Windows software is also an issue. It seems to be mostly closed-source and abandoned by the authors. I hope some Windows-running persons will do some cleaning in the Wiki.

l3u, 2017-11-24:
Nice :-)

Question is if Windows software (esp. if it's not Open Source and not maintained anymore) actually still acts a part. I think what most people do (with vcontrold) is monitoring and statistics of the heating/solarthermics. As we have cheap single-board computers like the Raspberry Pi nowadays, this is just perfect for this very job. You can't (or at least you don't want to) do this with a Windows machine.

And if you really want to use a Windows machine, you can compile vcontrold there via MinGW, can't you?

l3u, 2017-11-24:
Well okay, building via MinGW isn't that easy. Bitches about `syslog.h` in `common.c`, which one could cheap out by disabling the respective code (I think we should replace the depreceated `bzero` calls there by `memcopy` btw?), but next thing not to build then is `socket.c`. But possibly, one can build it via Cygwin? (Doesn't really fit here however ;-)