openv/openv #383: Probleme setaddr

Krawum, 2017-11-18:
Hallo!
Ich habe vor kurzem openv entdeckt und bin ganz begeistert über die Möglichkeiten. Ich würde mich auch gerne engagieren, falls es Bedarf gibt.

Ich lese aktuell Daten von einer Vitocal 300-G mit dem Protokoll KW2.

Dies ist mein Problem: Ich möchte die Neigung und Steigung der Heizkennlinie verändern. Lesen klappt, schreiben leider nicht.

Hat jemand eine Idee, woran das liegen könnte oder was ich ausprobieren könnte?

Vielen Dank!

Beim Aufruf in PuTTY kommt folgende Fehler:

> vctrld>setHKLSteigung 0.8
> DEBUG:Sat Nov 18 12:46:21 2017 : Befehl: setHKLSteigung 0.8
> DEBUG:Sat Nov 18 12:46:21 2017 : Send Exp:V*10 [V=0.800000]
> DEBUG:Sat Nov 18 12:46:21 2017 : Typ: short (Bytes: 08 )
> DEBUG:Sat Nov 18 12:46:21 2017 : >SEND: 04
> DEBUG:Sat Nov 18 12:46:21 2017 : >FRAMER: no preset result
> DEBUG:Sat Nov 18 12:46:21 2017 : Warte auf 05
> DEBUG:Sat Nov 18 12:46:23 2017 : <RECV: 05 (1980.0 ms)
> DEBUG:Sat Nov 18 12:46:23 2017 : >SEND: 01
> DEBUG:Sat Nov 18 12:46:23 2017 : >SEND: 00
> DEBUG:Sat Nov 18 12:46:23 2017 : >SEND: 02
> DEBUG:Sat Nov 18 12:46:23 2017 : >SEND: 30
> DEBUG:Sat Nov 18 12:46:23 2017 : >SEND: 07
> DEBUG:Sat Nov 18 12:46:23 2017 : >SEND: 01
> DEBUG:Sat Nov 18 12:46:23 2017 : >SEND: 08
> DEBUG:Sat Nov 18 12:46:23 2017 : >SEND: 00
> DEBUG:Sat Nov 18 12:46:23 2017 : >FRAMER: no preset result
> DEBUG:Sat Nov 18 12:46:27 2017 : <RECV: len=1 05 (3380.0 ms)
> DEBUG:Sat Nov 18 12:46:27 2017 : <RECV: received 05
> DEBUG:Sat Nov 18 12:46:27 2017 : Errstr matched, Ergebnis falsch (Retry:-1)
> DEBUG:Sat Nov 18 12:46:27 2017 : Ergebnis falsch, Abbruch
> DEBUG:Sat Nov 18 12:46:27 2017 : Fehler beim ausfuehren von setHKLSteigung 0.8
> ERR: Ergebnis falsch, Abbruch
> Fehler beim ausfuehren von setHKLSteigung 0.8
> vctrld>

Auszug aus der vcontrold.xml:

```
       <protocol name='KW2'>
       <macros>
         <macro name='SYNC'>
             <command>SEND 04;WAIT 05;SEND 01</command>
         </macro>
         <macro name='GETADDR'>
             <command>SEND F7</command>
         </macro>
         <macro name='SETADDR'>
             <command>SEND 00 02</command>
         </macro>
       </macros>
       <commands>
         <command name="sync">
              <send>SYNC</send>
         </command>
         <command name="getaddr">
              <retry>3</retry>
              <recvTimeout>150</recvTimeout>
              <send>SYNC;GETADDR $addr $hexlen;RECV $len $unit</send>
         </command>
 		<command name="setaddr">
 			<send>SYNC;SETADDR $addr $hexlen;SEND BYTES $unit;RECV 1 SR</send>
         </command>		
       </commands>
       </protocol>
```

Ich hatte anstatt "SEND 00 02" auch schon "SEND 01 F4" ausprobiert, was aber ebenfalls nicht funktioniert hat.

Auszug aus der vito.xml:

```
 <command name='setHKLSteigung' protocmd='setaddr'>
       <description>Heizkennlinie Steigung HK2*</description>
       <addr>3007</addr>
       <len>2</len>
       <error>05</error>
       <unit>UN</unit>
 </command>
```

MrIcemanLE, 2017-11-18:
So weit ich das verstanden habe müssen die Set-Befehle in einfache Anführungszeichen. Also so: 
vclient -h 127.0.0.1:1234 -c ‘setHKLSteigung 0.8‘

Krawum, 2017-11-18:
Danke für den Hinweis! Das war leider nicht die Ursache.

Ich habe es jetz aber hinbekommen. Es lag am SEND-Statement. Ich hatte "SEND 01 F4" falsch interpretiert. Die 01 gehört wohl bereits zum Sync-Statement, so dass für Set nur noch F4 gesetzt werden muss (siehe unten).
  
Viele Grüße

>      <protocol name='KW2'>
>        <macros>
>          <macro name='SYNC'>
>              <command>SEND 04;WAIT 05;SEND 01</command>
>          </macro>
>          <macro name='GETADDR'>
>              <command>SEND F7</command>
>          </macro>
>          <macro name='SETADDR'>
>              <command>SEND F4</command>
>          </macro>
>        </macros>
>        <commands>
>          <command name="sync">
>               <send>SYNC</send>
>          </command>
>          <command name="getaddr">
>               <retry>3</retry>
>               <recvTimeout>150</recvTimeout>
>               <send>SYNC;GETADDR $addr $hexlen;RECV $len $unit</send>
>          </command>
>  		<command name="setaddr">
>  			<send>SYNC;SETADDR $addr $hexlen;SEND BYTES $unit;RECV 1 SR</send>
>          </command>		
>        </commands>
>        </protocol>