openv/openv #384: VDensHO1/ device 20C2 - set commands gehen nicht

emichel11, 2017-12-09:
Hallo Zusammen,
ich habe versucht hier in dem Forum ein Hinweis zu bekommen warum bei mir alle SET Kommandos
nicht gehen. zB das Issue
https://github.com/openv/openv/issues/377
löst das Problem bei mir leider nicht. Ich habe eine VDensHO1. Die Abfragen an die Adressen 0xF8 und 0xF9 liefern folgends Ergebnis:
0x00F8 - getDevType
20 C2 00 04 00 00 01 10
0x00F9 - getCtrlId
received C2 00 04 00 00 01 10 
0x08E0 - getInventory
received 35 31 34 36 37 33 38 6F
Wenn ich zum Beispiel abfragen möchte wie die Schaltzeit eingestellt ist
0x2000 - getTimerM1Mo
received 3C FF FF 9D A0 9D B2 32
funtkioniert das. Aber beim Setzen einer Schaltzeit
0x2000 - setTimerM1Mo 7:00 8:00
[20249] Sat Dec  9 10:01:30 2017 : Command: setTimerM1Mo 7:00 8:00
[20249] Sat Dec  9 10:01:30 2017 : Cycle Time: 07:00 -> [38]
[20249] Sat Dec  9 10:01:30 2017 : Cycle Time: 08:00 -> [40]
[20249] Sat Dec  9 10:01:30 2017 : >SEND: 41
[20249] Sat Dec  9 10:01:30 2017 : >SEND: 0D
[20249] Sat Dec  9 10:01:30 2017 : >SEND: 01
[20249] Sat Dec  9 10:01:30 2017 : >SEND: F4
[20249] Sat Dec  9 10:01:30 2017 : >SEND: 20
[20249] Sat Dec  9 10:01:30 2017 : >SEND: 00
[20249] Sat Dec  9 10:01:30 2017 : >SEND: 08
[20249] Sat Dec  9 10:01:30 2017 : >SEND: 38
[20249] Sat Dec  9 10:01:30 2017 : >SEND: 40
[20249] Sat Dec  9 10:01:30 2017 : >SEND: FF
[20249] Sat Dec  9 10:01:30 2017 : >SEND: FF
[20249] Sat Dec  9 10:01:30 2017 : >SEND: FF
[20249] Sat Dec  9 10:01:30 2017 : >SEND: FF
[20249] Sat Dec  9 10:01:30 2017 : >SEND: FF
[20249] Sat Dec  9 10:01:30 2017 : >SEND: FF
[20249] Sat Dec  9 10:01:30 2017 : >SEND: 9C
[20249] Sat Dec  9 10:01:30 2017 : <RECV: len=1 06 (20.0 ms)
[20249] Sat Dec  9 10:01:30 2017 : <RECV: received 06
[20249] Sat Dec  9 10:01:30 2017 : >FRAMER: Command send
[20249] Sat Dec  9 10:01:30 2017 : >FRAMER: no preset result
[20249] Sat Dec  9 10:01:35 2017 : <RECV: read timeout
[20249] Sat Dec  9 10:01:35 2017 : <RECV: received
[20249] Sat Dec  9 10:01:35 2017 : >FRAMER: read failure
[20249] Sat Dec  9 10:01:35 2017 : Error in recv, terminating
[20249] Sat Dec  9 10:01:35 2017 : Error executing setTimerM1Mo 7:00 8:00

Einen TimeOut bekomme ich bei jedem Set Kommando.
Ich verwende das 300er Protokoll
Hat jemand eine Idee?? Ist die Firmware der Anlage zu alt??

Ich verwende das 300er Protokoll

speters, 2017-12-09:
Dein Log zeigt eine fehlerhafte Kommunikation, die wie eine Mischung aus KW und P300 Protokoll aussieht.
Es wird ein P300 Telegrammstart mit 0x41 gesendet und eine Telegrammlänge 0x0D.
Dann geht es à la KW Protokoll weiter, offenbar mit einem 0x01 als ACK und einem 0xF4 für's Schreiben.
Richtig wäre für P300 hier 0x00 zur Kennzeichnung einer Anfrage und dann 0x02 für Schreiben.

Deine Konfiguration ist also irgendwie im Argen.

emichel11, 2017-12-09:
Hallo Speters,
vielen Dank für deine Anmerkung!
Ich verwende die Konfiguration wie in Github
https://github.com/openv/vcontrold
Ich nahm eigentlich an dass es funktioniert.
Gibt es eine Beschreibung über das P300 Protokoll in Netz, um es zu korrigieren?
Vielen Dank!

hmueller01, 2017-12-09:
Stimmt. Die Config für P300 SETADDR ist falsch. 01 F4 statt 00 02. Solltest Du mal korrigieren Sönke.

emichel11, 2017-12-09:
Danke Sönke,
wie mache ich das??
In vito.xml steht
    <command name="setTimerM1Mo" protocmd="setaddr">
      <addr>2000</addr>
      <len>8</len>
      <unit>CT</unit>
      <description>Schaltzeit Montag M1</description>
      <device ID="2053"/>
    </command>

Die adresse ist do 2000 für SetTimer Montag??
Kannst du mirt einen Link zu einer korrekten Konfig schicken bitte?
Vielen Dank!

speters, 2017-12-09:
Schau mal in https://github.com/openv/vcontrold/blob/master/xml/300/vcontrold.xml#L284 , das Protokollproblem ist dort nun korrigiert.

hmueller01, 2017-12-09:
Genau das muss in der vcontrold.xml für das P300 Protokoll stehen: 
```
      <protocol name='P300'>
	  <pid>41</pid>
	  <macros>
		<macro name='GETADDR'>
		    <command>SEND 00 01</command>
		</macro>
		<macro name='SETADDR'>
		    <command>SEND 00 02</command>
		</macro>
	  </macros>
	  <commands>
		<command name="getaddr">
		     <send>GETADDR $addr $hexlen;RECV $len $unit</send>
		</command>
		<command name="setaddr">
		     <send>SETADDR $addr $hexlen;SEND BYTES $unit;RECV 1 SR</send>
		</command>
	  	<command name="gettestaddr">
		     <send></send>
		</command>
	  </commands>
      </protocol>

```

emichel11, 2017-12-09:
Hallo Speters,
vielen Dank für den Hinweis. Der Patch behebt das Problem mit setTimerM1xxx
Jetzt bleibt nur das Problem mit
setBetriebPartyM1
Das ist mehrfach beschrieben
vctrld>getBetriebPartyM1
0
vctrld>setBetriebPartyM1 1
OK
vctrld>getBetriebPartyM1
0

Ich setze es - es kommt ok, ist aber nicht geschaltet.
hast du dafür auch eine Lösung?? Ich möchte die Heizung fernsteuern. Da bin ich nicht immer und möchte die Heizung Remote vorab einschalten.
Ansonsten muss ich wohl auf das Python Script das hier mal gepostet wurde zurück greifen

speters, 2017-12-10:
(Mit dem Python-Script meinst du wohl
https://github.com/openv/openv/issues/120#issuecomment-339569862)

Die Adresse `0x2303` ist m.E. zum Setzen des Partymodus für deine VDensHO1 korrekt.

Ich habe aber einen Workaround gefunden:

Man kann in `0x27f2` angeben, nach wie vielen Stunden der Partymodus zurückgesetzt werden soll.
Probier doch mal, dafür folgende Einträge in der _vcontrold.xml_ anzulegen:

```
<command name="getPartyDauer" protocmd="getaddr">
      <addr>27F2</addr>
      <len>1</len>
      <unit>ST</unit>
      <description>Begrenze Partybetrieb auf 1..12 Stunden. 0 beendet Partybetrieb bei nächster Umschaltung in Normalbetrieb.</description>
</command>
<command name="setPartyDauer" protocmd="setaddr">
      <addr>27F2</addr>
      <len>1</len>
      <unit>ST</unit>
      <description>Begrenze Partybetrieb auf 1..12 Stunden. 0 beendet Partybetrieb bei nächster Umschaltung in Normalbetrieb.</description>
</command>
```

Vor dem Setzen des Partybetriebes nun `0x27f2` auf 0 setzen (stand bei meiner VScotHO1_4 auf 8 (Auslieferungszustand?)).
Jetzt klappt das Setzen des Partymodus in `0x2303`.
Allerdings: Wenn man nun die Partydauer in `0x27f2` ändert, schaltet sich der Partymodus wieder ab.

Bei anderen Steuerungen gibt es einen Zähler (z.B. an 0x2536 oder 0x253a) , der die verbleibende Zeit im Partymodus angibt. Womöglich existiert ähnliches auch hier und müsste einen Wert >0 enthalten, damit der Partymodus angeschaltet bleibt.
Für VDensHO1 und VScotHO1 hab ich das aber nicht ausfindig machen können.

closed 2017-12-11 by emichel11