openv/openv #385: Berechnung Checksumme beim Protokoll 300

Bjoern3003, 2017-12-21:
Ich habe vermutlich irgendwie nen Denkfehler oder verstehe das Protokoll noch nicht ganz.

Ich frage 0x80 0x00 (Aussentemperatur) ab und erhalte zurück:
Hex 0x41 0x07 0x01 0x01 0x08 0x00 0x02 0x59 0x00 0x6c
Dec:    41    07     01     01    08     00     02    89     00  108

Checksumme ist 108 (0x6c). Bekomme ich auch genau so ausgerechnet. Alles gut.

Frage ich 0x08 0x04 (VitoTempWWIst) ab, erhalte ich zurück:
Hex: 0x41 0x07 0x01 0x01 0x08 0x04 0x02 0xef 0xbf 0xbd 0x01
Dec:     41    07     01     01    08     04    02  239  191   189     01

Checksumme wäre 190 (0xbd 0x01), ausrechnen würde ich aber 451.

Vielleicht kann mir ja jemand auf die Sprünge helfen, wie ich das zu berechnen habe.

hmueller01, 2017-12-24:
Hallo,

die Checksumme ist nur das letzte Byte. Nicht die letzten 2 Bytes. Siehe [P300 Protokoll](https://github.com/openv/openv/wiki/Protokoll-300).

Trotzdem komme ich bei der VitoTempWWIst nicht auf 0x01. 
Mein Ergebnis wäre 0x82. ??? Komisch ...

Bjoern3003, 2018-01-01:
Hallo, das ist eben mein Problem. Es passt einfach bei diversen Berechnungen nicht.

Heute habe ich noch mal ein wenig getestet. Nun kommt die VitoTempWWIst korrekt an. Dafür ist nun getVitoTempWWAuslauf fehlerhaft.

Da bekomme ich 0x41 0x07 0x01 0x01 0x08 0x14 0x02 0xef 0xbf 0xbd 0x00 zurück.

Bjoern3003, 2018-02-03:
Servus, das Problem besteht leider nach wie vor. Kennt sich vielleicht jemand mit dem Protokoll etwas besser aus und kann mir helfen? Könnte es vielleicht auch ein Übertragungsfehler sein (Dafür ist ja die Checksumme gedacht)? Wieso dann aber immer an der gleichen Stelle? Die ersten paar Anfragen werden ja korrekt beantwortet. Starte ich die Anfrage dann wieder von vorn, klappt es wieder bis zur Abfrage von VitoTempWWIst einwandfrei.

Morg42, 2018-03-15:
Was mich wundert, sind die 0x02 Datenbytes in der Antwort, aber es kommen - anscheinend - 3 Bytes + CRC zurück. Das passt doch schon nicht.

Bei mir ist TempWWIstOben auch 0x010d... (ist aber eine V200WO1C)