openv/openv #387: Software available for decoding KM bus

boblegal31, 2018-01-13:
Hello,

I successfully built a KM bus to ethernet adapter using NCN5150 IC plus LPC1769 evaluation board.
I can now successfully transmit all the data received on KM bus to ethernet. I haven't tested the emission on KM bus though.
I would like to know if anyone has already started working on decoding the KM bus data.

Thanks in advance

speters, 2018-01-15:
Great to hear that someone is working on this!
There is some basic information in the [KM-Bus Wiki pages](https://github.com/openv/openv/wiki/KM-Bus).

If you have trouble understanding the german Wiki text, don't hesitate to ask here.

boblegal31, 2018-01-16:
Ok,

so reading the pages regarding KM Bus, I could get the following concept :
 - messages exchanged on the bus are frames of various length called telegrams (I can successfully capture these telegrams on my setup, and I can see the different command fields, being 0x00,0x31, 0xB1, 0xB3 and 0xBF)
 - messages are supposed to be exchanged in a peer to peer basis

What I'm not sure to fully understand, is the concept of master and slave. I'm not sure if the master is always the same (the boiler regulation) or if the master/slave scheme is for each telegram exchange and though at any point of time, there is always one unit that is master (the boiler regulation, the remote control unit, the DCF77 unit ...) and one other unit that is slave. I encline to think the first assertion is correct.

What I completely don't understand, is the concept of tables and associated CMD1/2/3.

Thanks in advance

François

boblegal31, 2018-01-17:
For those interested, attached is a capture of all the traffic I could see on my KM bus during abour 1h.
The capture is a wireshark format capture.

Every message received on KM bus is sent as an UDP frame, with UDP destination port being set to 4000 + the command byte of the KM bus telegram (so that sorting is easier).

Duplicated messages (mostly ping commands) only appear once in the capture.

For the record, my boiler setup is the following :
- Vitodens 333F type FR3B
- Vitotrol 300 unit attached for heater circuit 1
- EA1 internal extension with attached hot water circulation pump

Hope it helps.

François

[Capture_busKM_1h.zip](https://github.com/openv/openv/files/1640375/Capture_busKM_1h.zip)

timob0, 2018-09-09:
Hello,
I‘m working on something similar based on a diy USB - mbus adapter. I got a new vitocrossal 300 gas heating recently and I‘m trying to hook it up to mqtt for logging.

Based on the information in the wiki, I hacked up a python script which in the end  should emulate  a Vitotrol remote control. At the moment, it gets through the registration process so that the heater controller believes a remote ccntrol is connected and it sends down the ping and the status datagrams.
Did you get further wirh your project and do you have insight into the commands the vitotrol send to the heater to request data?

Best regards
timo

boblegal31, 2018-09-10:
Hi timo.

 

Unfortunately, I did not have time to spend on that project for a while now.

All I have is MBUS network captures so far.

 

François

timob0, 2018-09-11:
Hi Francois,

no worries, thats fine. I took a look at the captures you shared. Essentially, they confirm the information on the wiki, however since only the communication from master to slaves is visible, it‘s hard to figure out what are the requests sent by the slave to query data.
Re. the messages, the one with 21 bytes contains several temperatures + heater status flags and I‘m seeing the same on my setup. It seems to be sent on change of values by the master. I‘m currently using it for logging temperatures.
The message with 40 bytes however, I haven‘t seen it so far. That one seems to be sent following a ping from the master and I believe the ping reply needs to have a certain structure to request it.
Also I wonder if there is a way for a slave to actively send something to the master without waiting for the ping cycle, like when you press a button on the remote control.
Would you have a way to capture the messages sent by the remote control?

Best
timo

boblegal31, 2018-09-11:
Hi Timo,

 

I don’t think we have the same understanding of the protocol.

To me, it looks like the master is broadcasting all the information periodically, so that the slaves do not have to query for data.

On the other hand, whenever a slave needs to send data (a new temperature set point for instance) , I guess it has to be sent right after a ping message.

 

That being said, I tried to capture those messages that the slave (a vitotrol 300 remote control in my configuration) sends whenever I try to change a setting (heating program or temperature), but I could not see those messages. I’ll have to try again.

 

François

boblegal31, 2018-09-25:
Hi Timo,

 

I decided to look back at this, and I figured out I was wrong in my previous reply.

 

First thing to note :

-        Using KM bus, the slaves on the bus will never “see” (receive) the telegrams sent by the other slaves (as slave to master communication is done through current modulation).

 

This implies that we can’t see the vitotrol and the internal extension (the only slave I believe, in my case) replying to ping requests.

 

I believe the 40 bytes telegram is about the program schedule (heater and hot water). Do you have a vitotrol ?

 

About the status message (21 bytes), I found some time ago some decoding information :

“

 <https://www.haustechnikdialog.de/Forum/t/59578/Vitotronic-vom-PC-steuern-ueberwachen?page=3> https://www.haustechnikdialog.de/Forum/t/59578/Vitotronic-vom-PC-steuern-ueberwachen?page=3

 

The KM bus is now most likely the M-BUS, because I have the first useful results. Please forget the telegrams mentioned above. There was one more mistake in programming. The protocol is 1200 baud, 8 data bits, 2 stop bits and even parity. Then comes the trick! At a certain point, the bytes are encoded with XOR (Exclusive Or) 0xAA. Example (all in HEX, coding as of byte 8): 

11 00 BF 15 01 01 1D AA AA E1 CF 95 BB B9 AE E1 AA A0 AA 78 05 without coding 
11 00 BF 15 01 01 1D 00 00 4B 65 3F 11 13 04 4B 00 0A 00 D2 AF with (B) XOR 0xAA coding 

Byte 1: 11 start bit 
Byte 2:? 
Byte 3: Address Vitotrol300? 
Byte 4: Telegram length (0x15> 21d) 
Byte 5:? 
Byte 6:? 
Byte 7: Parameter address? 
Byte 8: Upper 4-bit switching valve, lower 4-bit burner on / off 
Byte 9:? 
Byte 10: Boiler temperature (Actual): 37.5 C (0x4B> 75d / 2> 37.5) 
Byte 11: Hot water temperature (actual): 50.5 C (0x65> 101d / 2> 50.5) 
Byte 12:? 
Byte 13:? 
Byte 14: Outside temperature (actual): 9.5 C (0x13> 19d / 2> 9.5) 
Byte 15: HZ  <https://translate.googleusercontent.com/translate_c?depth=1&hl=fr&rurl=translate.google.com&sl=de&sp=nmt4&tl=en&u=https://www.haustechnikdialog.de/SHKwissen/1576/Pumpe&usg=ALkJrhicOOc3Uw7kF_jAp2OF3zAl6lma6w> pump or WW pump on / off 
Byte 16: is then boiler temperature (actual) 
Byte 17:? 
Byte 18: Operating mode: 0A = night mode, C6 = day mode, 86 = economy mode, 06 = party mode, 00 = shutdown mode 
Byte 19:? 
Byte 20: Check sum? 
Byte 21: Check sum? 

Then there are still telegrams with 38 bytes (11 00 BF 26 01 01 ...) where, among other things, values ​​such as room temperature setpoint, night setpoint, party temp. Setpoint, hot water temp setpoint, burner starts, burner time, heating phases, Hot water phases and hot water circulation pump - phases have found. These telegrams are only sent if the Vitotrol300 is connected to the KM bus. 
Although there are also different telegrams that are sent without Vitotrol300, but I have no conclusive results. 

Unfortunately, I can only listen to the master (Vitodens333 or Vitotronic200), since the M-BUS is so structured. Even though my M-BUS level converter can send telegrams via current modulation, I can only receive voltage modulation. Now you would have to build an M-BUS master to simulate a heater as a master. Then you would also know the telegrams from the Vitotrol300. The consummation would be sure to simulate such :). 

I would like to compile an excel list to get this out. Unfortunately, I do not have a server of my own and Comenius, as he said, does not seem to be back online before May. Would that work for you hgy_ch? Incidentally hgy_ch, the KM-BUS is terminal 145, you had asked at some point.

 

“

 

I started consolidating the statuses and temperatures, but it not mature enough to be shared here so far.

 

François

timob0, 2018-09-25:
Hi Francois,

many thanks for the reply. I agree to your analysis of the communication. It is directional in that slaves can hear only the master - not each other - and the master would only hear it's slaves.
This also is reflected in the fact that there are different transceiver chips for each role. The TI TS721A which is used in the Viessmann Equipment is a M-Bus slave chip.
I built a USB interface by hooking up such a transceiver to the KM-Bus terminal 145 of the Vitrotronic and connecting the Tx/Rx lines to a USB UART which gives a serial interface to receive and send on the Bus in a slave device role.

I do not have external Viessmann equipment on the KM Bus, the Vitotronic controller is the only device.

I was hoping you would have a different hardware solution which would be capable to capture the traffic sent by the slave devices.

Here's what I have so far:
- Without other devices attached, the Vitotronic only sends out two types of telegrams
1) Device discovery, like the following ones, it essentially walks through the devices it expects to find on the bus and queries each one for their serial number:
11:00:33:0a:02:01 - f8:04:84:ca
11:00:33:0a:03:01 - f8:04:3f:d6
11:00:33:0a:01:01 - f8:04:49:ef
07:00:33:0a:01:01 - f8:04:80:a9
07:00:33:0a:02:01 - f8:04:4d:8c
10:00:33:0a:01:01 - f8:04:f6:6e
10:00:33:0a:ef:01 - f8:04:3e:74
2) Current date and time encoded in BCD, seems to be a broadcast since the target address is 0xff
ff:00:b3:16:00:01 - 01:_20_:02:_18_:03:_09_:04:_23_:05:_06_:06:_06_:07:_28_:c9:b0

A device on the KM Bus needs to answer to the device discovery message with its serial number in a certain time window after reception of the telegram, the M-Bus spec says it should be something around 50ms + 11 bit times =~ 60ms.

After successful device discovery, there is another exchange, probably to confirm the detected device.

Once this is complete, the Vitrotronic starts sending ping telegrams to the device and it also starts sending the 21 byte telegrams with temperatures and burner/pump statuses.

However, I have not seen the 40 byte telegrams in my setup yet which is why I think that the Vitotrol device can somehow request them. I have not been able to figure out yet how that happens.

What I managed to do though is to send room temperature data down to the Vitotronic which appears on its on screen menu.

The Python program to communicate with the Vitrotronic is attached. It's just a dirty hack and by no means complete. It does however send temperatures into MQTT for logging and it can receive the room temp to send into the heating, so it is a very light Vitotrol emulation.
Ultimately, I'd like to integrate the heating into my home automation which is based on MQTT for communication and Home Assistant as a dashboard.
There still is an issue with this: the Vitotronic sets an error condition after some time and alomost immediately cancels it. This is after varying periods of time, so I suspect that I still haven't got the bus timing right.

Please find the following attached in the ZIP file:
kmBusAgent.py - the km bus to mqtt hack - USE AT YOUR OWN RISK ONLY
data.txt - 0x1d telegrams in decoded (XOR 0xAA) form
log.txt - communication log between Vitotronic and the Python program
0x1d_decode - my notes on the 21byte telegram

Happy to hear back soon :-)

[kmbus.zip](https://github.com/openv/openv/files/2417079/kmbus.zip)

boblegal31, 2018-09-25:
Hello,

I'm not sure to fully understand the 0xBF command, for sending orders/commands to the master.
On the wiki page, it is stated : "A tab from this is structured as follows: [storage no.] [Heating circuit (?)] [Cmd] ..."

I don't understand what "[storage no.]" refers to (maybe the english translation is wrong. Can anybody help ?

What I'm willing to do here, is to send a shutdown command for example, so I need to build a telegram for that matter.

Thanks in advance

François

boblegal31, 2018-09-25:
Hi Timo,

thanks for the quick answer.
So for my example to send a "shutdown" command, I would go with
00:11:bf:11:01:01:14:01:c8:00:00:00:00:00:00:00:b9:a3
and so "[storage no.]" refers to the table no ?

François

timob0, 2018-09-25:
Hi Francois,

unfortunately, the wiki is not really clear on the data structure. The way I read it is that for the tab 0x14, the payload length is 9 bytes including the 0x14 itself. Then, it says the format is [mem#][heat#][cmd] ...
where:
mem# - memory number out of 15 possible ones, maybe 0 in this case ?
heat# - heating circuit number, probably 1 ?
cmd - would be the 0xc8 for shutdown
... -  remaining 5 bytes, all set to 0 ?

so the message _might_ be:
[header] 14:00:01:c8:00:00:00:00:00 [checksum]

I tried to read the memories using the 0xe7 comand with tab 0x14 in different variations, but didn‘t get anything back. Also tried 0xe8 since that is indicated further down the page - no luck.
 Thats why I‘d be keen on an actual capture from a Vitotrol, that should give clarity.

Please let me know how things go with the above

boblegal31, 2018-09-28:
Hi Timo,

I'll try as you mention (but probably XORing the values after 0x14 with 0xAA) and report back.
In the mean time, I tried to probe the bus during the slave replies using an oscilloscope.

Good news is I can see the actual reply using AC coupling an inscreasing the vertical sensivity. Bad news is I can then no logner decode the master request. I'll try anyway to decode some of the telegrams I can see. I'll keep you updated.

François

timob0, 2018-09-29:
Hi Francois,

you're right on the XOR'in the bytes following the 0x14, let me know if it works once you have time to try out.

As per the mbus documentation here: http://www.m-bus.com/mbusdoc/md4.php, I think you probably would see a slight voltage dip when the slaves are sending data and it should be significantly lower than the voltage change when the master is sending.
I wonder if it's possible to amplify that voltage dip and feed it into a serial port so to log the slave communication. 

Timo

timob0, 2018-09-29:
Update:
I added the shutdown command to my script and - it works:
> Received 8 bytes datagram: 11:00:00:08:01:01 - 08:88
> Reply sent: 00:11:bf:11:01:01:14:aa:ab:62:aa:aa:aa:aa:aa:39:26

Following the above, the heating goes into "off" mode.
Likewise, commands 0xc9 turns back on warmwater mode and 0xca turns on full operation.

Now need to figure out how to send the set points and how to request data from the heating. Will experiment a bit more.

timo

boblegal31, 2018-10-01:
Hi Timo.

Thanks for the confirmation. I had no luck doing the same (though I have never tried the emission part of my software). I figured out why. With a vitotrol on the bus, it seems all the slots are being used, as you can see on the following bus captures.

About the question on the possibility to amplify the small voltage difference on the bus corresponding to slave emission, I doubt this is going to be possible : the voltage drop may largely depend on the wiring between the master and slave. On my setup (see picture 3), I get between 200 and 300mV. Maybe using some fined tuned op amp comparator would do the trick.

[image]
[image]
[image]

timob0, 2018-10-01:
Hi Francois,
Nice captures, many thanks for sharing. There‘s quite a lot going on on the bus. I think I‘ll do the same to compare to the timing I have since I‘m still seeing an error message om the Vitotronic.
I think you could try to emulate a second Vitotrol, for this you‘d have to do the pairing sequence when the Vitotronic tries to discover new devices. Then, as the emulated Vitotrol send the commands.
I wonder however if it‘s possible to pair a 2nd Vitotrol to the same heating circuit unless you have more than one. 
Apart from that, I accidently discovered that I can setup the vacation program using the tab 14 record. I found this when I tried to query data using 0x7e / 0x8e. The vacation was showing as 8 March 1973 for one day. I believe this might be actually the systems zero date since all the other bytes on the the telegram were set to 0x00.
Didn‘t however get the heating to send back anything,so I actually suspect that tab 14 is only for controlling heating modes. To me that makes sense since the 0xbf is for transferring a record and the 3f one would be for requesting. I think I‘ll experiment into that direction.

-timo

boblegal31, 2018-10-02:
Hi Timo,

I was wondering what your test process is, since I have never seen the 11 00 33 0A 01 01 F8 04 telegrams, even after hours of bus capture. Do you first run your python script, then turn on the heater ?

François

timob0, 2018-10-03:
Hi Francois,

I didn‘t do anything special, just connected to the KMBus port on the heater and the USB on the PC. Then wirh a serial terminal I see the messages, from there developed the python script.
The heater sends the below discovery messages every few minutes:
11:00:33:0a:02:01 - f8:04:84:ca
11:00:33:0a:03:01 - f8:04:3f:d6
11:00:33:0a:01:01 - f8:04:49:ef
07:00:33:0a:01:01 - f8:04:80:a9
07:00:33:0a:02:01 - f8:04:4d:8c
10:00:33:0a:01:01 - f8:04:f6:6e
10:00:33:0a:ef:01 - f8:04:3e:74

So in your setup, do you see any of these or just not the specific one you mentioned?
I think it could be possible that the heater wouldn‘t query for a device already known as in your case you have a real Vitotrol, so maybe it‘s not sending the 11:00:33:0a:01:01 - f8:04 anymore.
This also would mean I‘m missing something in the communication that the heater expects so its not fully pairing, which explains why I‘m seeing an error message on the heater control panel for the remote control.

boblegal31, 2018-10-03:
I'll have to double check. Maybe my KMBus to Ethernet converting HW/SW is missing some frames. I'll try with another setup.

I incline to believe those frames (11:00:33:0a:01:01 - f8:04:49:ef) are sent periodically anyway, because I did in the past unplug the vitotrol from the bus for a few minutes, plug it back in and it started again working as before. Or maybe there is a timeout longer than 2 or 3 minutes before the vitotronic declares the slave as unavailable.

François

timob0, 2018-10-03:
Hi Francois,

here's a quick update:
- It turns out that the device once paired and responding to pings properly is not queried again
- Once a number of pings are not replied, the heater queries for the device again. After some time it raises an error condition visible on the control panel
-- This explains why you're not seeing the 11:00:33:0a:01:01... frames as longs as your Vitotrol is working properly. You should however see some 11:00:33:0a:02:01... or 11:00:33:0a:03:01... ones (see below capture), since you can connect up to 3  Vitotrols 
- I finally managed to query data from the heater. As it turns out, the record 0x14 is for sending data and switching heater programs only, whereas the record 0x15 is for querying (see below)
- My setup was wrong in terms of timing which became obvious when looking at your captures. The slave has to reply almost immediately for the master to properly recognize. So I did the following:
-- Removed all the delays from my python script as well as the repetition
-- Reduced the buffer depth to one byte on the USB serial converter
-- Raised the process priority from Normal to Realtime (experimenting with High atm)

Heater querying for devices - note the absence of the 11:00:33:0a:01:01 frame:
<10< 11:00:33:0a:02:01 - f8:04:84:ca
<10< 11:00:33:0a:03:01 - f8:04:3f:d6
<10< 20:00:33:0a:ed:01 - f8:04:c0:a0
<10< 20:00:33:0a:02:01 - f8:04:b3:a6
<10< 20:00:33:0a:03:01 - f8:04:08:ba
<10< 04:00:33:0a:01:01 - f8:04:50:23
<10< 01:00:33:0a:02:01 - f8:04:fc:91
<10< 01:00:33:0a:03:01 - f8:04:47:8d
<10< 01:00:33:0a:04:01 - f8:04:66:da
<10< 01:00:33:0a:01:01 - f8:04:31:b4
<10< 20:00:33:0a:ee:01 - f8:04:0d:85
<10< 20:00:33:0a:ec:01 - f8:04:7b:bc
<10< 04:00:33:0a:11:01 - f8:04:f1:e0
<10< 04:00:33:0a:12:01 - f8:04:3c:c5
<10< 04:00:33:0a:13:01 - f8:04:87:d9
<10< 07:00:33:0a:01:01 - f8:04:80:a9
<10< 07:00:33:0a:02:01 - f8:04:4d:8c
<10< 10:00:33:0a:01:01 - f8:04:f6:6e

Here's how to query data - the memory address is the two bytes following 0x15 and they're reflected on the reply frame from the heater. Currently running queries for memories 0x00 - 0xff
>>>> 00:11:bf:0c:01:01:**15**:**ab**:**ab**:4d:75:c2
<40< 11:00:bf:28:01:01 -  11:**ab:ab**:be:aa:aa:aa:aa:aa:7b:aa:aa:aa:aa:aa:aa:aa:51:a8:aa:aa:aa:7b:aa:aa:aa:e1:ab:ab:84:aa:ab:69:f3

-timo

boblegal31, 2018-10-03:
Timo,

I was just thinking while reviewing the data in BF:28:01:01:11
If you're correct, then we might be able to decode the data by reviewing the XML files from vcontrold as the memory addresses might/should be the same.

I had another concern in mind : let's say I have 2 vitotrols in my installation (I don't know if that's even possible), how one vitotrol know it has to reply for internal slot 1 and the other slot 2 ?

François

timob0, 2018-10-03:
Hi Francois,

I started looking at the 40 bytes records too, if we can make use of existing work that would be great and save a lot of effort. Attached is a file with what I have so far including a dump of the 40 byte messages (data40.txt) I got by looping through the address range from 00 - ff. Almost looks like the lower 3 address bits are masked out leading to the data being the same for 8 consecutive lines.

One portion with addresses from 0x18 - 0x48 I'm sure is the time programming for each weekday - would that be reflected in the vcontrold the same way?

Re. the Vitotrol, I think there probably is a way to set the device itself to a specific heating circuit either by hardware switches or maybe in the menu when it's first powered up?

[kmBusAgent.zip](https://github.com/openv/openv/files/2443813/kmBusAgent.zip)

boblegal31, 2018-10-04:
Hi Timo.

So I tried to register my unit as remote control for internal slot 2, which translates for vitotronic as remote control for heating circuit 2. That kind of work, as I can then successfully see the 0x1e messages. I get however an error status on the vitotronic, code BD (wrong configuration of code A0). That makes sense as I don't have any heating circuit 2. So all I get from vitotronic directed to internal slot 2 is the message :
11:00:bf:0b:02:01:1a:55:2d:9b:7d coming every 2 or 3 seconds. 

I tried to poke the vitotronic memory using the 0x15/0xE7 command, but I don't get anything in return. That may be linked to my "faulty" setup.

François

boblegal31, 2018-10-04:
Oh and yes, about any switches or configuration on the vitotrol, I checked the manual and I found none.

timob0, 2018-10-05:
Hi Francois,
when registering with slot 2, does the heater start sending the ping frames 11:00:00:08:02:01:08:88 ?
The BD error code also refers to a bus communication issue, I had this (code BC for circuit 1) with my Vitotrol emulation when the timing of my ping responses wasn't correct. 

I think in case of the secondary Vitotrol, you would need to use different "tabs" in the 0xbf command as per the Wiki (https://github.com/openv/openv/wiki/KM-Bus-Command-0xBF), I guess for reading memory this would be the 0x16 with command 0e7 and probably you'd get a "tab" 0x12 back.
This is somewhat weird in that the heating circuit assignment is supposedly coded in the "slot" as well. Maybe this is for devices like the Vitotrol 300a which can control multiple heating circuits.

The message you've seen for slot 2 (11:00:bf:0b:02:01:1a:55:2d:9b:7d), I can see it too for slot 1 and it refers to tab 0x19 instead of 0x1a which is in line with the above from the Wiki. 
I think it contains an indication of which memory locations changed so that the Vitotrol doesn't have to read all memory all the time. When XORing 55:2d it gives a bit vector b1111 1111 1000 0111
After reading all memory addresses its aa:a8 XOR aaaa = 0x0002 = b0000 0000 0000 0010
Will try to figure out the mapping from bit number to memory address.

On the physical device configuration, there seem to be separate installation and owner manuals for the Viessmann equipment. When I searched for a Vitotrol manual, I found a "Manual for the installer": https://www.3-liter-heizung.de/bilder/mta-vitotrol200.pdf
In there, the bus connection is explained and also there is a mention of dip switches on the devices backside which would control the heating circuit assignment. Maybe this paperwork wasn't handed out to you when the heating was installed.

boblegal31, 2018-10-05:
Hi Timo,

Yes, I do get the PING requests 11:00:00:08:02:01:60:A2 frames, to which I reply with the corresponding PONG frames, and I no longer see the 11:00:33:0a:02:01 frames, so I suppose the registration is OK. I do also get 00:11:bf:0c:02:01:1e which are identical to the 00:11:bf:0c:01:01:1d frames.

About the different tabs, I did not think about trying, because the 00:11:bf:11:02:01:14:aaab62:aa:aa:aa:aa:aa frames were working correctly. I will give a try and report back.

About message in tab 0x19, I've never seen it so far for circuit 1.

I checked again the documents. You're right, there is dip switches to select the heating circuit. I also checked the vitotronic 200 datasheet, and it states you can have ONE vitotrol 200 or 300 remote control on each heating circuit.

boblegal31, 2018-10-05:
Tried sending the E7 command in tab 0x16 works !

boblegal31, 2018-10-05:
However, all I get as a reply, whatever the address I encode in the request is : 
11:00:bf:28:02:01:12:aa:ab:83:aa:aa:aa:aa:aa:dd:90:aa:aa:aa:aa:aa:aa:ec:fa:aa:aa:aa:dd:90:aa:aa:e1:aa:aa:aa:aa:aa:91:38

But again, that might be linked to the non existant heating circuit.

boblegal31, 2018-10-05:
For reference, a capture of all the 0xBF telegrams, with my system registered on internal slot 2
[KM3.zip](https://github.com/openv/openv/files/2451423/KM3.zip)

timob0, 2018-10-05:
Great you go it working anyway and thanks for sharing the capture! I agree, the empty data likely is due to the non-existing heating circuit. In the data you shared earlier, there were lots of the 40 Byte telegrams for circuit 1, so maybe you can use those for monitoring. If controlling the heater with tab 14 works on the slot 2, you'd have both, the control and the logging.

Here's a thought on the tabs 14, 15, etc.: 
Maybe tab 14 is a more general one for controlling the heater as a whole, like shutdown and setting modes not neccessarily bound to one circuit.
The tabs 15, 16 and 17 would then be there to query data from one specific circuit
Likewise, the tabs 20, 21 and 22 would send the room temperature to the circuits 1 - 3

One the 40 byte messages, I think the tab 0x19 / 0x1a encodes the updated since last read memory positions as a 16 bit vector. The bytes appear to be swapped (low byte first) and each bit corresponds to a memory address (or block?) in 0x00,0x08,0x10, …

I think this could help in decoding the data since ideally changes of a temperature for instance would be located at one specific memory address.

Another thing I'd like to figure out is how to control the setpoints for normal and low temperature.

boblegal31, 2018-10-31:
Hi Timo,

So I've got my remote working quite well now, but I decided I had to did this topic a little more.
I disconnected my vitotrol from the base plate and observed by scope the signals. I ended up findind LVCMOS3.3 class UART signals, so I decided to tap the TX line from the vitotrol.

I found this :
- When setting the normal setpoint from 20° to 22°, the vitotrol sends out :
00:11:BF:11:01:01:15:A6:AB:67:BC:AA:AA:92:AA:9B:B3
- When setting the normal setpoint from 22° to 20°, the vitotrol sends out :
00:11:BF:11:01:01:15:A7:AB:67:BE:AA:AA:92:AA:AC:24

- When setting the reduced setpoint from 17° to 15° :
00:11:BF:11:01:01:15:A4:AB:64:A5:AA:AA:92:AA:AD:51
- When setting the reduced setpoint from 15° to 17° :
00:11:BF:11:01:01:15:A5:AB:64:BB:AA:AA:92:AA:EA:05

So the red byte is definitely the setpoint, but I don't really get the blue byte. It might be a compound of an address, with something else.

- When setting the party mode on with 20° setpoint :
00:11:BF:11:01:01:15:A3:AB:61:BB:AA:AA:92:AA:DC:0C
then PONG then
00:11:BF:11:01:01:15:A0:AB:77:BB:AA:AA:92:AA:46:DC
then PONG, PONG, PONG, then
00:11:BF:11:01:01:15:A1:AB:65:BE:AA:AA:92:AA:4B:31

- When setting the party mode off :
00:11:BF:11:01:01:15:A6:AB:66:BE:AA:AA:92:AA:38:A1

- When setting the holiday mode starting 31/10/2018 and ending 01/11/2018 :
00:11:BF:11:01:01:15:A7:AB:73:55:EB:70:F1:AA:FE:A3
then PONG, then
00:11:BF:11:01:01:15:A4:AB:70:AA:E8:70:F1:AA:C4:F5

I've got this message which I suppose is the current temperature :
00:11:BF:0C:01:01:20:78:AA:AA:D2:0A

When canceling holiday mode :
00:11:BF:11:01:01:15:A5:AB:71:AA:E8:70:F1:AA:50:70

I will try to gather more data to explain the blue byte, but this looks like a good starting point.

François

boblegal31, 2018-11-05:
Good to hear it works !

Interesting the fact that you send the setpoint to tab 14 (whereas my real vitotrol sends it to tab 15).

 

Attached is a capture of what the vitotrol sent during one of my trials. The (byte) does not seem to have an incremental variation.

 

François

dumpfheimer, 2020-09-16:
Hi guys,

I'm trying to replicate your work on a ESP32 with a MBUS<->TTL converter.
I managed to receive the discovery message (0x1100330A0101F80449EF) after specifically configuring the heater (Vitocal 200-S) to use the remote (address 2003->1).
The ESP32 immediately answers with 0x0011B3100101F811F938FA00FB053613 but nothing seems to happen.

I am using a HardwareSerial on 1200 baud with 8E1 (have tried multiple configurations).
Changing it to 8E2 does not seem to affect the receiving which makes me wonder if it actually changes anything?

Do you have any idea where I could start searching for errors?

Best,
Chris

timob0, 2020-09-17:
Hi Chris, 
from my memory, there were a couple of pitfalls:
1) There is a checksum and some XORing on the messages - last two bytes I think - which has to be correctly calculated for the replies sent back
2) Timing is critical, the reply has to come back without delay - this might not be a big issue on the ESP. Running on Windows I had to set process priority really high to make it work
3) Serial buffering - again probably only on PC operating systems - needs to be turned down as far as possible to prevent the UART chip from delaying the response

If timing and messages are right, then the discovery process will add your ESP as a remote automatically to the heater, it will show up on the config menu, I think A2 config item then changes from 0 to 2

Once you get it going, its quite straight forward. I'm happy to share my python code if that helps.

I haven't looked into this in a while since I got the Viessmann Wifi Gateway w.t. mobile App, but would be interested in working with you on the ESP front for monitor purposes.

Best,
Timo

dumpfheimer, 2020-09-17:
Thank you for your quick reply!

I'm fairly sure I have the correct checksums and have also tried answering immediately with the prepared byte buffer. I believe the message is correct as I have found the same response in the openv KM Bus Wiki page.

Serial buffering did give me an idea, though. I write the bytes seperately and have a flush after. Maybe a flush after every byte could help.

Also I have ordered another KM bus slave device (USB instead of USB<>TTL). Maybe I will have more success with that (thanks for the serial bus and priority tip).

I have tried letting the heater configuration unchanged but I then never get any discovery messages. Not even after several minutes. Do you have any thoughts on that?

Thanks for your help!

Best
Chris

boblegal31, 2020-09-17:
Hi Chris,

 

From the top of my head, the timing when the answer is sent had an importance.

I’ll have to go back to the source code and check that.

 

Anyway, as Timo replied, once you register your remote controller to the VitoTrol, you should be able to send frames after the ping ones, and you should start receiving the statuses frames

 

François

timob0, 2020-09-17:
Re the discovery messages, different heaters might behave differently and from a quick googling your config seems to be correct (2003:1). So the vitocal might not send discovery messages out of the box.
I assume the wiring of your setup is correct since you're seeing the message from the heater. 
The secret lies in the response message then, either timing or construction of the message. Since the vitocal 200 doesnt support all of the vitotrols, it might be an idea to explore in that direction, i.e the heater might consider the reply as from a unsupported vitotrol and ignore it. We would need the reply of a vitrol 200a to replicate that - @Francois, which one do you have? If its a 200a, can you share a capture?

-timo

dumpfheimer, 2020-09-17:
Hi Timo!

Thank you both for responding so quickly. I really appreciate it.

I think you just might have pointed me in the right direction. It might be that it simply does not support the 200 but only the 200a.. (https://www.heizungsforum.de/threads/passende-vitotrol-fuer-vitocal-200-s.6127/ german)
so many hours.. well.
I'm not sure if that is the difference but I am iterating over the sub-class-id at the moment and hoping the I'd 0005 is valid for the 200a too. If not I might try iterating over the IDs too... Scratch that I just figured out that would take 25 years. I'll see if the subclass does the trick.

Thanks again!

Best
Chris

timob0, 2020-09-17:
Hi Chris,
I think the class and subclass are good since the heater is querying for these (0x11, 0x01, 0x01), so the reply should match. Also I checked against my notes from late 2018 and found the request / response byte sequences are exactly the same as on your initial post, so the reply you constructed should work in principle.

However, when I looked at this Wiki: https://github.com/openv/openv/wiki/KM-Bus
I came across the following table:
>>
Die Vitotronic 200KW2 kennt folgende Klassen und Unterklassen:
Tabelle | Klasse | interner Slot | ID2 | Geräte | CMD1 | CMD2 | CMD3 | Kommentar
1 | 0x11 | 0x01 | 0x34,0x38 | Vitotrol200A/300 | X | X | X |  
2 | 0x11 | 0x02 | 0x34,0x38 |   | X | X | X |  
[...]
10 | 0x20 | 0xEE | 0x81 |   | X | X | X |  
11 | 0x04 | 0x01 | 0x11,0x12 | Funktionserweiterung (?) | X | X |  
<<
and further down:

>>
Die ID
Jedes Gerät hat eine 4 Byte ID.
ID1 | ID2 | ID3 | ID4
Klasse | interner Slot | SerienNummer 1 | SerienNummer 2
<<

Look at line 1 in the first table - it specifies **0x34** and 0x38 as ID byte 2 for **Vitotrol 200a** and 300 respectively. Which means that (if the table is correct) your response bytestream identifies istself as a Vitotrol 300, which as per the forum post you found doesn't work with the vitocal 200:
0x00 11 B3 10 01 01 F8 11 F9 **38** FA 00 FB 05 36 13 

You could try to replace the 10th response byte 0x38 with 0x34 and retry the communication. I'm not sure if ID bytes 3 and 4 are checked somehow, i.e. if you require a "valid" 200a serialnumber - don't think so. 

If that's successful, the heater would reply with a 9-byte message such as: 11:00:31:09:01:01:00:ee:4f
To which my code just replies:                        00:11:b1:0a:01:01:00:01:03:f7
After that, the communication is established.

If it's still not working, I'd revisit the wiring for the TX line and maybe make a measurement on both the ESPs TX line and the KMBUS wire to see if anything changes when you send data to the bus. After all, it could also be a faulty transceiver.

Hope this helps - let us know how it's going.

Cheers,
timo

dumpfheimer, 2020-09-17:
YES! It's alive! 0x34 is the way to go.

The heater is now sending 0x00, 0x3F and 0x0BF messages.
Also, I was able to send my room temperature and it triggered the cooling, so I guess that is working too.

I will keep you posted on further progress. I have built a "Smart Home Heating App" that controls the valves of the heating circuits (not quite sure of the correct term here; i mean the rooms / pipes that are controlled by the electric valves) using ESP8266 and automatically sets the heater into heating/cooling/warmwater mode. 

My intention of using the KM Bus is purely to pass a (fake-isch) room temperature to the heater to have a better cooling experience (but I am guessing the heating will work better too).

Maybe I can replace the optolink part with the pure km bus module to make it a bit simpler.

Thanks again, you were a big help!

Best,
Chris

timob0, 2020-09-18:
Wow, glad you could make it work 👍 Also your project sounds interesting. Actually I was thinking to use an ESP as a web gateway which would expose a REST Api that should be much simpler to integrate with things like Homeassistant and other automation devices than the pure KMBUS or the closed Viessmann Wifi Gateway.
As I dont have done anything with the ESP yet (but with arduinos) would you be open to sharing your code as a starting point? Maybe I could try to add the webservice part to it then.

Cheers 
timo

dumpfheimer, 2020-09-18:
Hi!

I am definitely willing to make it open source, but up until now everything is in a state in which I do not feel comfortable making things public - in respect to code quality mainly - just yet.

Please do not understand this as a promise of beautiful code later ;-)

I guess in a few days or weeks I will have the ESP/Vitotrol so far. For the actual heater control software and the ESP satellites that control the valves I believe I will need much more time. I do not see much sense in publishing it before that as it is very much modelled to fit my needs and has multiple hard coded things (eg addresses).

Best
Chris

dumpfheimer, 2020-09-18:
Hi!

Did any of you manage to put your finger on how to write the setpoint?
I tried what you wrote back in October 21st 2018:
00:11:BF:11:01:01:15:(A6):AB:67:[BC]:AA:AA:92:AA:9B:B3
Did you ever find out what A6 means? or AAAA92AA?
For me it had the strange effect that it started cooling after writing that (it did however change the temperature correctly).
This might actually be correct but I still wanted to check in with you just to be on the safe side and not mess up any internal things.

Best
Chris

boblegal31, 2020-09-18:
Hi,

 

I don’t remember if I succeeded or not in setting the setpoint.

It’s been quite some time since I last looked at that. I’m sorry.

 

François

timob0, 2020-09-18:
Hi Chris,

I understand you don´t want to share your code just now, that´s fine.

Re the setpoint thing, please see the code snippet below. One thing to keep in mind is that for whatever reasons the data following a command has to be XORed with AA, so the 8 bytes following 0x15 have that applied.
I haven´t figured out what the A6 byte means. Setting the value to 0 (aka AA) does work as well.
- 0xAB translates to 1 which I believe is the heating circuit to control.
- 0x67 translates to 0xCD and sets the "normal" temperature whereas 0xCE would set the reduced temp
- 0xBC translates to 22 and is just the setpoint temp as an integer number, i.e. no fractions possible
I found that any bytes sent after that are just garbage and ignored, so it´s safe to leave them out which is why the respective lines below are commented out.

Below is a code snippet from the stuff I did for the Vitocrossal heater I have, yours might be different in nuances. 
I´m also attaching the full script (without warranty) since there is a lot more you can read out from the heater memory such as the heating schedule etc. 

Cheers
timo

# Construct reply buffer - to send commands to the heating by using tab 0x14
						# databyte 3 contains the command:
						# 0xC8 Abschalten
						# 0xC9 Warm Wasser
						# 0xCA Heizung und Warm Wasser
						# 0xCB Partybetrieb An
						# 0xCC Partybetrieb Aus
						# 0xDC Sparbetrieb An
						# 0xDD Sparbetrieb Aus
						# if state change requested
						if (control_state_request!=heater_state_unset):
							print("New operation mode requested ", file=log, flush=True)
							print(control_state_request, file=log, flush=True)
							print("from state ", file=log, flush=True)
							print(control_state_active, file=log, flush=True)							
							tmp_state = 0xCA
							tmp_data = 0x00
							if (control_state_request==heater_state_off):
								tmp_state = 0xC8
							elif (control_state_request==heater_state_water):
								tmp_state = 0xC9
							elif (control_state_request==heater_state_water_heater):
								tmp_state = 0xCA
							elif (control_state_request==heater_state_party):
								tmp_state = 0xCB
							elif (control_state_request==heater_state_save):
								tmp_state = 0xDC
							elif (control_state_request==heater_state_setpoint_hi):
								tmp_state = 0xCD
								tmp_data = set_temp_normal & 0xff
							elif (control_state_request==heater_state_setpoint_lo):
								tmp_state = 0xCE
								tmp_data = set_temp_reduced & 0xff
							# reset state request
							control_state_request = heater_state_unset
							reply = bytearray(0)
							reply.append(0x00) # dest class
							reply.append(0x11) # source class						
							reply.append(0xbf) # query master
							reply.append(0x00) # lenght byte
							reply.append(0x01) # destination slot
							reply.append(0x01) # source subclass
							reply.append(0x14) # select table 0x14
							reply.append(0x00 ^ 0xAA) # data 1 - memory?
							reply.append(0x01 ^ 0xAA) # data 2 - heating circuit
							reply.append(tmp_state ^ 0xAA) # data 3 - turn on				
							reply.append(tmp_data ^ 0xAA) # data 4
							# reply.append(0x00 ^ 0xAA) # data 5
							# reply.append(0x00 ^ 0xAA) # data 6
							# reply.append(0x00 ^ 0xAA) # data 7
							# reply.append(0x00 ^ 0xAA) # data 8
							reply[3] = len(reply) + 2
							cs = crc16( reply )
							reply.extend( cs.to_bytes(2, 'little') )

[kmBusAgent.zip](https://github.com/openv/openv/files/5248465/kmBusAgent.zip)

timob0, 2020-09-22:
Hi Chris,

the goal was to integrate the heating with Homeassistant so that I could set operating modes and target temperature from around the house using a browser and monitoring temperatures / states. Also I have a homematic room temp sensor from which I fed data into the heating emulating the vitotrol room sensor which didn't turn out too useful.
What I really was after but could not find was the heater exhaust temperature to calculate the efficiency and with that continously optimize the setup. 
Also, I'm not too happy with the script running on Windows server due to timing issues when the machine is running under load. In that case the timing gets too far off and I'm getting a error message on the heater indicating failure of the remote control.
That's why I'd be keen on porting the script to the ESP which wouldn't suffer from the timing problems. Add a REST Api and it would be a perfect interface to any home automation solution.

Cheers
timo

dumpfheimer, 2020-09-23:
Got it, thanks.

This definitely makes the read functions mandatory.

I am trying to figure out what the best behavior for read/write differences should be. Right now I let the ESP go silent (and cause an error on the heater) if it does not receive any commands over wifi for i think 5 minutes. I did this to prevent sending wrong room temperatures for a longer period and maybe cause overheating or "undercooling". My belief was as soon as the remote disappears, the heater will ignore the current room temperature.

Do you have knowledge of any internal timer in the heater?
To elaborate: If I send the current room temperature 16 Celsius and then nothing for an hour, will the heater still think the room temperature is 16 Celsius?

If not I will stick to my current going-silent-method. 
But then I have the problem what to do if it is used as read-only. It's not possible to read the values without triggering the Error on the heater.

Possible solutions:
- go silent (and cause an error on the heater) if no current room temperature was received for X minutes
- write back the current desired room temperature as current room temperature if no current room temperature was received for X minutes
- decision must be made at compile time (could cause problems when switching)

Any ideas/thoughts?

Best,
Chris

PS
I keep saying desired room temperature, with that I mean the setpoint.

boblegal31, 2020-09-23:
I think, to avoid the error on the heater, you just have to replay to PING packets.

If you don’t generate any more traffic than that, it should be OK.

 

François

dumpfheimer, 2020-09-23:
Hi!

Sorry I believe there is a misunderstanding. I currently purposely let an error occur to be sure the heater is not using an old, possibly wrong, current room temperature. (The ESP does not measure any temperature itself but relies on another system, that could fail)

If there is another way to tell the heater there is NO current room temperature that would be ideal. 

Best
Chris

timob0, 2020-09-23:
Hi,

I‘m not sure how the heater handles the room temp in case it‘s not getting periodic updates. My guess is that it would treat the Vitrotrol (which we‘re emulating) as failed and falling back to the regular heating program based on the outside temp. Would need to check with Viessmann to be sure though.

I think there is a config item on the heater to explicitly switch between outside temp and room temp regulation which can be used to ignore the Vitotrols room sensor. If we manage to control that remotely, then you could use this as a fallback in case you don‘t get temp readings. Maybe the Vitotrol manual has clues on this, I‘ll check my heater manual later for the config item.

Another thought is separating the room temp and the remote control feature:
1) you could try to emulate 2 Vitotrols, one would send the room temp and commands and the other would be for monitoring. I believe @Francois tried that out with the real and the emulated Vitotrol connected same time. This would depend on the heater to support multiple remote controls which varies by model.

2) I think there is a separate temp sensor available from Viessmann in case the one built into the Vitotrol can’t be used (Scenario: You have a dedicated reference room for the temp but the Vitotrol is in a more convenient location)
Emulating that device as well could be another option, however we‘d need a real one to analyse it.

Cheers
timo

dumpfheimer, 2020-09-24:
@timob0 
I have an optolink -> wifi translator for ESP running already. I am glad to share the code with you if you are still trying to read the exhaust temperatures. I am sure it is available over Optolink.

timob0, 2020-09-27:
@dumpfheimer  that would be really great if you could share the code. 
Many thanks in advance :-)

PS: I looked up the config item re. the room temp sensor, this seems to be item b0 with a factory default of "0" which means it will ignore the room temperature. This is for the Vitocrossal heater I have, yours could be different.

dumpfheimer, 2020-10-08:
Hi!

Sorry for the long delay. I have finally published my Optolink proxy project here: https://github.com/dumpfheimer/OptoProxy
Please note that I did not have time to clean it up. If you have problems compiling it feel free to contact me.

My heater does not have a b0 address (all of them are 4 digit hex). But I was able to find the setting. Thank you anyway

dumpfheimer, 2020-10-08:
Hi!

I was able to get the vitotrol project to an acceptable level to be published. If you want to take a look: https://github.com/dumpfheimer/WiFiVitotrol

Best
Chris

boblegal31, 2022-03-17:
If anybody is interested in this topic, I had to dig a little deeper into this as I modified my heating installation by adding another heating circuit. This seem to change the way I have to interact over KM bus a lot.
I have some captures of data received ant sent by my vitotrol 300 when acting as sensor/remote control for both heating circuits.
I can share those here if anybody is interested.

François

boblegal31, 2022-03-29:
I don't know if anybody is still interested in this topic, but I found out how to set the current time. For the date, it is still not clear. I'll check that later.

So for the records, the time is set with the 0xD4 command to group1 (0x14 - 0x17)
The telegram is 17 bytes long and has the following format :
00:11:bf:11:01:01:14:aa:ab:7e:[date lsb]:[date msb]:aa:aa:aa:CRC:CRC

So the date is calculated in seconds on a 16 bits value + 0x4c00. The overflowing bits are discarded. This value is then, of course, XORed with 0xAAAA, the the 8 lsb are inserted after the command, followed by the 8 msb.

For example, 8h30 would be (30 * 60) + (8 * 3600) = 30600 -> 0x7788
0x7788 + 0x4c00 = 0xC388
0xC388 ^ 0xAAAA = 0x6922

The telegram would be 
00:11:bf:11:01:01:14:aa:ab:7e:22:69:aa:aa:aa:CRC:CRC

Now for 20h45 : (45 * 60) + (20 * 3600) = 74700 -> 0x123cc
0x123cc + 0x4c00 = 0x16fcc
We now discard the MSB :
0x6fcc ^ 0xaaaa = 0xC566
The telegram would be 
00:11:bf:11:01:01:14:aa:ab:7e:66:c5:aa:aa:aa:CRC:CRC

François

mvimercati, 2022-09-15:
Hi guys, congrats for your work!
Do you think is it possible to build an interface that emulates a viessmann extension? e.g.: EA1: mine is not working very well, the replacement is expensive and maybe I cannot find it easily.
I found some TSS721A to uart/ttl converter and I would like to understand if it is easy enough to understand the protocol and simulate the missing device.

dumpfheimer, 2022-09-15:
It's most likely possible. Just a matter of effort. If you have a (kind of) working extension you could try to sniff the traffic so you have na idea how the devices communicate.

What are extensions for, actually?

mvimercati, 2022-09-15:
EA1 connected to Viessmann 242-F, and managing the thermostat contact input. Actually the broken extension gives problems to the whole bus and when connected I get random errors also from other extensions (SM1, managing the solar module) on the KM. I'm not sure I can sniff something good from that.
I'm going to try to replace the TSS721A, hoping that will fix the problem... and eventually I can replace the EA1 with a new one.
Having an alternatve solution would be good, even considering that those parts may disappear in the future....
So the extensions are slaves and the boiler is the master? Do you think I can connect on the bus in parallel with the extension and sniff the traffic from both side?

[10 more replies]