openv/openv #388: Adressen für Verbrauchsabfrage

Hardy74, 2018-01-16:
Hallo,

da die App von Viessmann ja die Gas- und Stromverbräuche der Heizung anzeigen kann, müßte das via Kommando doch auch mit openv möglich sein!? Quasi Befehle wie getGasVerbrauch(vonDatum, bisDatum), oder getGasVerbrauch(monat), selbiges für Strom.

Hat da jemand Erfahrung oder gar schon passende Befehle/Adressen parat? Idealer Weise für eine VScotHO1?

Morg42, 2018-02-12:
Hi,
hab ich bei meiner WP (WO1C / 204d) noch nicht gefunden. Da ich elektronische Stromzähler habe, mache ich das über die (ist ja getrennt nach Haushalt und Heizung). Aber das ist sicher keine Lösung für jede Installation...

MrIcemanLE, 2018-02-13:
Hallo Morg42,
ich habe die gleiche Steuerung und lese aktuell auf folgenden Adressen aus:
`<command name="Jahresarbeitszahl" protocmd="getaddr">
	<addr>1680</addr>
	<len>2</len>
	<unit>CO</unit>
	<description>Statistik Energiebilanz Wertebereich 0 - 10</description>
</command>

<command name="JahresarbeitszahlHeizen" protocmd="getaddr">
	<addr>1681</addr>
	<len>2</len>
	<unit>CO</unit>
	<description>Statistik Energiebilanz Wertebereich 0 - 10</description>
</command>

<command name="COPWW-Betrieb" protocmd="getaddr">
	<addr>1682</addr>
	<len>2</len>
	<unit>CO</unit>
	<description>Statistik Energiebilanz Wertebereich 0 - 10</description>
</command>

<command name="COPHeizbetrieb" protocmd="getaddr">
	<addr>1690</addr>
	<len>2</len>
	<unit>CO</unit>
	<description>Statistik Energiebilanz Wertebereich 0 - 10</description>
</command>`

Allerdings liefert der COPHeizbetrieb (0x1690) anscheinend einen Momentanwert und COPWW-Betrieb (0x1682) einen über sehr lange Zeit konstanten Wert. Bei den anderen Adressen bekomme ich nur sehr hohe Werte die für mich unplausibel sind.

[image]
[image]

dima-357, 2020-11-21:
Please tell me the request address V100
boiler pump in operation or not /
It is not in the list of addresses of Viessmann-adresses