openv/openv #390: OpenV für Vitoligno 300-C VL3C 32kw

aukator, 2018-01-17:
Hallo, ich versuche gerade meine Heizung in Openhab2 einzubinden. Lese schon eine Weile (noch im openvwiki) aber hab noch keinen Startpunkt gefunden. 
Hier ein paar Fragen, bei dem ich Hilfe brauche:
wird meine Vitoligno 300 C unterstützt?
kann ich den USb Optolink Adapter nutzen, der bei meiner Heizung an die Vitoconnect100 kommt?
Einen Raspberry hab ich noch über. Nutze schon einen für Openhab2...
Ich würd gern den zweiten Raspi einfach mal beschreiben, den Adapter des Vitoconnect100 anschliessen und es ausprobieren. Hab nur Respekt davor etwas kaputt zu machen...
Vielen Dank im vorraus...

luismauriz1, 2018-03-05:
Hallo, ich habe eine Vitoligno 300P. Grundsätzlich wird jedes Gerät unterstützt, wenn du ein unterstütztes Protokoll hast. In deinem Fall meine ich P300. Du musst lediglich deine vito.xml und vcontrold XML konfigurieren. Dabei bin ich dir gerne behilflich und kann dir auch meine zukommen lassen. Solange du nur Daten liest, passiert erst einmal nichts. Gerne stehe ich dir für Fragen zur Verfügung
LG
Luis

luismauriz1, 2018-03-05:
Den Adapter von der Vitoconnect 100 habe ich auch, funktioniert prima!
LG

aukator, 2018-03-05:
@louismauriz1
Vielen Dank für die Info. Das hilft mir sehr. Ich werde es jetzt einfach mal versuchen. Und dann werd ich mich sicher bei dir melden;) 
Wenn ich es zum laufen bekomme, stelle ich meinen Lernprozess hier gerne zur Verfügung. Grüße

stheine, 2018-11-26:
> Hallo, ich habe eine Vitoligno 300P. Grundsätzlich wird jedes Gerät unterstützt, wenn du ein unterstütztes Protokoll hast. In deinem Fall meine ich P300. Du musst lediglich deine vito.xml und [...]

Hallo,

ich habe auch den Vitoligno 300P mit dem vcontrold laufen. Ebenso momentan nur als Logger.

Eigentlich würde ich gerne die Betriebsart und die Raumtemperatur setzen können, habe mich aber noch nicht getraut irgendwelche Werte zu schreiben.

Hat da schon jemand Erfahrungen?

Gruss
  Stefan

hydrax0, 2020-04-04:
Hi, ich bin neu hier.
Ich habe eine Vitoligno 300-c und hätte gerne etwas Unterstützung bei den "vconrol.xml" und "vito.xml. 
Wenn jamand etwas für die Vitoligno 300-P hat, dann wäre ich natürlich auch  dankbar dafür.

luismauriz1, 2020-04-04:
Hi, ich besitze eine Vitoligno 300-P und habe in der Vergangenheit auch mit openv nativ gearbeitet. 
Wie möchtest du die Wete auslesen bzw wie möchtest du sie dir anzeigen lassen?

Da ich OpenHab verwende und zu dem Zeitpunkt der Einrichtung die Openhab Kopplung von diesem Forum noch nicht existiert hat habe ich das ganze über ein anderes Programm gelöst. 
Ich schaue mal, ob ich meine alten Vito.xml finden kann

aukator, 2020-04-04:
Hallo luismauriz1, 
Wie hast du es denn gelöst? Ich nutze auch openHAB und bin da interessiert. Wollte es mit der vito XML probieren. Kam aber noch nicht dazu. Lg und danke für deine Hilfe.

luismauriz1, 2020-04-04:
Hi,
Schau hier in diesem Forum vorbei:
https://github.com/steand/optolink
dieses System ist schon fertig konfiguriert. Du musst lediglich die Start Datei anpassen und automatisch starten lassen. Dann funktioniert das ohne Probleme. Direkt in Openhab

Das ist das Binding:
https://www.openhab.org/addons/bindings/vitotronic/

hydrax0, 2020-04-04:
Hallo und vielen Dank für die rasche Antwort!
Ich habe auch eine KNX Installation und habe openHAB auch schon mal am laufen gehabt. Bis mir mein Raspi abgeraucht ist (bzw. die SD-Karte) und ich keien Backup hatte und es noch nicht neu aufgesetzt habe.
Danke für die Hinweise, ich werde mich dort mal umsehen, klingt sehr vielversprechend!!!

aukator, 2020-04-04:
Dar ich dich vielleicht noch bitten, mir eine Anleitung zu geben. Bin recht neu, was raspberrys angeht. 
Und Check das nur halb. Natürlich nur wenn du Zeit und Lust drauf hast. 
Könnte dann auch anbieten, das mit ner Fotostrecke hier online zu stellen. Lg

luismauriz1, 2020-04-14:
Hi, bitte entschuldige meine späte Antwort. Schreib in dem Forum von Steand einen Issue Eintrag. Dort kann ich dir sehr gerne helfen :)
Wenn ich das in diesem Forum schreibe, könnte das für Verwirrung sorgen, da die Schnittstellen sich doch ziemlich unterscheiden.
LG

hydrax0, 2020-04-16:
@aukator : es gibt eine sehr detailierte Anleitung, wie du den "Daemon" vcondtrold am Raspberry installieren kannst. https://github.com/openv/openv/wiki/vcontrold-mit-Raspberry-Pi eine andere Quelle wäre "https://projects.webvoss.de/2017/11/05/interfacing-vitovalor-300-p-with-a-raspberry-pi/". 
Habe diese Anleitungen selber mal ausprobiert und sie funktionieren einwandfrei. Beim Adapter hab ich erst mal ein "Steckbrett" verwendet und mir die Transistoren erst mal gespart, weil die Dioden auch direkt angesteuert werden können, wenn der Raspberry nicht weit weg von der Heizung ist und man kein langes Kabel braucht. Ich habe die Widerstände angepasst, damit nicht mehr als max 16mA vom Raspberry gezogen werden...

hydrax0, 2020-04-16:
@luismauriz1 Ich habe im Forum von Steand einen Eintrag eröffnet. [https://github.com/steand/optolink/issues/19](url). Hoffe ich bekomme was raus. 
Danke im Voraus

DocDirac, 2021-02-24:
Hallo zusammen, 
da unter diesem Issue in der Vergangenheit verschiedene Fragen behandelt wurden hoffe ich, das es ok ist, wenn ich mich hier anhänge.
Ich betreibe eine Vitoligno 300-P mit Solarthermie zur Heizkreisunterstützung. Ich nutze vcontrold zum Loggen verschiedener Anlagenparameter. Bis jetzt aber nur solche für die ich die Adressen im Netz finden konnte. Was mir fehlt ist die Kollektortemperatur. Hat da jemand schon etwas ermittelt? Ich kann mir vorstellen, das man sie unter der 0x3900 findet, da unter dieser Adresse die Temperatur für den Vorlaufsensor des Zweiten Mischerkreises ausgelesen werden könnte, wenn man statt Heizkreis/Solarthermie die Konfiguration Heizkreis/Heizkreis fahren würde.
Ich bin mir aber unsicher, ob ich es einfach auf einen Versuch ankommen lassen kann, indem ich die vito.xml und mein Loggerscript anpasse:

`    <command name="getTempKoll" protocmd="getaddr">
      <len>2</len>
      <unit>UT</unit>
      <description>Ermittle die Kollektortemperatur in Grad C</description>
      <addr>3900</addr>
		<unit>UT</unit>
        <len>2</len>
    </command>`

Wenn ich nur lesend zugreife sollte doch eigentlich nichts passieren, oder?
Danke

stheine, 2021-02-24:
> Wenn ich nur lesend zugreife sollte doch eigentlich nichts passieren, oder?

@DocDirac, sehe ich auch so. Lesen habe ich auch mal einiges durchprobiert. Den 300-P habe ich auch, allerdings ohne Solar. Ich logge seit mehreren Jahren die wichtigen Parameter.
Was ich gerne machen wuerde, mich allerdings nocht nicht getraut habe, ist das Wechseln des Betriebsmodus Heizen und Warmwasser zu nur Warmwasser und umgekehrt. Wenn dazu also jemand die richtigen Parameter kennt...