openv/openv #391: Vitoconnect und Raspberry Pi gemeinsam an Vitodens 300

Aschhoff, 2018-01-20:
**Eine einfache Möglichkeit die Optolink Schnittstelle gemeinsam zu nutzen.**

Ich habe eine kleine Platine mit zwei Relais und einem Transistor gebastelt, damit schalte ich die vier Leitungen des Optolink einmal auf das Vitoconnect und alle 15 Minuten kurz auf den Raspberry.
Im Cronjob steuer ich die Ports mit Scriptbefehlen und somit die Relais an, mache die Datenabfrage und schalte wieder zurück. 

Datenbank und Visualisierung funktioniert seit Monaten prima.
Im Anlagentagebuch der App VitoCare hab ich pro Tag ein paar Einträge "Verbindung zur Heizung verloren"
Ich denke das ist ganz ok.

powerzumsel, 2018-02-02:
HiHi,
genau sowas schwebt mir auch vor.
Kannst du mehr Infos geben?

Aschhoff, 2018-02-02:
wenn denn Interresse besteht, mach ich mal ne kleine Zeichnung und poste das Script.

powerzumsel, 2018-02-02:
interessiert mich sehr.
Finde es schade, dass Viessmann die Daten für sich behält und man nur über die App rankommt.
Ne offene REST API im OPTO1 wär schön.
Neben der Relaisumschaltung, hatte ich an eine Modifizierung der EigenbauUSB Schnittstelle gedacht, um die optischen Signale zum einen ans Optolink UND an die EigenbauUSB Schnittstelle zu leiten

Aschhoff, 2018-02-02:
Dachte ich auch erst dran, aber dann kollidieren eventuell die Sendedaten. Harte Umschaltung scheint da besser.
So Schaltplan, Script und Detail hinzugefügt.

Aschhoff, 2018-02-17:
Die Idee ist gut, aber man müsste die Daten vor dem seriell USB Wandler abgreifen also TTL. Das Datenprotokoll ist ja bekannt. Ob openv einfach mitlesen kann???
In meiner Anwendung frage ich Vorlauf ist und soll Brennermodulation usw alle 15 min ab. Die Daten werden Vissmann wohl nicht interessieren.
Zur Schaltung: laut Foren soll bei USB erst die + - dann erst die Daten geschaltet werden! Schau mal in eine USB Buchse, dann siehst du, dass die äusseren Kontakte etwas länger sind also + - zuerst verbunden werden.

warmup72, 2018-02-17:
Wenn die Möglichkeit eines Mitschnitts am USB-Port besteht, wäre es ggf. auch eine Möglichkeit gezielt daten über die VitoSoft abzufragen. Es gab ja im Netz mal eine 4 Wochen gültige Installation. Diese sollte sich doch auf einer VMware zum Leben erwecken lassen.

Jojo-A, 2018-02-20:
@Aschhoff genau so hatte ich das gemeint. Die Daten müßten TTL-seitig mitgeschnitten werden, nicht USB-seitig. Ich habe gerade mal mein mitgeliefertes OPTOLINK-Kabel aufgeknackt. Mit etwas Löt-Geschick müßte man da eigentlich dran kommen. Ich bräuchte also nurnoch ein Tool, was die Daten anhand eines definierten Protokolls entschlüsselt.
Zur Schaltung: jo, die Versorgungsleitungen bekommen beim einstöpseln zuerst Kontakt. Das hat aber auch den Grund, daß der Controller im USB-Geräte (z.B. so ein FTDI-Chip) erstmal auf einem definierten Potential liegen sollte, bevor man ihn mit Signalen füttert. Ich bin relativ sicher, daß ein Umschalten den Signalleitungen reicht, solange beide Hosts das gleiche GND-Potential haben. Einfach mal ausprobieren :) .

@warmup72 also ich kenne keine Möglichkeit, um "am USB" Daten mitzuschneiden (außer natürlich mit Logic-Analyzer, etc...). Aber das ist m.E.n. auch nicht nötig, da die OPTOLINK-Schnittstelle eine Serielle ist (also eigentlich so erstmal nix mit USB zu tun hat).

Grüße

Aschhoff, 2018-02-21:
nimm einen TTL USB Wandler, klemm dich parallel auf die Sendeleitung des Optolink.
Ich denke mit vclient und vcontrold (ist ja eine Art Terminalprog für die Heizung) sollte man mitlesen können.

Jojo-A, 2018-02-21:
@Aschhoff jo, so einen Wandler hab ich bereits bestellt :) . Ist nur leider noch auf dem Weg... Und dann werd ich mal mein Glück mit dem vclient versuchen. Hab nur leider noch keine Gehversuche damit unternommen, darum werd ich bestimmt Schwierigkeiten bekommen. Mal sehen...

Jojo-A, 2018-03-27:
Ich habe nun soweit erstmal alles zusammen, das heiß ich habe den Wandler und habe auch schon Abgriffe auf das Platinchen in dem Original-OPTOLINK-Kabel gelötet. In einem normalen Terminal-Programm (hterm) kann ich auch schon sehen, daß haufenweise Datenpakete ankommen. So weit, so gut.
Leider hab ich keine Vorstellung, wie das mit vcontrold und ggf vclient nun funktionieren soll. Ich verstehe die Anleitung einfach nicht. Könnte mir jemand erklären, wie ich jetzt, wo ich den rohen Vitoconnect-Traffic mitlesen kann, das ganze per vcontrold entschlüsselt kriege?
Grüße

ang2l, 2018-09-06:
Vielen Dank erstmal für eure Bemühungen!

Wir sind gerade am Neubau eines EFH und haben als Heizung eine Sole-Wasser-Wärmepumpen von Viessmann (Vitocal 300-G) vorgesehen.

Ich würde gerne mittels einem Raspberry und der Software Volkszähler den Temperaturverlauf der Heizung bzw. der Heizkreisläufe überwachen.

Da die Heizung einen Vitoconnect 100 mitgeliefert bekommt und ich gerne die 5-Jahre-Garantie mitnehmen würde, finde ich eure Lösungsansätze klasse!

Da ich elektrotechnisch nicht ganz so fit bin, hätte ich die Bitte, dass du Ashhoff den Teil unter dem Text „Kleinrelais“ nochmal kurz wiederholst. Ich kann es leider nicht genau lesen. Ein Foto von deiner fertigen Platine wäre auch super!

Danke dir!

Grüße
manu

Jojo-A, 2018-09-07:
@ang2l Hi, nee mit dem Auslesen bin ich noch nicht weiter gekommen. Irgendwie fallen in einem Haus immer ein Haufen anderer Aufgaben an, die höhere Priorität und WAFs haben ;) .
Ich habe jetzt mal vcontrol kompiliert und installiert. Jetzt muß ich mal gucken, wie das mit dem vclient funktioniert. WENN ich was zustande bringe, werde ich es aber berichten.

challo2018, 2022-12-08:
Ist zwar schon ein sehr altes Thema. aber habt ihr hier inzwischen eine Lösung?

challo2018, 2022-12-09:
Grundsätzlich bin ich bei dir, dass man auch eine Lösung verwenden kann. Ich verwende aktuell die API. Allerdings sind hier einige funktionen nicht steuerbar, ich hoffe hier mit vcontrold mehr möglichkeiten zu haben. Zustzlich kosten neue Datenpunkte in der Viessmann API monatlich geld und nicht gerade wenig. 
Wenn man Vitoconnect verbunden hat gibt es auf die Heizung eine Zusatzgarntie. 
Daher bin ich aktuell am überlegen auf eine Lösung umzubauen die beides bedienen kann und mehr Datenpunkte liefert.