openv/openv #392: Benötige Hilfe beim Raspi vcontrol Startscript

mikeww1, 2018-02-02:
Ich bin neu hier und habe heute mein Viessmann Verbindungskabel erhalten.
Als Zielgerät verwende ich einen Raspberry Zero W auf dem Openhabianpi installiert ist.
Entspricht Jessie soweit ich das verstehe. Dort läuft openhab2.
Ich konnte der Anleitung problemlos folgen bis zum Pkt6
Ein manueller Start von vcontrold funktioniert. 
[18:11:43] openhabian@openHABianPi:/etc/udev/rules.d$ vcontrold -n
[6113] Fri Feb  2 18:12:58 2018 : TCP socket 3002 geoeffnet
^C

Dann habe ich das shell vcontrol (ohne Endung) mit nano erstellt und das Beispielshell reinkopiert. Den Befehl: sudo chmod u+x vcontrol habe ich ausgeführt.
Jetzt wollte ich testen aber erhalte nur die folgende Fehlermeldung:
**[18:42:21] openhabian@openHABianPi:/etc/init.d$ sudo sh vcontrol
Usage: /etc/init.d/vcontrold {start|stop|status|restart|force-reload}
[18:42:37] openhabian@openHABianPi:/etc/init.d$ sudo sh vcontrol start
[....] Starting vcontrol (via systemctl): vcontrol.serviceFailed to start vcontrol.service: Unit vcontrol.service failed to load: No such file or directory.**

Leider sind meine Linux Kenntnisse sehr beschränkt und ich kann das Script kaum lesen.
Jetzt habe ich zur Sicherheit auch noch das nächste Kommando eingegeben:
sudo update-rc.d vcontrol start 99 2 3 4 5 . stop 99 0 1 6 .

Das wird aber mit folgendem quittiert:

**update-rc.d: warning: start and stop actions are no longer supported; falling back to defaults
insserv: script vcontrol.sh: service vcontrold already provided!**

Was immer das bedeuten mag ? Ich bitte um Hilfe bei den finalen Schritten.

mpseybold, 2018-03-03:
Hallo Mike,
ich bevorzuge das neure `systemd` system als die betagten und unübersichtlichen init scripte.
> ... [Service]
Type=forking
ExecStart=/usr/local/sbin/vcontrold -x /etc/vcontrold/vcontrold.xml
...

Für die restlichen Einstellungen so eines `bla.service` units die du in deinem spezielle Fall brauchst schaust du am besten in die `systemd` Doku.

ManuelK83, 2022-02-27:
Hallo,
kann einer eine kurze Anleitung schreiben wie man das script nun startet.  
Also quasi  systemd. Ich weiss nicht wo ich die oben genannten Parameter eintragen soll.
Der usb Port wäre bei mir auch ttyusb0

Hoffe mir kann einer helfen.

schaedeldx, 2022-09-26:
Hatte das gleiche Problem, im Bereich "after" funktioniert "systemd-udev-settle.service" nicht, habe es nun rausgelöscht, dann startet der Dienst. Habe es nicht überprüft, aber das exisitert vermutlich nicht.
Zu beachten ist auch, dass der korrekte Pfad zur vcontrold angegeben wird. Ich habe es selbst kompiliert und dadurch ist es im Home-Verzeichnis...
Das USB-Device gebe ich in der Regel über die ID an: /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0
Dadurch benötige ich keine Regelanpassung oder muss die Nummerierung berücksichtigen.

Hoizi007, 2023-11-09:
> Hatte das gleiche Problem, im Bereich "after" funktioniert "systemd-udev-settle.service" nicht, habe es nun rausgelöscht, dann startet der Dienst. Habe es nicht überprüft, aber das exisitert vermutlich nicht. Zu beachten ist auch, dass der korrekte Pfad zur vcontrold angegeben wird. Ich habe es selbst kompiliert und dadurch ist es im Home-Verzeichnis... Das USB-Device gebe ich in der Regel über die ID an: /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0 Dadurch benötige ich keine Regelanpassung oder muss die Nummerierung berücksichtigen.
> 
> Aktuell startet es so: [Unit] Description=vcontrold daemon After=syslog.target
> 
> [Service] Type=forking ExecStartPre= ExecStart=/home/pi/openv/vcontrold/build/vcontrold -x /etc/vcontrold/vcontrold.xml ExecReload=/bin/kill -HUP $MAINPID Restart=always RestartSec=120 StandardOutput=null
> 
> [Install] WantedBy=multi-user.target

bekomme hier leider die Meldung 

Nov 09 05:35:50 raspberrypi systemd[1]: /etc/systemd/system/vcontrold.service:1: Assignment outside of section. Ignoring.

was kann ich da noch versuchen?

lg.

J-A-U, 2023-11-09:
Fehlt in der Datei das erste `[Unit]` oder sind da (Leer-) Zeichen mit in der Zeile die da nicht hingehören?

Hoizi007, 2023-11-09:
die leerzeichen bezogen auf die vcontrold.xml , vito oder dem startup script?