openv/openv #393: Auslesen IST Temperatur Werte

Brookyman, 2018-02-04:
Hallo, ich habe meine vito.xml genau nach Angaben geändert. Die Datenpunkte habe ich direkt von Viessmann aus einer PDF genau für meine Anlagen ID. Somit gehe ich erst einmal davon aus das es nicht an den Datenpunkten liegt.

Mein Problem besteht darin, dass ist alle Soll Werte auslesen kann. Also z.B. VorlauffsollHK1, Betriebsart, Stellungen von Ventilen, SollTemperaturen usw.... Jedoch wenn ich die Sekundär Vorlauf- /Rücklauftemperatur, Aussentemperatur, Primärtemperatur, etc... auslesen will kommt im Telnet immer folgende Meldung:

>ERR: >FRAMER: addr was still active FE06
>FRAMER: Command send
>FRAMER: ERROR address 01C1 code 4
Fehler recv, Abbruch
Fehler beim ausfuehren von Außentemperatur

Kann mir hier wer sagen was dieser Fehler bedeutet? Er ist bei allen IST Temperaturen der selbe nur mit anderer Adresse.

domschi, 2018-02-24:
Hallo Brookyman,

Haben Sie den fehler gefunden? ich habe das gleiche Problem. 

DS

Gmoawirt, 2018-08-24:
Hallo,

habe genau dasselbe Problem. Gibt es vielleicht schon Lösungsansätze? Danke!

mxwi, 2022-01-21:
Hi, bin gerade zufällig darüber gestolpert. Ist ja schon 1-2 Jahre her. Aber probiert doch mal den Datenpunkt 01**0**1 statt 01**C**1.