openv/openv #394: Systematische Datenhaltung für Datenpunkte/Adressen?

Morg42, 2018-02-12:
Hallo,

da sich im Wiki Dateien nicht so komfortabel unterbringen lassen und auch die jeweilige Formatierung vielleicht nicht jedermanns Sache ist - und ich die alte "Datensammlung" hier noch gar nicht wieder gefunden habe:

ist es geplant, die xml-Dateien der User oder alternativ die Datenpunkte ohne XML-Format für verschiedene Systeme bereitzustellen?

Mir sind ein paar Möglichkeiten eingefallen; wenn Interesse an so etwas besteht, dann können wir das ja gern mal diskutieren:

<ul>
<li>im vcontrold-Repository, hier im Repository oder in einem eigenen Repository eine Verzeichnisstruktur mit Anlagen und Dateien, z.B.

/ID1/user1/vito.xml
/ID1/user2/vito.xml
/ID1/user2/erlaeuterungen.txt
/ID2/user3/vito.xml
...

Vorteile: Dateien können direkt übernommen werden
Nachteile: uneinheitliche Benennung, Vergleichen/Zusammenführen aufwendig

<li>große Tabelle mit allen Werten (ähnlich wie bei wikispaces)

Vorteil: Vergleichbarkeit, Übersicht
Nachteil: praktisch unmöglich zu pflegen, XML muss selbst geschrieben werden

<li>sowas wie eine Datenbank, web-gestützt, sortier-/filterbar

Vorteil: Übersicht, Vergleichbarkeit, ggf. XML-Export
Nachteil: muss jemand schreiben und hosten
</ul>

Falls jemand zufällig weiß, ob es ein Web-Tool gibt, mit dem man die DB umsetzen könnte, schau ich mir das gerne an und versuche mal eine Umsetzung...

speters, 2018-02-13:
Ein Webfrontend zu einer DB existierte mal. Leider waren weder Datenbank noch das Script offen verfügbar, so dass nach Einstellung durch den privaten Betreiber die Community in die Röhre guckte.

Mir ist die Lösung mit den XML-Dateien in einem Repo die sympathischste, weil sie offen, einfach, unabhängig und dauerhaft ist.
Das von `vcontrold` gewählte XML-Format ist leider nicht ganz ideal (sh. https://github.com/openv/vcontrold/issues/15), aber verbreitet und brauchbar.

Ich will aber deinem Tatendrang nicht im Wege stehen:
DBs kann man recht fix mit Django, Flask oder CRUD Frontends wie https://github.com/scoumbourdis/grocery-crud ins Netz bringen.

ramack, 2018-02-14:
Hi
Ich hatte damals den webapace zur Verfügugn gestellt. Der Autor der skripte war aber nicht mehr interessiert. Wenn ihr wollt können wir das schon wieder zum Leben erwecken aber ich kann halt nicht viel Zeit investieren. Wenn es nach mir gegangen wäre hhätte man damals auch schon eine Editiermöglichkeit eingebaut...

Also wenn sich jemand beteiterklärt die Skripte zu pflegen können wir gerne wieder den Phpserver und die MySQL db ans laufen bringen... einfach melden

speters, 2018-02-14:
@ramack 
Super, dass da noch was existiert. Vielleicht hast du ja die Möglichkeit, die Dinge (Scripte, Dumps) hier auf GitHub als Repository zur Verfügung zu stellen.

@Morg42 
Durch "genauere Betrachtung" von Viessmanns Vitosoft-Demo (Setup.exe mit 7zip entpacken, nach ggf. UTF16 kodierten XML Dateien Ausschau halten) und ein bisschen Script-Hilfe per Python käme man flugs an alle Infos z.B. als sqlite-DB:

[image]

[image]

Das oben ist natürlich nur ein fiktives Szenario und die Bilder sind erstunken und erlogene Mockups, klar.

Wie aber realiter mit solchen Infos umgehen, ohne auf urheberrechtliche Probleme zu stossen? Viessmann ist in seiner Infopolitik ja derart restriktiv, dass ich kaum glaube, dass sie die Daten auf Nachfrage hin unter eine brauchbare Lizenz stellen.

Morg42, 2018-02-14:
Ich habe die Viessmann-Dateien mal durchgeprüft, aber das Problem scheint mir, dass viele Adressen auch in manchen Anlagen gehen und in anderen nicht. Dazu kommt noch, dass es Datenpunkte gibt, bei denen Viessmann behauptet, dass sie stimmten, aber sie stimmen realiter nicht.

Das sind einige der Punkte, warum wir überhaupt die Daten zusammensammeln....

Es muss nicht zwingend ein Webinterface sein, aber das macht für die Bearbeitung und Übersicht den Zugang einfacher. Wenn man dann ggf. sogar noch eine XML-Importfunktion hätte ;)

Webspace mit entsprechenden Servern wäre nicht das Problem. Wenn noch ein Grundstock an alten Skripten oder Datenbanken existiert, wäre das doch schon mal ein Anfang...

ramack, 2018-02-16:
Mh, es sieht so aus als hätte ich keine Backups von der VM auf der die vito-DB gelaufen ist erstellt :-( Und da ich letztes Jahr auf einen neuen Host umgezogen bin hat es die vito-VM gekostet. Tut mir leid.

Hatte versucht Hanspeter (@vitoopen) zu fragen, ob die DB noch genutzt und/oder gepflegt wird, aber mangels Rückmeldung das Thema dann nicht weiter verfolgt.

Ich finde es weiterhin sinnvoll eine zentrale DB aufzubauen, die gemeinschaftlich editierbar sein sollte, Im- sowie Export für vcontrold-xml wäre dabei natürlich super.

vitoopen, 2018-02-18:
@ramack sorry dass ich nicht reagiert habe.
Habe bei mir noch eine Kopie am laufen (http://gysin.dtdns.net/vitoopen_rampro/), die auf dem gleichen Stand wie die deiner VM sein sollte.

Das Repository ist eine MySQL DB. Dazu kommen ein paar PHP scripts und etwas CSS.
Leider kann ich die DB so nicht herausgeben, denn einiges ist direkt von der Vitosoft SW (Stand 2008) übernommen.
Für eine neue Datenbank könnten die Daten jedoch in einem passenden Format exportiert werden.

ramack, 2018-02-18:
Ich verstehe zwar nicht recht warum du Bedenken hast die DB öffentlich bereitzustellen dir einen Export jedoch vorstellen kannst aber egal. Ich fände es super, wenn du die PHP Skripte und einen DB dump hier einchecken könntest...

@speters

Morg42, 2018-02-18:
@vitoopen:

Möchtest du denn deine DB als "offizielle" openv-Datenbank (weiter) betreiben? Oder möchtest du die eher als persönliche Referenz benutzen?

Wenn du bereit wärst, die Skripte "freizugeben", würde ich anbieten, eine offizielle DB aufzusetzen, in die jeder/angemeldete/ausgewählte Daten eintragen und importieren und jeder XML-Dateien (oder ggf. noch andere) exportieren könnte.

Webspace usw hätte ich, auch einen passenden Domainnamen könnte ich bereitstellen (openv.de ist leider schon vergeben... ;) )

Wenn du das nicht möchtest, dann wäre es gut, wenn du dich entsprechend dazu positionieren könntest.

Morg42, 2018-03-05:
Hm.. hat jemand von euch Kontakt zu @vitoopen? Ist er im Urlaub?

Oder laufe ich hier einem toten Pferd hinterher?

Ich bin nicht so richtig motiviert, eine Datenbank "from scratch" neu zu schreiben, zumal vernünftige und - wie ich denke - nutzbare Ansätze vorhanden sind. Wenn ich auf einer Basis (und sei sie auch ohne ggf. eingeschränkt nutzbare Daten) aufsetzen kann, mache ich das gern. Das muss hinterher nicht "meine" DB werden, dafür ist das hier der falsche Rahmen - aber wenn ich raten müsste, was hinter einer dynamic DNS webadresse steckt, wäre zumindest für den Anfang "richtiger" Webspace möglicherweise praktischer. 

Wenn ich hier ein Feedback bekomme, dass ich das machen soll - gern. Ich bin da ziemlich flexibel, was Ausgestaltung und Umsetzung angeht. Wenn ich Feedback bekomme, dass ich das nicht tun soll - auch gut. Da bin ich auch flexibel. Meine Heizung läuft, und den einen Punkt, den ich gern hätte, kann derzeit offensichtlich nichtmal Viessmann klären, also bin ich für meinen Bedarf soweit bedient.
Wenn ich gar kein Feedback bekomme, lasse ich es. Nur für mich muss ich da kein Projekt aufziehen...

ramack, 2018-03-05:
Also ich finde es vernünftig eine gemeinsame Datenbank zu pflegen und würde es befürworten das was @vitoopen damals begonnen hat weiter auszubauen und die Möglichkeit vorzusehen, dass man auch Daten eintragen und ändern kann. Am Webspace soll es nicht scheitern, ich kann gerne wieder ein VM dafür zur Verfügung stellen. Viel mehr als Infrastruktur und vielleich das ein- oder andere review kann ich aber aus Zeitgründen gerade nicht beisteuern.

Morg42, 2018-03-07:
Dann warten wir mal auf ein Feedback von @vitoopen...  an die DB bzw. die Oberfläche würde ich mich schon ransetzen, und auf einen späteren endgültigen Server umzuziehen wäre dann ja auch kein Problem, wenn er oder wir da Vorlieben haben...