openv/openv #398: Intepretation von COP und JAZ werten

domschi, 2018-02-26:
Hallo,
ich nutze seit neustem die vcontrold zum loggen unserer Vissmann Vitocal 200-S (WO1C). Es klappt alles sehr gut und ich möchte mich bei der Community hier bedanken für die gute Arbeit.

ein Problem habe ich allerdings noch:
die Ausgabe von diversen JAZ und COP Zahlen kann ich nicht interpretieren.
 <command name="getJAZ" protocmd="getaddr">
         <addr>1680</addr>
         <len>2</len>
         <unit>CO</unit>
         <description>Statistik - Energiebilanz: Jahresarbeitszahl (0..10)</description>
     </command>
     <command name="getJAZHeiz" protocmd="getaddr">
         <addr>1681</addr>
         <len>2</len>
         <unit>CO</unit>
         <description>Statistik - Energiebilanz: Jahresarbeitszahl Heizen (0..10)</description>
     </command>
     <command name="getJAZWW" protocmd="getaddr">
         <addr>1682</addr>
         <len>2</len>
         <unit>CO</unit>
         <description>Statistik - Energiebilanz: Jahresarbeitszahl WW (0..10)</description>
     </command>
     <command name="getCOPHeiz" protocmd="getaddr">
         <addr>1690</addr>
         <len>2</len>
         <unit>CO</unit>
         <description>Statistik - Energiebilanz: COP Heizbetrieb (0..10)</description>
     </command>
getJAZ zum Beispiel, gibt bei mir Zahlen über 8000 aus und getCOPHeiz von 37 - 0.
Ich dachte es sollten nur Zahlen zwischen 1..10 ausgegeben werden. Mich interessiert die Energiebilanz und die COP Werte unserer WP. Vielleicht kann mir einer von euch auf die Sprünge helfen. 
Danke 
Domschi

MrIcemanLE, 2018-02-26:
Hallo @domschi 
Ich stehe vor dem gleichen Problem. Habe in diesem Issue hier mal meine Daten gepostet: #388 

Hinter der 0x1682 scheint ein gleitender Durchschnitt (12 Monate?) zu stecken. Diese Werte musst du durch 10 teilen.
Die 0x1690 scheint ein Momentanwert zu sein, der je nach Zeitpunkt des Auslesens auch mal unrealistisch hohe Werte annehmen kann (Peaks in meinem Diagramm)

Die 0x1680 und 0x1681 kann ich auch nicht interpretieren. Bei mir aktuell 8490 und 10792

domschi, 2018-02-26:
Danke MrIcemanLE für die Antwort,

Bei Beobachtung des 0x1690 COP Heizbetrieb fällt mir auf, dass ich nur Werte !=0 bekommen wenn die WP Strom zieht. Es gibt eine Korrelation zwischen COP Werten und Leistungsaufnahme der Heizung. Würde ich den dann erhaltenen Wert auch durch 10 teilen bekäme ich plausible Werte zwischen 2,5 – 4. Vielleicht gibt es noch Andere die die gleiche Beobachtung machen.

Domschi

MrIcemanLE, 2018-02-26:
> Bei Beobachtung des 0x1690 COP Heizbetrieb fällt mir auf, dass ich nur Werte !=0 bekommen wenn die WP Strom zieht.
Ich denke, dass 0x1690 ein Momentanwert der Anlage ist **unabhängig** von der Betriebsart (Heizen/WW)

Mich würden aber eher die Adressen der wöchentlichen Werte interessieren. So wie hier: #262

wuselAUT, 2019-01-25:
Mir kommt vor, dass die Zahlen bei 0x1680 um einen Offset von 4000 verschoben sind und dann durch 1000 zu dividieren. Wenn ich deine Zahlen nutze würde das eine JAZ von 4.49 ergeben, was sich mit dem von dir geposteten Diagramm gar nicht schlecht decken würde @MrIcemanLE. Bei mir wird sich dabei eine JAZ von 4.2 ergeben, was auch nicht unrealistisch klingt.

MrIcemanLE, 2019-01-26:
@wuselAUT ich habe noch einen anderen Ansatz verfolgt. Die Werte von 0x1680 teile ich durch (256*10) also 2560. Damit komme ich auf plausibel Werte, die der JAZ Heizen entsprechen würden. Also wieder einem gleitenden Mittelwert über die letzten 12 Monate (möglicherweise).

Ähnliches habe ich bei der 0x1681 gemacht. Hier habe ich nur durch 256 geteilt und komme (möglicherweise) auf die JAZ WW. 

Diese Annahmen müsste ich mal mit Stichproben belegen. Da sich die Werte aber nur sehr langsam ändern ist das recht schwer. 

Zumindest decken sich die so errechneten Werte mit der Anzeige.

Die JAT gesamt (aktuell 3,7) habe ich noch nirgends gesehen oder errechnet.

Aktuell habe ich folgende Werte:

0x1680: 10277 / 2560 = 4.014453125 => JAZ Heizen lt. Anzeige
0x1681:  797.6 / 256 = 3.11562490625 => JAZ WW lt. Anzeige
0x1682: 31 / 10 = 3.1 => JAZ WW lt. Anzeige
0x1690: 46 / 10 = 4.6 => Momentanwert des COP bei Betrachtung des Diagramms
0x1691:  0

MrIcemanLE, 2019-01-26:
BTW:

Hat einer von euch an der WO1C schon mal diese Werte getestet:
https://github.com/openv/openv/issues/420#issuecomment-453774576

`ElektroEnergie_Warmwasser_1Woche 0x16D0, 2bytes factor=0.1
WaermeEnergie_Warmwasser_1Woche 0x16D2, 2 bytes, factor=0.1
ElektroEnergie_Heizen_1Woche 0x16D1 2 bytes, factor=0.1
WaermeEnergie_Heizen_1Woche 0x16D3, 2 bytes, factor=0.1`

Ich bekomme bei allen eine Null.

h0nIg, 2022-09-10:
Hi, das Problem ist warscheinlich die Byte Länge. In den vito.xml die ich gefunden habe war die JAZ immer mit 2 byte definiert, es sind aber 1 byte. Damit passt die JAZ bei mir

```
     <command name="getJAZWW" protocmd="getaddr">
         <addr>1682</addr>
         <len>1</len>
         <unit>CO</unit>
         <description>Statistik - Energiebilanz: Jahresarbeitszahl WW (0..10)</description>
     </command>
```

msglueck, 2023-04-28:
Hi zusammen, kann bestätigen @h0nIg hat recht. Hab mir noch eine Unit Definition dazu gebaut die den Wert durch 10 teilt um 3.8 bzw. 5.6 zu erhalten.

```
    <unit name='JAZ'>
	   <abbrev>JAZ</abbrev>
	   <calc get='V/10' set='V'/>
	   <type>int</type>
	   <entity></entity>
	</unit>
```
und die JAZ Kühlen ist folgende Konftiguration:

```
     <command name="getJAZAC" protocmd="getaddr">
         <addr>1683</addr>
         <len>1</len>
         <unit>JAZ</unit>
         <description>Statistik - Energiebilanz: Jahresarbeitszahl AC Kühlen (0..10)</description>
     </command>
```