openv/openv #400: Vitocal 300A mit Vitotronic 200 (WO1B) - ErrorStates

s4ndst0rm, 2018-03-01:
Wie komme ich denn an die ErrorStates? Ich kann zwar abfragen ob es eine Störung gibt (Müsste Adresse 0A82 sein) aber ich hätte gerne den ErrorState.
Ist das die Adresse: 7507?
Weiterhin hätte ich dann auch gerne noch den Detailfehler: In der 7507 steht dann ja "nur" ein "Fehler im Kältekreis" aber nicht was tatsächlich den Fehler ausgelöst hat. Beim nächsten Start der WP ist dieser erweiterte Fehler auch weg..

Danke!

warmup72, 2018-04-15:
Hat sich bei dem Thema schon was getan?
Niemand eine Idee?
Leider sind die hier im Forum bekannten Adressen für Errorstate (7507, 7510, 7519, 7522, 752B, .. -  als Error01ff) alle nicht für die WO1B geeignet:-(

Das Einzige, was ich anbieten kann sind
Anzahl-Sammelstörung - 0511
Stunden-Sammelstörung - 0591
Status-Störung - 7579

Hab mich als beholfen die Anzahl abzufragen, ob ungleich zur vorherigen Auslesung. Das Problem ist aber, dass hier NUR Störungen gezählt werden. Reine MELDUNG (also die beispielsweise die allseits bekannte MELDUNG 07) wird hier nicht gezählt :-(

Würde mich freuen, wenn jemand eine Idee oder gar einen Code hätte um auch diese erfassen zu können.

s4ndst0rm, 2018-11-13:
Ich habe auch nochmal die Vitosoft Dateien durchgesehen aber leider ohne Erfolg.

Da ich auch die Meldung 07 habe und nie rechtzeitig an der Steuerung bin um den eigentlichen Errorcode zu sehen, kann ich auch nichts analysieren :(

Ich wäre weiterhin über jede Idee dankbar!

warmup72, 2018-11-21:
Guten Abend,
nein ich habe auch nix gefunden. Alle Mailbenachrichtungen auf Grund von anderen Datenpunkten ermittelt scheitern. Da die Meldung 07 eben auch nur eine Meldung und kein Fehler ist, wird der interne Fehlerzähler auch nicht hochgezählt. 
Nachdem ich nun 6 Jahre ohne Meldung 07 gelebt habe, kommt sie auch wieder bei mir. Sehr selten allerdings von 01-03 2018 waren es 5x und seit Mitte 10.2018 gerade 1x