openv/openv #403: I am lost in the multiples versions.  Please help !

JacquesBB, 2018-03-05:
Dear All, 

Sorry for writing in english. 

This site is great as it includes a lot f information, but I must say I am lost in the multiples versions that are added  without much clarification. 
I have  a Vitalodens 300  with a Vitotronic 200 KW2. 
I have already made an optolink 5 years ago, and I am resuming this in order to be able to remotely control my heater. 

I have no firm idea of which software or which platform to use. I could use anything, from PC, to Rasp PI, or linux machine, or NAS server. I have all. I have no trouble with one or another programming language. 

My question is : **Which software will fullfil my needs.** 

This is where I need clarification. 
I understand that many softwares will just record the parameters of the heater, but will not allow easily to remote control the heater. Am I wrong ? 

I have hooked a PC (widows on my optolink) and installed v-control 1.3 . 
This is fine but I do not want to leave this PC on all the time, and will need to remotely access to it, which is not so easy. 

5 years ago, I tried to make a radio link with the optolink. It was OK, I could display the info, but it was not very easy to chage the parameters. Maybe there were some delays that make things difficult. 

Now I need to access through the internet, from very far away and change the parameters of the heater from there. 

So please help me to decide which is the best solution before I jump into the making of it. 

- I have a linux station permanently on, and hooked to the web
- I know how to use RaspPi, and could add one devoted to that ( loww power) 
- I could use a ESP8266, ESP32 
- I have a NAS permanently on. 

Thansk for any advice

bertmelis, 2018-03-05:
Are you sure you can't use the 300 protocol? "Vitalodens 300 with a Vitotronic 200" sound like it's a rather recent installation?

JacquesBB, 2018-03-05:
Thanks for the answer. 
How can I check ? 
I have the v-control 1.3 installed. Is there a  way to test the 300 protocol with it ? 
Right now, it  has recognised  V200KW2. 

I will only be able to make the tests next week end, but I can collect information meanwhile, 
and even prepare some software.

bertmelis, 2018-03-05:
I've got no experience with the Windows software, and there's no source to see how it does this test.

On the Linux version, you'll have to define your device in the xml-files so you can define yourself which protocol to follow (and see if it's working).

JacquesBB, 2018-03-06:
The advantage of the Windows software is that it has a nice GUI. 
The problem is that it is no longer active and the code is not provided. 

The unix software is active, and from what I understand it reduces now to a single one vcontrold. 

After looking to the different solutions, what I decided  for now is : 

- keep the v-control windows software on my heater as it works, provides a nice interface, and allows 
all tuning of the heater parameters. But I will shut it down most of the time as I do not want  to waste power with a PC, even a low power pc. 

- Adopt your  ESP8266 library to be able to remotely access to  the heater, with a low poser device that will be always on. This will be limited to a few orders. 
From now on, I will thus communicate on your thread.

smbunn, 2018-07-18:
I have identical boiler hardware and have been using a Raspberry Pi for the last 6 years.  It runs vcontrold and I link this to Munin on the same raspberry pi that provides my graphical feedback.  I have managed to write to temperature setpoints successfully, but for some reason I cannot change the mode on my boiler (e.g. to take it out of standby into HW only)