openv/openv #409: Discovering commands (for V200KO2B or others)

berkinet, 2018-04-11:
There have been a number of posts, here and elsewhere, regarding the supported commands/addresses for the V200 KO1B & KO2B. But, as far as I can tell, nobody has ever resolved the issue completely.  The information I have found in the address list  (https://github.com/openv/openv/wiki/Adressen) shows just 17 known addresses, and several of those do not work. On the other hand, some supported addresses, like room temp are reported wrong, and see to be incorrectly identified.  Also, some posts indicate the KO[12]B will support the KW and 300 protocols. But, I cannot get the 300 to work.

So, my 2 questions...
1) Does anyone have an up-to-date vito.xml and control.xml for this device?
2) Is there a way to walk through all the possible command numbers to, air least, discover what exists? If so, I could then try to correlate the address with measurements from the V200 itself.

Thanks,

Richard

bertmelis, 2018-04-11:
I'll write a long version next week but, in z nutshell, this is what I did:
- download a copy of Vitosoft (demo version)
- extract the xml-files with the defenitions of the datapoints. Don't install but open using a zip-file manager like 7-zip of similar.
- find your device ID in this file
- correlate the datapoints with the datatypes (use a second xml-file that resides in the same directory)

Excel and vlookup are a great help.

I too short on time right now to write a complete walkthrough but stay tuned.

Remark: I'm not sure you aren't violating copyright laws. But in the EU you're quite free to do this for interoperabilty reasons. And since we're doing this to connect our heaters to our home automation systems...

berkinet, 2018-04-11:
Ok... Much thanks for the pointers. I downloaded the demo, found 100 entries for Typ KO2B in DPDefinitions.xml. But, couldn't find any cross references in any other files. Interestingly, while several documents on this (GitHub) site identify my unit as a 209F, there is no mention of that device number in any of the xml files. I do find 2098, and others.

I suspect (actually hope) I have misinterpreted your instructions...

bertmelis, 2018-04-11:
You can always install the software and see if the software recognizes your device. You have 90 days to evaluate if it's working and which data you can extract.

Vitosoft worked out of the box with the diy USB optolink.

berkinet, 2018-04-11:
I'd have to run Vitosoft in a VM... only Macs here. But, I could do that. Right now I have a RPi as the interface device.

berkinet, 2018-04-11:
Ok. I have exported the system and have the following (first few lines only)
<?xml version="1.0" encoding="utf-8"?><ParameterSets Version="0.0.26.4683"><ParameterSet Id="1" CompanyId="24" DatapointId="3" DatapointTypeId="340" CreatedDate="2018-04-11T07:52:00.7660229-07:00"><Name>KO2B</Name><Description>heater parameters</Description><ParameterValues
<ParameterValue TypeId="1" ValueTypeId="1"><Value Type="System.String">ecnStatusEventType~0</Value></ParameterValue>
<ParameterValue TypeId="7" ValueTypeId="7"><Value Type="System.Double">64.6</Value></ParameterValue>
<ParameterValue TypeId="9" ValueTypeId="9"><Value Type="System.Double">20</Value></ParameterValue>
<ParameterValue TypeId="10" ValueTypeId="10"><Value Type="System.Double">10.2</Value></ParameterValue>
<ParameterValue TypeId="12" ValueTypeId="12"><Value Type="System.Double">64.3</Value></ParameterValue>

The problem is, I cannot associate any of the Typeids or ValueTypeids with anything that looks like a name or address.

Any ideas?

philverh, 2018-06-26:
Interpreting the data available in Vitosoft config files is a multiple step exercise. 
First I identified my device in one of the Geraeteliste.html files, then looked at ecnDataPointType.xml to get a complete list of all addresses for my device. The looking further in the ecn*.xml files I found the group my device belongs to, the values available for this group, their category (statistics, settings, current value, ...) length, their coding (Int, ouble, text, binary, ...) and if they could be read or set.  

Quite some work, but a reliable source.
Philippe.

promy, 2018-08-07:
@bertmelis : i could not find a demo version of vitosoft? (i need to register as a customer on the viessmann site?) Could you point me toward the software of provide me with the config file for the vitocal 300-G?

philverh, 2018-08-07:
On 07/08/2018 13:27, promy wrote:
>
> @bertmelis <https://github.com/bertmelis> : i could not find a demo version of
> vitosoft? (i need to register as a customer on the viessmann site?) Could you point
> me toward the software of provide me with the config file for the vitocal 300-G?
>
Right now I can't find the Vitosoft SW either. Strange.
A couple of weeks ago I downloaded the demo, and looked up the parameters for my
device (2046, aka Vitocal 300-G).
I ended up with a massive number of items which can be read or updated. I'm in the
process of generating (holidays slowed down the process ;-) ) the .xml files to work
with vcontrold.

promy, 2018-08-07:
Strange that the post disappeared. My browser crashed so maybe thats the issue...
Looks like we have the same machine :-)
I found a version of vitosoft via the german site...
Maybe we can share the results

philverh, 2018-09-10:
I did not mess with the temp probe, but went into the installer/contractor menu (with the required password) and swithched on (Manual relay settings) the primary pump and the NC. Of course this gets reset after half an hour or something like that, if there is no interaction with the console. I only used it for tests, so was not bothered by the automatic reset. The relais status info is available, and could be set progammatically, although it may be quite more complex than on the display of the unit. 

For the SPF, I haven't yet looked into it, but it is definitely a part of the statistics, 
I have a device type  VBC700_BW_WW for which I find in the Viessmann data that the COP for the heating should be at address 0x1690 (and 0x1691 for the domestic hot water). Of course SPF is not equivalent to COP, as it is an average over a time span. I cannot immediately find the address of the remaining SPF data (last 12 months, one of which is displayed on the system overview), but as it appears on the boiler, it should be hiding somewhere. 

I've been quite busy with other projects lately, but plan on restarting this heating related stuff at the end of the month.

mactoolz, 2018-11-04:
Hello everybody

I would like to record the theme again. What about the K01B / K02B control?

That seems to be a very difficult control?

greeting
MacToolz