openv/openv #413: "set" Befehle

Goersch, 2018-05-06:
Hallo,

jemand eine Idee, warum set-Befehle (z.B. setBetriebArtM2) zwar mit einem "OK" quitiert werden, aber keine Umschaltung der BA erfolgt? (alle get Befehle funktionieren)

`vctrld>setBetriebArtM2 WW
DEBUG:Sat May 5 19:01:50 2018 : Befehl: setBetriebArtM2 WW
DEBUG:Sat May 5 19:01:50 2018 : WW -> 00
DEBUG:Sat May 5 19:01:50 2018 : >SEND: 04
DEBUG:Sat May 5 19:01:50 2018 : >FRAMER: no preset result
DEBUG:Sat May 5 19:01:50 2018 : Warte auf 05
DEBUG:Sat May 5 19:01:51 2018 : <RECV: 05 (330.0 ms)
DEBUG:Sat May 5 19:01:51 2018 : >SEND: 01
DEBUG:Sat May 5 19:01:51 2018 : >SEND: F4
DEBUG:Sat May 5 19:01:51 2018 : >SEND: 33
DEBUG:Sat May 5 19:01:51 2018 : >SEND: 01
DEBUG:Sat May 5 19:01:51 2018 : >SEND: 01
DEBUG:Sat May 5 19:01:51 2018 : >SEND: 00
DEBUG:Sat May 5 19:01:51 2018 : >FRAMER: no preset result
DEBUG:Sat May 5 19:01:51 2018 : <RECV: len=1 00 (10.0 ms)
DEBUG:Sat May 5 19:01:51 2018 : <RECV: received 00
DEBUG:Sat May 5 19:01:51 2018 : 00 -> OK
DEBUG:Sat May 5 19:01:51 2018 : OK
OK
vctrld> `

Gibt es evtl. Einschränkungen, wenn eine "VC 100 LAN1" (Verbindung für Vitodata 100) im Einsatz ist?

Danke !
Georg

Goersch, 2018-05-11:
Keiner eine Idee ?

Hab nun vieles ausprobiert - aber ohne erfolg.

Es handelt sich um eine VScotH01 Typ 20CB - Ich fahre das KW Protokoll

Goersch, 2018-05-11:
bin jetzt auf die letzte hier bei GIT liegende Version umgestiegen
bin auf Protokoll P300 umgestiegen
"get" funktioniert
"set" leider nicht

> Vctrld>setBetriebArtM2 H+WW
> 
> DEBUG:Fri May 11 18:53:04 2018 : Command: setBetriebArtM2 H+WW
> DEBUG:Fri May 11 18:53:04 2018 : H+WW -> 03
> DEBUG:Fri May 11 18:53:04 2018 : >SEND: 41
> DEBUG:Fri May 11 18:53:04 2018 : >SEND: 06
> DEBUG:Fri May 11 18:53:04 2018 : >SEND: 00
> DEBUG:Fri May 11 18:53:04 2018 : >SEND: 02
> DEBUG:Fri May 11 18:53:04 2018 : >SEND: 33
> DEBUG:Fri May 11 18:53:04 2018 : >SEND: 01
> DEBUG:Fri May 11 18:53:04 2018 : >SEND: 01
> DEBUG:Fri May 11 18:53:04 2018 : >SEND: 03
> DEBUG:Fri May 11 18:53:04 2018 : >SEND: 40
> DEBUG:Fri May 11 18:53:04 2018 : <RECV: len=1 06 (30.0 ms)
> DEBUG:Fri May 11 18:53:04 2018 : <RECV: received 06
> DEBUG:Fri May 11 18:53:04 2018 : >FRAMER: framer_set_actaddr framer_current_addr = 0133 (was FFFF)
> DEBUG:Fri May 11 18:53:04 2018 : >FRAMER: Command send
> DEBUG:Fri May 11 18:53:04 2018 : >FRAMER: no preset result
> DEBUG:Fri May 11 18:53:04 2018 : <RECV: len=1 41 (10.0 ms)
> DEBUG:Fri May 11 18:53:04 2018 : <RECV: received 41
> DEBUG:Fri May 11 18:53:04 2018 : <RECV: len=1 06 (0.0 ms)
> DEBUG:Fri May 11 18:53:04 2018 : <RECV: received 06
> DEBUG:Fri May 11 18:53:04 2018 : <RECV: len=1 03 (0.0 ms)
> DEBUG:Fri May 11 18:53:04 2018 : <RECV: len=1 02 (0.0 ms)
> DEBUG:Fri May 11 18:53:04 2018 : <RECV: len=1 33 (10.0 ms)
> DEBUG:Fri May 11 18:53:04 2018 : <RECV: len=1 01 (0.0 ms)
> DEBUG:Fri May 11 18:53:04 2018 : <RECV: len=1 01 (0.0 ms)
> DEBUG:Fri May 11 18:53:04 2018 : <RECV: len=1 01 (0.0 ms)
> DEBUG:Fri May 11 18:53:04 2018 : <RECV: len=1 41 (10.0 ms)
> DEBUG:Fri May 11 18:53:04 2018 : <RECV: received 03 02 33 01 01 01 41
> DEBUG:Fri May 11 18:53:04 2018 : >FRAMER: framer_reset_actaddr framer_current_addr = FRAMER_NO_ADDR (was 0133)
> DEBUG:Fri May 11 18:53:04 2018 : >FRAMER: ERROR address 3301 code 1
> DEBUG:Fri May 11 18:53:04 2018 : Error in recv, terminating
> DEBUG:Fri May 11 18:53:04 2018 : Error executing setBetriebArtM2 H+WW
> ERR: >FRAMER: ERROR address 3301 code 1
> Error in recv, terminating
> Error executing setBetriebArtM2 H+WW

Goersch, 2018-05-14:
hab nun eine Vitocom 100 WLAN aktiviert - BA Umschaltung funktioniert damit; es liegt also nicht an der Heizung oder deren Einstellung.

Kann mir jemand sagen, wie ich die Kommunikation zwischen Heizung und der Vitocom protokollieren kann um den Fehler zu finden?

bertmelis, 2018-05-14:
Your Viessmann is returning an error code. Are you sure you're using the right address? How many heating circuits are defined?

Goersch, 2018-05-14:
there is just one heating circuit - M2; get works - so it should be the right address

speters, 2018-05-15:
@Goersch wie bertmelis bereits sagte, antwortet die Heizung, dass dein set-Versuch einen Fehler nach sich zieht.

Du sagtest, dass du nur einen Heizkreis hast, versuchst aber die Betriebsart vom zweiten Heizkreis zu setzen.

Bei lediglich einem Heizkreis wird i.d.R. die Betriebsart an Adresse 0x2301 gesetzt.

bertmelis, 2018-05-15:
No, it means that you can read and write to 2301 but only read at 3301 (and a default value is returned?).

See, although no second heating circuit is defined, the memory is there and is accessible. But it hasn't been configured so writing to it returns an error. You could argue that reading the address should also result in an error.

Goersch, 2018-05-15:
ok, but there is one issue to think about: only 3301 returns the right value. When I change the mode e.g. via display, 3301 changes to the correct value.
I have read 2301 once yesterday. It returns not the correct mode („Off“ instead of „heating water“); which makes sense, because there is no M1)

Goersch, 2018-05-15:
auf der 3500 kommt ein Wert "2", der sich aber auch nach Umschaltung der BA (via App) nicht ändert
bei der 3501 kommt eine Fehlermeldung im Antworttelegramm der Heizung (Status 3 Byte 3)

Welche DB meinst Du?

speters, 2018-05-16:
Oh, ich hatte "%Betrieb%" als Stichwort bei meiner Suche durch die DB gemacht. Asche auf mein Haupt...

Das Gesuchte war aber mit "BA" abgekürzt: _BedienteilBA_GWGA1~0x2323_
 und _BedienteilBA_GW GM2~0x3323_ mit 0=Abschalt;1=Nur WW;2=Heizen + WW;3=Dauernd Reduziert;4=Dauernd Normal.

Für dich mit deinem M2 müsste das also 0x3323 sein.

Nun aber ... hoffentlich ...

speters, 2018-05-16:
Sparbetrieb müsste bei Dir auf 0x3302 liegen.

Goersch, 2018-05-18:
ja, danke, funzt :)

klopsi007, 2018-05-23:
Hat jemand von euch die Adresse für einen Reset bei Störung / quittierung ?  Könnte es die sein ? 0xA050 oder doch diese 0x5724 ? VitoPendHO1

passuff, 2018-08-07:
Hast du mittlerweile herausgefunden wie die Adresse für den Reset Befehl lautet?
Gruß

Goersch, 2018-09-27:
Ob noch jemand die richtige Adresse für die Aktivierung/Deaktivierung des Partybetriebs für mich hat?
Danke!
Georg

dima-357, 2020-11-22:
Please tell me the request address V100
boiler pump in operation or not /
It is not in the list of addresses of Viessmann-adresses

UliGr, 2022-05-03:
Hallo,
ist es möglich, die Speicherladepumpe mit dem Set Befehl anzusteuern? Ich will in der Übergangszeit das vom Dach erwärmte Warmwasser in den Kessel zurücklaufen lassen, damit die Fußbodenheizung unterstützt wird.

WEnn ich den Status von 0 auf 1 änderne, bekommt man eine OK (mit vclient), aber der Status bleibt unverändert bei 0. 

Ich denke mal, dass es allgemein funktionieren sollte.
In der Default-Vito.xml (auf Github) ist die Zirkulationspumpe bei den "set" Befehlen eingetragen. (setPumpeStatusZirku mit Adresse 0x0845)
Diese ist aber ebenfalls nicht ansteuerbar. 
Mit der Software Vitosoft300 ist es mir möglich, die Pumpen anzusteueren.
Die Software kommuniziert ja über das selbe Protokoll. (KW)

Laut der Doku im Wiki hat die Speicherladepumpe die Adresse 0x0846.
Die Heizungsteuerung ist eine V200KW2
Temperaturen, Heizkurve setzen usw funktoniert ohne Probleme.

Folgend ist meine Konfiguration, welche ich von der Zirkulationspumpe übernommen habe:

    </command>
    <command name="setPumpeStatusSp" protocmd="setaddr">
      <addr>0846</addr>
      <len>1</len>
      <unit>RT</unit>
      <description>Setze den Status der Speicherladepumpe</description>
      <device ID="2053">
        <addr>01</addr>
        <len>1</len>
        <bit>2</bit>
        <precommand>getPumpeStatusSp</precommand>
        <unit>BA</unit>
      </device>
    </command>

Gruß Uli

Tonitom, 2022-11-07:
Bitte um Hilfe,
ich verstehe nicht, wie ich mit Telnet die Betriebsart setzen kann "setBetriebArtM2 H+WW".
Wenn ich den Befehl setBetriebArtM2 absetze bekomme ich "ERR: command unknown" zurück. Der "get" Befehl funktioniert.

Muss ich die vito.xml anpassen? Etwa so?

```
    <command name="setBetriebArtM2 H+WW" protocmd="setaddr">
      <addr>3323</addr>
      <len>1</len>
      <unit>BA</unit>
      <description>Setze Betriebsart H2</description>
      <device ID="20CB"/>
    </command>
```
Die DeviceID hab ich mit der WindowsApp "voIdent" auslesen können

Tonitom

Goersch, 2022-11-08:
command name muss ohne H+WW - ich vermute bei Dir gleichen Fehler wie bei mir damals (falsche Adresse - siehe oben)

Tonitom, 2022-11-08:
Hi, sorry ich stehe auf dem Schlauch.
Der Screenshot ist von der /etc/vcontrold/vito.xml. 
Genügt das nicht? Oder muss noch mehr angepasst werden?

[image]

Goersch, 2022-11-08:
da sieht command name ok aus - vorher hattest Du noch „ H+WE“ dran hängen

Tonitom, 2022-11-08:
Hallo, ja, hab schon viel getestet.

Wenn ich mit "commands" die Befehle auflisten lasse, dann ist "setBetriebArtM2" nicht dabei. Trotz der angepassten vito.xml.

Wäre es möglich die vito.xml und die vcontrold.xml zu teilen?

Gruß
Tonitom

Tonitom, 2022-11-08:
wow, das war meine schwierigste Raspi Geburt. Danke für die Unterstützung.

[image]

mathias-ch1903, 2023-02-05:
Hallo zusammen.
Ich habe den Thread hier und viele andere Quellen sorgfältig durchgelesen. Trotzdem bin ich noch zu keiner Lösung gekommen. Ich habe folgendes Problem an einer Gasheizung mit Vitotronic 200 Steuerung (gemäss getDevType: V200KW2 ID=2098 Protokoll:KW) mit zwei Kreisen (M1/M2)

Seit einiger Zeit nutze ich vcontrold / vclient.

Lesen funktioniert gut, ich bekomme mit "getBetriebArtM1" (2301) & "getBetriebArtM2" (3301) die korrekte am Steuergerät eingestellte Betriebsart zurückgemeldet.

Ich habe in der vito.xml auch "setBetriebArtM1" und "setBetriebArtM2" konfiguriert. Das auf den Adressen:
setBetriebArtM1: 2301
setBetriebArtM2: 3301

Die Heizung kann ich damit auch abschalten (setBetriebArtM1 ABSCHALT / setBetriebArtM2 ABSCHALT), die entsprechenden Lesungen getBetriebArtM1 / getBetriebArtM2 melden danach auch ABSCHALT.

Schalte ich die Heizkreise wieder ein (setBetriebArtM1 H+WW, setBetriebArtM2 H+WW) melden die Lesungen getBetriebArtM1 / getBetriebArtM2 zwar "H+WW", aber die Heizung geht nicht in Betrieb (Kesselsoll bleibt unten, Pumpenstatus M1&M2 aus)

Das ganze am Steuergerät beobachtet zeigt auch dass die LED der Kreise auf H+WW wechseln, aber es tut sich nichts.

Erst von Hand am Steuergerät zurück auf zB. nur WW und dann auf H+WW, dann hört man es klicken und die Heizung und Pumpen fahren hoch.

Brauche ich eine andere Adresse zum schreiben? Oder kann diese Heizungssteuerung das gar nicht (siehe Kommentar hier [1])?

Ausgaben:
`setBetriebArtM1 H+WW:
DEBUG:Mon Feb  6 00:15:55 2023 : Command: setBetriebArtM1 H+WW
DEBUG:Mon Feb  6 00:15:55 2023 : Process XXXXX tries to aquire lock
DEBUG:Mon Feb  6 00:15:55 2023 : Process XXXXX got lock
DEBUG:Mon Feb  6 00:15:55 2023 : >FRAMER: open device /dev/ttyUSB0 ProtocolID 00
DEBUG:Mon Feb  6 00:15:55 2023 : Configuring serial interface /dev/ttyUSB0
DEBUG:Mon Feb  6 00:15:55 2023 : H+WW -> 04
DEBUG:Mon Feb  6 00:15:55 2023 : >SENT: 04
DEBUG:Mon Feb  6 00:15:55 2023 : >FRAMER: no preset result
DEBUG:Mon Feb  6 00:15:55 2023 : Waiting for 05
DEBUG:Mon Feb  6 00:15:57 2023 : <RECV: 05 (2000.0 ms)
DEBUG:Mon Feb  6 00:15:57 2023 : >SENT: 01
DEBUG:Mon Feb  6 00:15:57 2023 : >SENT: F4
DEBUG:Mon Feb  6 00:15:57 2023 : >SENT: 23
DEBUG:Mon Feb  6 00:15:57 2023 : >SENT: 01
DEBUG:Mon Feb  6 00:15:57 2023 : >SENT: 01
DEBUG:Mon Feb  6 00:15:57 2023 : >SENT: 04
DEBUG:Mon Feb  6 00:15:57 2023 : >FRAMER: no preset result
DEBUG:Mon Feb  6 00:15:57 2023 : <RECV: len=1 00 (50.0 ms)
DEBUG:Mon Feb  6 00:15:57 2023 : <RECV: received 00
DEBUG:Mon Feb  6 00:15:57 2023 : 00 -> OK
DEBUG:Mon Feb  6 00:15:57 2023 : OK
OK
`

`setBetriebArtM2 H+WW:
DEBUG:Mon Feb  6 00:13:52 2023 : Command: setBetriebArtM2 H+WW
DEBUG:Mon Feb  6 00:13:52 2023 : Process XXXXX tries to aquire lock
DEBUG:Mon Feb  6 00:13:53 2023 : Process XXXXX got lock
DEBUG:Mon Feb  6 00:13:53 2023 : >FRAMER: open device /dev/ttyUSB0 ProtocolID 00
DEBUG:Mon Feb  6 00:13:53 2023 : Configuring serial interface /dev/ttyUSB0
DEBUG:Mon Feb  6 00:13:53 2023 : H+WW -> 04
DEBUG:Mon Feb  6 00:13:53 2023 : >SENT: 04
DEBUG:Mon Feb  6 00:13:53 2023 : >FRAMER: no preset result
DEBUG:Mon Feb  6 00:13:53 2023 : Waiting for 05
DEBUG:Mon Feb  6 00:13:55 2023 : <RECV: 05 (2000.0 ms)
DEBUG:Mon Feb  6 00:13:55 2023 : >SENT: 01
DEBUG:Mon Feb  6 00:13:55 2023 : >SENT: F4
DEBUG:Mon Feb  6 00:13:55 2023 : >SENT: 33
DEBUG:Mon Feb  6 00:13:55 2023 : >SENT: 01
DEBUG:Mon Feb  6 00:13:55 2023 : >SENT: 01
DEBUG:Mon Feb  6 00:13:55 2023 : >SENT: 04
DEBUG:Mon Feb  6 00:13:55 2023 : >FRAMER: no preset result
DEBUG:Mon Feb  6 00:13:55 2023 : <RECV: len=1 00 (60.0 ms)
DEBUG:Mon Feb  6 00:13:55 2023 : <RECV: received 00
DEBUG:Mon Feb  6 00:13:55 2023 : 00 -> OK
DEBUG:Mon Feb  6 00:13:55 2023 : OK
OK
`
[1] https://forum.iobroker.net/topic/2715/viessmann-adapter/206?page=25&lang=de

Danke & Grüsse