openv/openv #414: Vitodens 333 - 20C2 - Config Dateien?

mikenussknacker, 2018-06-18:
Hallo zusammen,

ich habe ein vcontrold auf meine Raspi3 mit der einfachen Optolink Schnittstelle hier aus dem Forum ans laufen gebracht. Die Hardware scheint zu klappen, jetzt suche ich für meine Heizung noch passende xml-Configs. 

Eine Standard vito.xml aus liefert:

[1228] Mon Jun 18 11:01:41 2018 : >SEND: 04
[1228] Mon Jun 18 11:01:42 2018 : <RECV: len=1 05 (790.0 ms)
[1228] Mon Jun 18 11:01:42 2018 : <RECV: received 05
[1228] Mon Jun 18 11:01:42 2018 : >FRAMER: closed
[1228] Mon Jun 18 11:01:42 2018 : >SEND: 16
[1228] Mon Jun 18 11:01:42 2018 : >SEND: 00
[1228] Mon Jun 18 11:01:42 2018 : >SEND: 00
[1228] Mon Jun 18 11:01:42 2018 : <RECV: len=1 06 (20.0 ms)
[1228] Mon Jun 18 11:01:42 2018 : <RECV: received 06
[1228] Mon Jun 18 11:01:42 2018 : >FRAMER: opened
[1228] Mon Jun 18 11:01:42 2018 : >SEND: 41
[1228] Mon Jun 18 11:01:42 2018 : >SEND: 05
[1228] Mon Jun 18 11:01:42 2018 : >SEND: 00
[1228] Mon Jun 18 11:01:42 2018 : >SEND: 01
[1228] Mon Jun 18 11:01:42 2018 : >SEND: 00
[1228] Mon Jun 18 11:01:42 2018 : >SEND: F8
[1228] Mon Jun 18 11:01:42 2018 : >SEND: 08
[1228] Mon Jun 18 11:01:42 2018 : >SEND: 06
[1228] Mon Jun 18 11:01:42 2018 : <RECV: len=1 06 (50.0 ms)
[1228] Mon Jun 18 11:01:42 2018 : <RECV: received 06
[1228] Mon Jun 18 11:01:42 2018 : >FRAMER: addr was still active FE06
[1228] Mon Jun 18 11:01:42 2018 : >FRAMER: Command send
[1228] Mon Jun 18 11:01:42 2018 : >FRAMER: no preset result
[1228] Mon Jun 18 11:01:42 2018 : <RECV: len=7 41 (0.0 ms)
[1228] Mon Jun 18 11:01:42 2018 : <RECV: len=1 20 (20.0 ms)
[1228] Mon Jun 18 11:01:42 2018 : <RECV: received 41 0D 01 01 00 F8 08 20
[1228] Mon Jun 18 11:01:42 2018 : <RECV: len=7 C2 (0.0 ms)
[1228] Mon Jun 18 11:01:42 2018 : <RECV: len=1 15 (10.0 ms)
[1228] Mon Jun 18 11:01:42 2018 : <RECV: received C2 00 05 00 00 01 1E 15
[1228] Mon Jun 18 11:01:42 2018 : 20 C2 00 05 00 00 01 1E -> UNKNOWN
[1228] Mon Jun 18 11:01:42 2018 : UNKNOWN

Die vorletzte Zeile ist wohl der genaue Typ: 20C2. Hier im github hilft mir die Suchfunktion nicht weiter. Gibt es da schon etwas passenderes zum herunterladen und wo finde ich das?

Besten Dank schon mal für die Hilfe

Michael

philverh, 2018-06-26:
Sorry, my German allows me to read but not to answer... 
I found a very complete and reliable source of information about the addresses and formats of the data in the demo version of the Vitosoft 300 software from Viessmann. 
When unzipping the demo (it's an .exe file but it can be unzipped with something like 7-Zip) you get a bunch of strangely named directories. You could also just install the demo but avoid launching the program, as it looks like some updates are done immediately, which change some of the info I was looking for. 
Looking through the files, you will find .XML files with a name starting with "ecn". Example: ecnEventType.xml or ecnEventTypeGroup.xml.
I found a file called "geraeteliste.html", which lists all suported devices, with their commercial name and the technical name of the device. (just open the file with your preferred web browser to display it with formatting. You may need to select one in a language you understand, as there are many files with this name.) This technical name is used futher down. In the .xml files I found my device, the category it belongs to and the different groups of information used by Vitosoft to logically group data. In another .xml file I found for each data element the address where it can be found, the length of the data element, the representation (integer, binary, text, ...) and the info if it is read-only or can be written.

From the info available in those files I could build the vcontrold command files I need for my device, based on info from a verified source. 

Hint: get started by looking at the ecnDataPointType.xml file, looking for your device. In there, I found a list of all addresses in text format (name~address) for each device. The remaining info needs to be assembled from the other files.

Hope this helps. 
Philippe.

mikenussknacker, 2018-06-29:
Dear Philippe

thank you so much for your answer. This is exactly what you need to get all data points with description of the values.

Additionally to your info here some of my findings:

- Geraeteliste.html can be used for some textual description of device and the official device name

- more precisely you can look into ecnDataPointType.xml
   there you find between identification-tag the technical id of the device,           
  
                   eg.<Identification>20C2</Identification>.

   the official name can be found in this secton, too: <ID>VDensHO1_4</ID> (some lines above)
   the most interesting tag is <EventTypeList> .... </EventTypeList> in this section: 
    This shows *all* datapoints for the device in a very long, semi-colon separated list, that are supported, e.g.

       ....;BauartWarmwasser~0x8851;Bedien_WW_Solltemperatur~0x6300;....

Before the ~ there is the technical name of the datapoint, after the ~ the techn. address. (But this is not unique, sometime it references the same address. Which bit to use is described in the follwing table)

Now you can look into ecnEventType.xml:
For each technical datapoint name, Viessmann calls it eventtype, just search for it, there is some quite good readable textual information about the usage, valid values, datatype and if it can be read or if it can be written.

I have tried some of these values - yeah, it works perfectly. Now the work begins :-)

Best regards

Michael