openv/openv #415: Adresses for 204B / ID CU401B_G

Kalmetic, 2018-07-05:
Dear community,

I try to get and set values but I am unable to identify the correct adresses.
Requesting the type via telnet gives me : 204B. (Request via 

After having downloaded the Vitosoft software I found the ID CU401B_G in the file geraeteliste.html. But I was unable to find nether the 204B nor the CU401B_G in any other file.

Trying different code I found:
- Outsidetemperatur = 0101
- Pufferspeichertemperatur oben = 010B
- water temperature = 010D

Do you gyus have any tips for me how to get more adresses?

thanks so far.
Kalmetic

Kalmetic, 2018-07-19:
Can anybody help me with that?

Thanks
Kalmetic

bertmelis, 2018-07-19:
What is the commercial typename of your boiler and controller? (eg vitodens 200 with vitotronic 200)

Kalmetic, 2018-07-19:
Hello
the Commercial Name is: vitocal 300-g with vitotronic 200 WO1C.

I installed the vitosoft Software and it says vitocal xxx-g with vitotronic 200.

Kalmetic, 2018-07-19:
found some protocol information in vitosoft. will keep you informed!

ynghill, 2018-11-26:
Hallo
Ich spreche kein Deutsch, also habe ich Google als Übersetzer benutzt.
Ich protokolliere meine Daten über ein optolink rpi in eine mysql-Datenbank, von der ich schnell Daten zu meiner Android-App sowie zu PC zur Temperaturkurvenanalyse abrufen kann.
Ich habe eine Wärmepumpe VT 200 WO1C mit Systemtyp 204b.
Mit Adresse 0x088E kann ich Datum und Uhrzeit nicht abrufen.
Eine Antwort kann wie folgt aussehen: 018201a600210045.
Die Antwort ändert sich mit der nächsten Frage, also passiert etwas.
Hat jemand eine andere Adresse für Datum und Uhrzeit?
Sucht auch nach der Adresse des Umlaufpumpens und des Kompressordrucks.
Wenn Sie die Softwareversion lesen können, ist es sinnvoll, diese in Verbindung mit den Adressen verschiedener Geräte einzugeben.
Mit freundlichen Grüßen
Yngve

ynghill, 2018-11-26:
Meine Wärmepumpe ist eine Vitocal 300-G mit Vitotronic 200 (Typ WO1C)
Yngve

ynghill, 2018-11-26:
Mina adresser
Yngve

0x00F8 | 204b | Version
-- | -- | --
0x0101 | 0.0 | Ute
0x0103 | 7.6 | Prim. fram
0x0104 | 4.9 | Prim. retur
0x0105 | 37.2 | Sek. fram
0x0106 | 36.7 | Sek. retur
0x0108 | 72.0 | Hetgas
0x010B | 36.1 | Acctank
0x010D | 50.2 | Varmvatten
0x0114 | 32.5 | Temp
0x048d | På | Pump HK1
0x048E | På | Pump HK2
0x0490 | Av | VVC pump
0x0494 | Värme | Värme/VV
0x0500 | 698 | Ant starter
0x0580 | 216.32 | Drifttid komp.
0x088E | 01ec021500250043 | Vito tid
0x1680 | 4.0 | COP total
0x1681 | 4.7 | COP värme
0x1682 | 3.8 | COP varmvatten
0x1800 | 39.5 | Börvärde
0x1801 | 32.2 | Börvärde
0x2000 | 21.0 | Rums borv.
0x2001 | 16.0 | Rum red, bv.
0x2006 | 5.0 | Värmekurva offset
0x2007 | 0.5 | Värmekurva
0x2022 | 20.0 | Party bv.
0x3000 | 21.0 | Rum bv.
0x3001 | 16.0 | Rum red, bv.
0x3006 | 2.0 | Värmekurva offset
0x3007 | 0.4 | Värmekurva
0x3022 | 20.0 | Party bv.
0x6000 | 50.0 | Börvärde
0xB000 | 02 | Driftsätt
0xB001 | 02 | Driftsätt
0xB440 | Av | Primärpump
0xb441 | På | Sekundärpump värme
0xb442 | Av | Sekundärpump varmv.
0xB443 | Av | Kompressor

philverh, 2018-11-26:
@ynghill I have a Vitocal 300-G, but it identifies itself as system type 2046 (not 204B as you said above). 
The adresses of the different data elements are different for almost each device type in the Viessmann range. I used a small software called VitoIdent which talks to the Viessmann devide and tells you the system type it reports.

As an example, I extracted the following addresses for my device: (It is a CSV file which I renamed to TXT to be able to attach it here.) The format is "field;address".
[Datapoints VBC700_BW_WW 2046.txt](https://github.com/openv/openv/files/2615085/Datapoints.VBC700_BW_WW.2046.txt)

Of course each field needs to be qualified by a type, length, value range, if it can be written or not, ... but you need to make 100% sure you're working with the right system type first.