openv/openv #416: Node-RED als Front-End

Werner-G, 2018-07-14:
Ich habe eine funktionierende Anbindung Optolink - RasPi - FHEM - MQTT - Node-RED. Noch vor der Heizperiode würde ich gern ganz auf FHEM verzichten. Auch OpenHAB möchte ich nicht deswegen installieren. Kann mir hier bitte jemand erklären, welche Software dafür nötig ist? Idealerweise würde ich die Heizung über einen ESP8266-Chip auslesen.

bertmelis, 2018-07-15:
https://github.com/bertmelis/VitoWifi

Werner-G, 2018-07-16:
Hi Bert, this sounds good. I understand I can use a WEMOS D1mini and load and set everything through the Arduino- tool, right? For testing, I could use the FTDI-inputs of my USB-Optolink-adapter in parallel or is this already 5V-level?
I am not in a hurry in the moment - when do you think you will release the major update?

bertmelis, 2018-07-16:
Release will be somewhere this week. But I actually only did a refactor. I didn't add any features altough it'll be easier to add "custom" data point types.

My hardware setup is like this: https://github.com/openv/openv/wiki/Bauanleitung-ESP8266. So the transistor and LED are driven by the 3.3V, directly from the esp8266's pins. The pins are the ones from the onboard UART (labelled rx and tx). So you cannot use the USB Serial interface while connected to the boiler. Other way around, you must (physically) disconnect from the boiler when flashing through USB.
ESP32 is the same but it has 3 UART interfaces and you can choose the pins. So there you can keep your USB Serial connected for flashing/communication.

So, while I had a lot of pleasure from my USB-Optolink-adapter, I don't use it anymore.

Werner-G, 2018-08-21:
Hi Bert, yesterday I made my first try with an D1mini, connected to IR-diode and -transistor - but failed. I compiled your basic.ino, and expected to see something in my mqtt-spy. Do I also have to include a MQTT-lib / WIFI-lib? Sorry, I am new in the Arduino-world. Can you please give me a kind of 'setup for dummys' please?

On the other hand, my raspberry is just 1m away from my Viessmann boiler and connected through the USB-optolink-adapter. For sure there is a more simple way to generate MQTT out of the datapoints. 

But indeed, I would prefer the wireless approach.

bertmelis, 2018-08-21:
I'll add an example that includes WiFi and MQTT. See https://github.com/bertmelis/VitoWiFi/issues/45