openv/openv #417: Speicherzugriffsfehler

Micky14, 2018-07-19:
Hallo,
für unsere neue Viessmann Heizung probiere ich mich auch gerade an vcontrold aus.
Bin nach der Anleitung https://openv.wikispaces.com/vcontrold+mit+Raspberry+Pi vorgegangen bis zum Punkt "6. Service starten (inkl. automatischer Start)". Mit "vcontrold" kann ich der serrvice starten und ich kann mich auch mit Telnet an meinem Server/Port anmelden. vctrld wird dabei geöffnet und ich erhalte unter Telnet mit "Version" folgende Ausgabe: "Version: 0.98.2_IPv6". Bei Eingabe von "getTempA" bei einer Telnet-Verbindung stürzt diese ab (wird geschlossen). Bei anderen Eingaben unter vclient wie z.B. "help" kommt eine entsprechende Antwort.

Bei Eingabe "vclient -c getTempA" in der Konsole (putty) erhalte ich die Fehllermeldung "Speicherzugriffsfehler".  

Kann dies am Opto-Kabel liegen oder eventuell am Raspi? In einem anderen Forum habe ich etwas von fehlerhaften Kernel-Modul bezüglich des Opto-Kabels gelesen. War aber kein Raspi.

Auch bezüglich der Geräte-ID bin ich mir nicht sicher, welche ich verwenden soll, Habe bereits mehrere ausprobiert (z.B. 2098, 20CB u. 209F) - finde leider unter http://openv.wikispaces.com/Adressen keine korrekte Adresse für meine Hardware. Kann ich die irgendwie über das Panel der Regelung ermitteln?

Meine Hardware: 
Optokabel von Vitoconnect 100 (jungfräulich aus Verpackung genommen)
Raspi Pi 1 Model B
Software Pi: Aktuelle stretch-lite 
Heizung: Vitocrossal 300 mit V'itotronic 200 KW6B (Bj 2018)

Viele Grüße,

Micky14

bertmelis, 2018-07-19:
if you'd have Windows laptop or similar you can try to use https://github.com/openv/openv/wiki/voIdent to find out your device id.

Micky14, 2018-07-20:
Thanks bertmelis,
i found my device:
Device       VScotHO1
ID           20CB
ExtID        0351
Protocol     300
Description  Gas-Brennwertkessel mit witterungsgeführter Kessel- und Heizkreisregelung für 2 Heizkreise mit Mischer und 1 Heizkreis ohne Mischer

Habe es nun mit den beiden Dateien vito.xml und vcontroldxml aus dem Verzeichnis xml_p300 versucht (dabei die Adresse <allow ip='192.168.0.0/24'/> für mein Netzwerk angepasst) und ins Verzeichnis /etc/vcontrold/ übertragen,

Dennoch gleicher Fehler wie zuvor:
Telnet meldet sich mit vctrld, Version lässt sich ausgeben, bei "getTempA" bricht Telnet ab.
Bei Consoleneingabe:
root@raspberrypi:/etc/init.d# vclient -h 127.0.0.1:3002 -c getTempA
erhalte ich Kommunikationsfehler:
[549] Fri Jul 20 08:00:37 2018 : exit mit count=0
[549] Fri Jul 20 08:00:37 2018 : Fehler bei der Server Kommunikation

bei 
root@raspberrypi:/etc/init.d# vclient -c getTempA
Speicherzugriffsfehler

Viele Grüße
Micky14

Micky14, 2018-07-21:
Habe es nun geschafft, dass ich mit Telnet auf vctrld zugreifen kann und auch die Werte der Anlage auslesen kann.
In der kopierten vcontrold.xml hatte ich die Schnittstelle "/dev/ttyUSB0" nicht korrekt eingetragen :-(

Das direkte Aufrufen des vclient liefert immer noch Speicherzugriffsfehler - habe aber keine Ahnung, ob der Befehl überhaupt funktionieren müsste (hatte ich so in einem anderen Post gelesen):
root@raspberrypi:/etc/init.d# vclient -c getTempA 
Speicherzugriffsfehler

Wichtiger wäre nun, dass der vcontrold-Server beim hochfahren automatisch startet. Funktionniert bei mir nur, wenn ich "vcontrold" eingebe:
root@raspberrypi:~# vcontrold

Danach habe ich über Telnet zugriff und kann dort die Werte auslesen.

Das Startup script /etc/init.d/vcontrol habe ich ertellt (lässt sich mit nano /etc/init.d/vcontrol auch anzeigen/ändern), 

danach:
cd /etc/init.d/
sudo chmod u+x vcontrol
sudo update-rc.d vcontrol start 99 2 3 4 5 . stop 99 0 1 6 .

Beim letzten Befehl erhalte ich folgende Meldung:
update-rc.d: warning: start and stop actions are no longer supported; falling back to defaults

Die kann ich nicht beurteilen. Beim Aufruf von
sudo /etc/init.d/vcontrol start
bekomme ich dann folgende Fehlermeldung:
sudo: unable to execute /etc/init.d/vcontrol: No such file or directory

Der Server startet nur, wenn ich manuell "vcontrold" eingebe.

Viele Grüße,
Micky14

createch2, 2018-10-18:
ich habe exakt das gleiche Problem, leider erst zu spät gesehen und ein eigenes Issue aufgemacht. Gabs hier mittlerweile schon eine Lösung?

Micky14, 2018-10-18:
Ich habe für den Autostart mir mittlerweile ein eigenes Script geschrieben - damit funktioniert der Aufruf von vcontrold bei Stromausfall oder reboot problemlos.
Im Netz hatte ich zuvor folgende Anleitung gefunden:
1.	Erstellen des vcontrol Start-Script mit folgendem Inhalt und dieses unter /etc/init.d speichern. 
2.	In das Verzeichnis wechseln in dem das Script openhab liegt
cd /etc/init.d
3.	Die Datei ausführbar machen
sudo chmod a+x vcontrol
4.	Gruppe und Eigentümer der Datei ändern
sudo chgrp root vcontrol
sudo chown root vcontrol
5.	Überprüfen der Änderungen ls -l Die Ausgabe müsste wie folgt aussehen: -rwxr-xr-x 1 root root 1757 Apr 13 23:27 /etc/init.d/vcontrol
 
6.	Nachdem das Script ausführbar ist, diesen noch in die Runnnlevel eintragen
sudo update-rc.d vcontrol defaults
7.	Nun kann  das script mit dem Befehl
sudo /etc/init.d/vcontrol start
gestartet, bez mit
sudo /etc/init.d/vcontrol stop
gestoppt werden.

Hier mein Script vcontrol (beim Erstellen im Notpad++ war irgendwie die Kodierung UTF-8 erforderlich):

#! /bin/sh
### BEGIN INIT INFO
# Provides: vcontrold
# Required-Start: $syslog
# Required-Stop: $syslog
# Default-Start: 2 3 4 5
# Default-Stop: 0 1 6
# Short-Description: vcontrold server
# Description:
### END INIT INFO
 
case "$1" in
    start)
        echo "vcontrold wird gestartet"
        # Starte Programm
        /usr/local/sbin/vcontrold
        ;;
    stop)
        echo "vcontrold wird beendet"
        # Beende Programm
        killall vcontrold
        ;;
    *)
        echo "Benutzt: /etc/init.d/vcontrold {start|stop}"
        exit 1
        ;;
esac
 
exit 0

Micky14, 2018-10-18:
Ich habe für den Autostart mir mittlerweile ein eigenes Script geschrieben - damit funktioniert der Aufruf von vcontrold bei Stromausfall oder reboot problemlos.
Im Netz hatte ich zuvor folgende Anleitung gefunden:
1.	Erstellen des vcontrol Start-Script mit folgendem Inhalt und dieses unter /etc/init.d speichern. 
2.	In das Verzeichnis wechseln in dem das Script openhab liegt
cd /etc/init.d
3.	Die Datei ausführbar machen
sudo chmod a+x vcontrol
4.	Gruppe und Eigentümer der Datei ändern
sudo chgrp root vcontrol
sudo chown root vcontrol
5.	Überprüfen der Änderungen ls -l Die Ausgabe müsste wie folgt aussehen: -rwxr-xr-x 1 root root 1757 Apr 13 23:27 /etc/init.d/vcontrol
 
6.	Nachdem das Script ausführbar ist, diesen noch in die Runnnlevel eintragen
sudo update-rc.d vcontrol defaults
7.	Nun kann  das script mit dem Befehl
sudo /etc/init.d/vcontrol start
gestartet, bez mit
sudo /etc/init.d/vcontrol stop
gestoppt werden.

Hier mein Script vcontrol (beim Erstellen im Notpad++ war irgendwie die Kodierung UTF-8 erforderlich):

case "$1" in
    start)
        echo "vcontrold wird gestartet"
        # Starte Programm
        /usr/local/sbin/vcontrold
        ;;
    stop)
        echo "vcontrold wird beendet"
        # Beende Programm
        killall vcontrold
        ;;
    *)
        echo "Benutzt: /etc/init.d/vcontrold {start|stop}"
        exit 1
        ;;
esac
 
exit 0

derfritz1, 2020-10-21:
Und ließ sich das Problem mit dem Speicherzugriffsfehler beheben? Habe genau das gleiche. Telnet-ABfrage funktioniert problemlos schon immer. Vcontrold startet auch ohne Probleme...

Danke

Micky14, 2020-10-22:
Nein, bei EIngabe von
root@raspberrypi:/etc/init.d# vclient -c getTempA
bekomme ich immer noch Speicherzugriffsfehler. Habe oben ja geschrieben, dass ich mir mit dem Befehl sowieso nicht sicher war.
Bei Anmeldung über Telnet "telnet localhost 3002" bekomme ich den Kommando-Prompt vctrld> Hier ging mal der Befehl getTempA - bei einem heutigen Test aber nicht mehr - habe mittlerweile ja aber auch eine andere vito-Datei.

Mit commands werden nun die gültigen Befehle angezeigt:

**vctrld>commands** 

getVitoBetriebsart: Betriebsart
setVitoBetriebsart: Setze Betriebsart
getVitoBetriebParty: Partybetrieb
getVitoBetriebParty2330: Partybetrieb_2330
setVitoBetriebParty: Setze Partybetrieb
setVitoBetriebParty2330: Setze Partybetrieb_2330
getVitoTempPartySoll: Solltemperatur Partybetrieb in °C
setVitoTempPartySoll: Setze Warmwassersolltemperatur Partybetrieb in °C
getVitoBetriebFerien: Ferienbetrieb
getVitoBeginnFerien: Abreisetag
setVitoBeginnFerien: Setze Abreisetag
getVitoEndeFerien: Rückreisetag
setVitoEndeFerien: Setze Rückreisetag
getVitoBetriebSpar: Sparbetrieb
setVitoBetriebSpar: Setze Sparbetrieb
getVitoTempRaumNorSoll: Raumsolltemperatur Normal in °C
setVitoTempRaumNorSoll: Setze Raumsolltemperatur Normal in °C
getVitoTempRaumRedSoll: Raumsolltemperatur reduzierter Betrieb in °C
setVitoTempRaumRedSoll: Setze Raumsolltemperatur reduzierter Betrieb in °C
getVitoTempAussen: Außentemperatur in °C
getVitoTempWWIst: Warmwassertemperatur in °C
getVitoTempWWSoll: Warmwassersolltemperatur in °C
setVitoTempWWSoll: Setze Warmwassersolltemperatur in °C
getVitoTempSpeicher: Speichertemperatur in °C
getVitoTempWWAuslauf: Auslauftemperatur Warmwasser in °C
getVitoStatusPumpeSpeicher: Status Speicherladepumpe
getVitoTempKesselIst: Vorlauf- bzw. Kesseltemperatur in °C
getVitoTempKesselSoll: Kesselsolltemperatur in °C
getVitoStatusFlamme: Flammenstatus
getVitoLaufzeitBrenner: Brenner Betriebsstunden
getVitoStartsBrenner: Brennerstarts
getVitoStatusPumpeIntern: Status interne Pumpe
getVitoDrehzahlPumpeIntern: Drehzahl interne Pumpe
getVitoKennlinieNeigung: Neigung Heizkennlinie
setVitoKennlinieNeigung: Setze Neigung Heizkennlinie
getVitoKennlinieNiveau: Niveau Heizkennlinie
setVitoKennlinieNiveau: Setze Niveau Heizkennlinie
getVitoBetriebsartHK: Aktuelle Betriebsart des Heizkreises
getVitoTempVLSoll: Vorlaufsolltemperatur in °C
getVitoTempRLIst: Rücklauftemperatur in °C
getVitoStatusPumpeHK: Status Heizkreispumpe
getVitoTempRaumHK: Heizkreis Raumtemperatur in °C
getVitoVolStromHK: Volumenstrom des Heizkreises in Liter
getVitoStatusPumpeZirku: Status Zirkulationspumpe
getVitoTempSolarkol: Solar-Kollektortemperatur in °C
getVitoTempSolarspeicher: Solar-Speichertemperatur in °C
getVitoLaufzeitSolar: Solar-Betriebsstunden
getVitoLeistungSolar: Solar-Wärmeenergie
getVitoErtragTagSolar: Solarertrag aktueller Tag
getVitoBetriebStatusSolar: Solar-Status Pumpe
getVitoStatusUmschaltventil: Status Umschaltventil Warmwasser/Heizen
getVitoTimerMoHeizen: Schaltzeit Montag
setVitoTimerMoHeizen: Setze Schaltzeit Montag
getVitoTimerDiHeizen: Schaltzeit Dienstag
setVitoTimerDiHeizen: Setze Schaltzeit Dienstag
getVitoTimerMiHeizen: Schaltzeit Mittwoch
setVitoTimerMiHeizen: Setze Schaltzeit Mittwoch
getVitoTimerDoHeizen: Schaltzeit Donnerstag
setVitoTimerDoHeizen: Setze Schaltzeit Donnerstag
getVitoTimerFrHeizen: Schaltzeit Freitag
setVitoTimerFrHeizen: Setze Schaltzeit Freitag
getVitoTimerSaHeizen: Schaltzeit Samstag
setVitoTimerSaHeizen: Setze Schaltzeit Samstag
getVitoTimerSoHeizen: Schaltzeit Sonntag
setVitoTimerSoHeizen: Setze Schaltzeit Sonntag
getVitoTimerMoWW: Schaltzeit Warmwasser Montag
setVitoTimerMoWW: Setze Schaltzeit Warmwasser Montag
getVitoTimerDiWW: Schaltzeit Warmwasser Dienstag
setVitoTimerDiWW: Setze Schaltzeit Warmwasser Dienstag
getVitoTimerMiWW: Schaltzeit Warmwasser Mittwoch
setVitoTimerMiWW: Setze Schaltzeit Warmwasser Mittwoch
getVitoTimerDoWW: Schaltzeit Warmwasser Donnerstag
setVitoTimerDoWW: Setze Schaltzeit Warmwasser Donnerstag
getVitoTimerFrWW: Schaltzeit Warmwasser Freitag
setVitoTimerFrWW: Setze Schaltzeit Warmwasser Freitag
getVitoTimerSaWW: Schaltzeit Warmwasser Samstag
setVitoTimerSaWW: Setze Schaltzeit Warmwasser Samstag
getVitoTimerSoWW: Schaltzeit Warmwasser Sonntag
setVitoTimerSoWW: Setze Schaltzeit Warmwasser Sonntag
getVitoTimerMoPumpeZirku: Schaltzeit Zirkulationspumpe Montag
setVitoTimerMoPumpeZirku: Setze Schaltzeit Zirkulationspumpe Montag
getVitoTimerDiPumpeZirku: Schaltzeit Zirkulationspumpe Dienstag
setVitoTimerDiPumpeZirku: Setze Schaltzeit Zirkulationspumpe Dienstag
getVitoTimerMiPumpeZirku: Schaltzeit Zirkulationspumpe Mittwoch
setVitoTimerMiPumpeZirku: Setze Schaltzeit Zirkulationspumpe Mittwoch
getVitoTimerDoPumpeZirku: Schaltzeit Zirkulationspumpe Donnerstag
setVitoTimerDoPumpeZirku: Setze Schaltzeit Zirkulationspumpe Donnerstag
getVitoTimerFrPumpeZirku: Schaltzeit Zirkulationspumpe Freitag
setVitoTimerFrPumpeZirku: Setze Schaltzeit Zirkulationspumpe Freitag
getVitoTimerSaPumpeZirku: Schaltzeit Zirkulationspumpe Samstag
setVitoTimerSaPumpeZirku: Setze Schaltzeit Zirkulationspumpe Samstag
getVitoTimerSoPumpeZirku: Schaltzeit Zirkulationspumpe Sonntag
setVitoTimerSoPumpeZirku: Setze Schaltzeit Zirkulationspumpe Sonntag
getVitoStatusStoerung: Status Störung
getVitoStoerung1: Störung Meldung 1
getVitoStoerung2: Störung Meldung 2
getVitoStoerung3: Störung Meldung 3
getVitoStoerung4: Störung Meldung 4
getVitoStoerung5: Störung Meldung 5
getVitoStoerung6: Störung Meldung 6
getVitoStoerung7: Störung Meldung 7
getVitoStoerung8: Störung Meldung 8
getVitoStoerung9: Störung Meldung 9
getVitoStoerung10: Störung Meldung 10
getVitoSystemzeit: Systemzeit
setVitoSystemzeit: Setze Systemzeit
getVitoAnlagenschema: Anlagenschema

Vcontrol läuft nun seit 2 Jahren problemlos. Verwende vcontrol mit entsprechendem Adapter bei meiner ioBroker-Visualisierung. Heizungssteuerung und Anzeige der Heizungswerte sind fehlerfrei.

hmueller01, 2020-10-23:
Hallo @derfritz1 @Micky14 ,

vclient sollte ohne Speicherfehler gehen, wenn ihr
```
> vclient -c getTempA -h localhost
```
verwendet. Ich habe da gerade noch mal ein [Update](https://github.com/openv/vcontrold/pull/73) gemacht.
Bitte mal diese "latest" git Version testen. Dann sollte auch
``` 
> vclient -c getTempA
```
ohne Fehler gehen.

PS: Fehler am vcontrold sollten unter [openv/vcontrold](https://github.com/openv/vcontrold) gemeldet werden, nicht hier im "Anwendungsbereich".