openv/openv #418: Vitotronic 150 Typ KB1 Protokoll: KW ID:2092 // Hilfe bei vito.xml

Knallix, 2018-08-26:
Hallo,

wie muss meine vito.xml aussehen ?
Kann mir jemanden einen Ansatz nennen ?

Ich verbinden mich mit mit Optolink mit FT232RL "
Dieser ist an einem Raspberry angeschlossen wo vcontrold drauf läuft.
Es besteht auch schon eine Verbindung zu IObroker.
Allerdings ist die ID in der XML ja falsch. Wie muss ich da vorgehen ?

Knallix, 2018-09-27:
Kann mir keiner Helfen ? oder einen Ansatz liefern ?

bertmelis, 2018-09-27:
What is the type of your heater?

Knallix, 2018-09-28:
I need the binary code. so that I can create the vito.xml.
To see here:
`<command name = 'getDevType' protocmd = 'getaddr'>
<Addr> 00F8 </ addr>
<Len> 2 </ len>
                <Unit> DT </ unit>
                <description> Determine Device Type of Attachment </ description>
                <device ID = "2053" protocmd = 'getvaddr'>
                    <Addr> F8 </ addr>
                    <Len> 4 </ len>
<error> 05 05 05 05 </ error>
                </ Device>
        </ Command>`

Knallix, 2018-12-10:
Help Please :-(

SuperNils, 2022-08-12:
@Knallix Hattest du Erfolg? 

Ich habe auch eine Vitotronic 150. Ist es möglich mit vcontrold eine Verbindung zur Vitotronic 150 herzustellen?

Knallix, 2022-08-12:
Hallo, nein leider nicht .
Hatte Verbindung nur zum Laptop und dort auch die Befehle aus einer Text Datei, nur leider nie richtig zum Ziel gekommen .
Leider hatte keiner auf meine Frage eine Antwort .

steini86, 2022-11-18:
Habe auch eine Vitotronic 150 und konnte einige Werte auslesen.
Es gab wohl früher eine Sammlung von vito.xml Dateien, aber der Server ist wohl offline. Meine vito.xml als Vorlage habe ich von hier: https://homematic-forum.de/forum/viewtopic.php?t=18926#p173085
Genauso wie die vcontrold.xml. Muss natürlich angepasst werden (ID auf 2092, IPs, etc..). 
Dinge wie Außen-, Warmwasser und Kesseltemperatur geht. Einige Dinge liefern immer 20° zurück, da ist wohl kein Sensor verbaut (Vor- und Rücklauftemperatur z.B.?)

steini86, 2022-11-18:
Bisher lese ich nur Werte aus, an das Setzen habe ich mich noch nicht gemacht. Das meiste fände ich aber auch nicht interessant/nützlich.
Gerade habe ich testweise einige Befehle versucht, und die Warmwassersolltemperatur funktioniert mit der aktuellen vito.xml schonmal nicht. Interessant wäre vielleicht die Timer zu verstellen, das finde ich immer sehr Mühsam am Gerät. Aber das schaut so auch recht Mühsam aus. Ansonsten gäbe es Befehle für zum Beispiel Zeitkonstanten, die für die Aussentemperatur konnte ich verstellen, grundsätzlich kann man also Dinge schreiben. Aber da will ich eigentlich nichts verstellen.. Dinge wie eine Zirkulationspumpe, die man setzen könnte habe ich nicht.

SuperNils, 2022-11-18:
Kannst du die Soll-Raumtemperatur verstellen? Anwendungsfall: Wenn keiner Zuhause ist, dann Temperatur etwas runter.

steini86, 2022-11-19:
Bei Warmwasser stimmt wohl der command nicht. Raumsolltemperatur kann ich ändern, mir ist nur noch nicht ganz klar was die verschiedenen Werte (M1/M2/M3) machen.
```
vctrld>getTempRaumNorSollM1
20.000000 Grad Celsius
vctrld>setTempRaumNorSollM1 21
OK
vctrld>getTempRaumNorSollM1
21.000000 Grad Celsius
vctrld>getTempRaumNorSollM2
20.000000 Grad Celsius

vctrld>getTempRaumRedSollM1
15.000000 Grad Celsius
vctrld>setTempRaumRedSollM1 16
OK
vctrld>getTempRaumRedSollM1
16.000000 Grad Celsius
```

steini86, 2022-12-08:
Weil ich verschiedene Dinge zusammenkopiert habe sehen die Dateien grausam aus und enthalten nicht funktionierende commads. Ich wollte sie vor Veröffentlichung eigentlich aufräumen, aber bevor du garnichts bekommst hier zumindest startende Dateien zum Anfangen. Falls du da Zeit investierst, teile sie gerne wieder ;-)
Mindestens Zeile 12 in vcontrold.xml (serial port) musst du anpassen.
Weil ich hier keine xml Dateien anhängen darf musst das das ".txt" am Ende noch entfernen.
[vito.xml.txt](https://github.com/openv/openv/files/10185908/vito.xml.txt)
[vcontrold.xml.txt](https://github.com/openv/openv/files/10185910/vcontrold.xml.txt)

Knallix, 2022-12-08:
> Weil ich verschiedene Dinge zusammenkopiert habe sehen die Dateien grausam aus und enthalten nicht funktionierende commads. Ich wollte sie vor Veröffentlichung eigentlich aufräumen, aber bevor du [...]

DANKE ! DANKE ! Läuft erstmal. :-)
Jetzt muss ich noch den Dienst Automatisch starten, dann ist alles perfekt.

Raudi1, 2023-02-23:
Funktioniert grundsätzlich auch bei meiner Calotronic 150. Musste allerdings bei setTempWWsoll in der vito.xml len=1 ändern, damit der entsprechende Befehl funktioniert. Manche Befehle, wie z.B. die Sparmodusbefehle liefern auch nur Fehlermeldungen zurück. Scheint die Steuerung einfach nicht zu können. Werde noch weiter alles ansehen, damit ich dann die Heizung effizienter gestalten kann.

SuperNils, 2023-03-19:
@steini86 
Danke für das Bereitstellen der xml Dateien. In Kombination mit iobroker war die gesamte Installation in kurzer Zeit eingerichtet und lauffähig.

SuperNils, 2023-03-23:
Jetzt wäre es noch cool die Setzbefehle setTempWWsoll und setTimerWW bei der Vitotronic 150 funktionsfähig zu bekommen. Falls jemand dazu Hinweise hat, bitte hier teilen.

Raudi1, 2023-03-23:
Wie schon gesagt, musste ich bei meiner Calotronic150 unter setTempWWsoll in der vito.xml len=1 setzen. Wenn der get Befehl klappt, dann einfach versuchen beim set Befehl die gleiche len zu nehmen war da mein Ansatz, der auch funktioniert hat.

SuperNils, 2023-03-25:
@Raudi1 Es funktioniert tatsächlich. Danke.

Außerdem ist es möglich die Warmwassersolltemperatur zu setzen. Dazu einfach die entsprechenden setTimmerWW-Befehle zur Vito.xml hinzufügen.
Jetzt muss ich nicht dauert in den Keller laufen :)

SuperNils, 2023-07-13:
Hat jemand es geschafft die Neigung und das Niveau auszulesen?
Folgender Eintrag hat ledier nicht funktioniert:
```
<command name="getNeigungM1" protocmd="getaddr">
<addr>2305</addr> 
<len>1</len> 
<unit>UN</unit> 
<description>Ermittle Neigung Heizkennlinie M1</description> 
</command>
<command name="getNiveauM1" protocmd="getaddr">
<addr>2304</addr> 
<len>1</len> 
<unit>ST</unit> 
<description>Ermittle Niveau Heizkennlinie M1</description> 
</command>
```
@Raudi1, @Knallix, @steini86

bp112, 2024-06-16:
Hallo zusammen :)

Bei meiner Calotronic150 (Vitotronic 150 Typ KB1 Protokoll: KW ID:2092) funktioniern die beiden .XML dateien Super.

Hat vieleicht jmd schon eine aktuelle Version geschrieben und möchte sie hier Teilen ?

Vielen Danke, ich bin aufjedenfall schon ein ganzes Stück weiter.

Raudi1, 2024-06-16:
> Hallo zusammen :)
> 
> Bei meiner Calotronic150 (Vitotronic 150 Typ KB1 Protokoll: KW ID:2092) funktioniern die beiden .XML dateien Super.
> 
> Hat vieleicht jmd schon eine aktuelle Version geschrieben und möchte sie hier Teilen ?
> 
> Vielen Danke, ich bin aufjedenfall schon ein ganzes Stück weiter.
> 
> @Raudi1, @Knallix, @steini86 @SuperNils

Hier meine aktuelle config. Ist aus verschiedenen Quellen zusammengewürfelt. Habe auch selber mit ein paar Sachen experimentiert. Größtenteils einfach get Befehle in set Befehle umgewandelt. Habe auch nicht alles getestet und wenn ich mich richtig erinnere funktioniert auch nicht alles. Ich setze die Betriebsart, um den Brenner basierend auf den Thermostaten an und aus zu machen(WW/H+WW) und stelle wenn ich duschen will (basierend auf dem Kalender oder manuell) die WW Temperatur hoch. Die Raumtemperatur lässt sich zwar setzen, kann aber nicht benutzt werden, da diese sich auf den eingestellten Wert vom Drehregler zurück stellt. Lese außerdem aktuell über das Vcontrol Home Assistant Addon folgende Werte aus: 
 - getStatusStoerung:FLOAT
  - getTempWWist:FLOAT
  - getTempWWsoll:FLOAT
  - getTempKist:FLOAT
  - getTempKsoll:FLOAT
  - getBetriebArt:STRING`
```

<vito>
<devices>
	<device ID="2092" name="V150 KB1" protocol="KW2" /> 
</devices>
<commands>
<command name="getTempA" protocmd="getaddr">
<addr>0800</addr> 
<len>2</len> 
<unit>UT</unit> 
<description>Ermittle die Aussentemperatur in Grad C</description> 
</command>

<command name="getTempAtp" protocmd="getaddr">
<addr>5525</addr> 
<len>2</len> 
<unit>UT</unit> 
<description>Ermittle die Aussentemperatur in Grad C (Tiefpass)</description> 
</command>
<command name="getTempAged" protocmd="getaddr">
<addr>5527</addr> 
<len>2</len> 
<unit>UT</unit> 
<description>Ermittle die Aussentemperatur in Grad C (Gedaempft)</description> 
</command>

<!--  WARMWASSER 
--> 
<command name="getTempWWist" protocmd="getaddr">
<addr>0804</addr> 
<len>2</len> 
<unit>UT</unit> 
<description>Ermittle die Warmwassertemperatur in Grad C</description> 
</command>
<command name="getTempWWsoll" protocmd="getaddr">
<addr>6300</addr> 
<len>1</len> 
<unit>UTI</unit> 
<description>Ermittle die Warmwassersolltemperatur in Grad C</description> 
</command>
<command name="setTempWWsoll" protocmd="setaddr">
<addr>6300</addr> 
<len>1</len> 
<unit>UTI</unit> 
<description>Setze die Warmwassersolltemperatur in Grad C</description> 
</command>

<command name="getSpeichervorrang" protocmd="getaddr">
<addr>27A2</addr> 
<len>1</len> 
<unit>ST</unit> 
<description>Ermittle den Status Speichervorrang Heizkreis ohne Mischer A1 bzw. mit Mischer M1</description> 
[... 1239 lines cut ...]
</command>
<command name="getVentilStatus" protocmd="getpaddr">
<description>Ermittle Zustand Umschaltventil WW/Heizen</description> 
</command>

<!--  Befehl 'getVentilStatus' sprechend analog Viessmann geaendert und ergaenzt von ceteris paribus 2010 
--> 
<command name="getUmschaltventil" protocmd="getaddr">
	<addr>0A10</addr> 
	<len>1</len> 
	<unit>USV</unit> 
	<description>Ermittle den Status des Umschaltventils WW/Heizen</description> 
</command>

<!--  Befehl 'setVentilStatus' sprechend analog Viessmann geaendert und ergaenzt von ceteris paribus 2012 
--> 
<command name="setUmschaltventil" protocmd="setaddr">
	<addr>0A10</addr> 
	<len>1</len> 
	<unit>USV</unit> 
	<description>Setze den Status des Umschaltventils WW/Heizen</description> 
</command>
</commands>
</vito>
```

bp112, 2024-06-16:
Danke Raudi1 für deine Antwort,

```
user@heizung:~ $ vclient getBetriebArt
getBetriebArt:
H+WW FS
user@heizung:~ $ vclient setBetriebArt
[21444] Sun Jun 16 20:03:35 2024 : exit with count=0
[21444] Sun Jun 16 20:03:35 2024 : Error communicating with the server
user@heizung:~ $ vclient setBetriebArt WW
[21446] Sun Jun 16 20:03:46 2024 : exit with count=0
[21446] Sun Jun 16 20:03:46 2024 : Error communicating with the server
user@heizung:~ $

```
Leider bekomme ich das ändern der Betriebsart immer noch nicht hin. 
Wo ist mein Fehler ?

Raudi1, 2024-06-16:
> Danke Raudi1 für deine Antwort,
>
> ```
> user@heizung:~ $ vclient getBetriebArt
> getBetriebArt:
> H+WW FS
> user@heizung:~ $ vclient setBetriebArt
> [21444] Sun Jun 16 20:03:35 2024 : exit with count=0
> [21444] [...]

Benutzt du die Vito.xml von mir?

bp112, 2024-06-16:
Ja, ich hab sie direkt ausprobiert.
 irgendwas geb ich da noch falsch ein

SuperNils, 2024-06-17:
@Raudi1 
> Ich setze die Betriebsart, um den Brenner basierend auf den Thermostaten an und aus zu machen(WW/H+WW)
Das klingt interessant. Könntest du das genauer erklären?

Ich prüfe zwischen 8:00 und 18:00 Uhr stündlich die Außen- und Innentemperatur und die aktuelle/zukünftige Sonneneinstrahlung. Dann setze ich die Heizzeiten entsprechend, d.h. die Heizung läuft dann entsprechend eine oder mehr Stunden im reduzierten Heizbetrieb.

Raudi1, 2024-06-17:
> @Raudi1
>
>
> Das klingt interessant. Könntest du das genauer erklären?
>
> Ich prüfe zwischen 8:00 und 18:00 Uhr stündlich die Außen- und Innentemperatur und die aktuelle/zukünftige Sonneneinstrahlung. [...]

Ich habe im ganzen Haus Thermostate, die ich in Home Assistant eingebunden habe. Wenn dort mindestens ein Thermostat eine gewisse Ventilöffnung aufweist, dann schalte ich die Betriebsart auf H+WW. Wenn dann die gewünschte Raumtemperatur erreicht ist, schließen die Thermostate wieder und die Betriebsart wird auf WW gesetzt. Ist so ziemlich die einfachste Regelung.