openv/openv #420: WO1B, V200WO1,VBC700 - zusätzliche Datenpunkte für die Tabelle

s0170071, 2018-09-09:
ein paar hab ich noch, die fehlen in der Tabelle und stehen nicht in der Anleitung: 
(bei mir Vitocal300+ Vitotronic WO1B)
Quelle: http://docplayer.org/2343818-Viessmann-viessmann-gmbh-u-co-kg-04-2012-1-hex-2-dez-1.html

Betriebsmodus, 0xb000 -  1byte, r/w, 0=aus, 1=ww, 2 = ww+heizen/kühlen
1xWW Bereitung, 0xb020, 1byte, r/w, 0=aus, 1=ein

ElektroEnergie_Warmwasser_12Monate,  0x1670, 4 bytes, factor=0.1 
WaermeEnergie_Warmwasser_12Monate   0x1660, 4 bytes, factor=0.1 
ElektroEnergie_Heizen_12Monate    0x1650, 4 bytes,factor=0.1
WaermeEnergie_Heizen_12Monate   0x1640, 4 bytes,factor=0.1
ElektroEnergie_Warmwasser_1Woche  0x16D0, 2bytes factor=0.1
WaermeEnergie_Warmwasser_1Woche     0x16D2,  2 bytes,  factor=0.1
ElektroEnergie_Heizen_1Woche  0x16D1 2 bytes, factor=0.1
WaermeEnergie_Heizen_1Woche  0x16D3, 2 bytes, factor=0.1 
(die Wochenzähler werden Montags,0:00 Uhr zurückgesetzt, die Jahreszähler sind rollierend)

Die DPs hier findet man noch in der Anleitung "Vitotronic 200 Serviceanleitung für die Fachkraft" Dok. Nr. 5681 896. Die Nummern der Datenpunkte für Optolink sind identisch mit den Nummern der Parameter in der Steuerung / Anleitung.

KuehlkennlinieNiveau  0x7110, R/W, 4 bytes
KuehlkennlinieSteigung  0x7111, R/W, 4 bytes
MinVLTemp_Kuehlung   0x7103, R/W, 4 bytes
TemperaturDifferenz_Kuehlung     0x7004, R/W, 4 bytes
Freigabe_Kuehlung          0x71fe, R/W, 1 bytes

TemperaturDifferenz_Heizung    0x7003, R/W, 4 bytes

ultrazauberer, 2020-03-02:
> Ich habe eine Vitotronic 200 - WO1C mit einer Vitocal 300 A. Leider liefern die Datenpunkte bei mit nur Nullen. Kennt jemand diese Datenpunkte für die WO1C (Typ 204C)?
> 
> ```
> ElektroEnergie_Warmwasser_1Woche 0x16D0, 2bytes factor=0.1
> WaermeEnergie_Warmwasser_1Woche 0x16D2, 2 bytes, factor=0.1
> ElektroEnergie_Heizen_1Woche 0x16D1 2 bytes, factor=0.1
> WaermeEnergie_Heizen_1Woche 0x16D3, 2 bytes, factor=0.1
> ```

Hast du hierfür schon etwas gefunden? Genau diese Werte benötige ich für meine Auswertung. Ich habe ebenfalls eine WO1C.

s0170071, 2022-12-01:
[Hier ](https://github.com/openv/openv/issues/480#issuecomment-1333585921) sind die Datenpunkt für die WO1C. Da gibt es einige Überschneidungen.

s0170071, 2022-12-11:
Die konsolidierte Liste aus https://github.com/openv/openv/wiki/Adressen für die WO1B

<body>

  | V200WO1,VBC700, WO1B |   |  
-- | -- | -- | --
  | Addr | Values Factor | Bytes
Gerätekennung | 0x00F8 | 2046 2047 2049 |  
  |   |   |  
Aussentemperatur (Sensor 1) | 0x0101 | / 10 | 2
Speichertemperatur (Sensor 5) | 0x010B | / 10 | 2
Brauchwassertemperatur oben | 0x010D | / 10 | 2
Brauchwassertemperatur unten | 0x010E | / 10 | 2
Vorlauftemperatur Anlage | 0x010A | / 10 | 2
Vorlauftemperatur M2 | 0x0114 | / 10 | 2
Vorlauftemperatur M3 | 0x0115 | / 10 | 2
Vorlaufsolltemperatur M3 | 0x1004 | / 10 | 2
Kollektortemperatur | 0x0111 | / 10 | 2
Speichertemperatur | 0x0113 | / 10 | 2
Raumtemperatur Soll | 0x2000 | / 10 | 2
Red. Raumtemperatur Soll | 0x2001 | / 10 | 2
Party Raumtemperatur Soll | 0x2022 | / 10 | 2
Heizkennlinie Niveau | 0x2006 | / 10 | 2
Heizkennlinie Steigung | 0x2007 | / 10 | 2
Primärkreis Vorlauftemperatur (Luft) | 0x0103 | / 10 | 2
Primärkreis Rücklauftemperatur (Luft) | 0x0104 | / 10 | 2
Sekundärkreis Vorlauftemperatur (Wasser) | 0x0105 | / 10 | 2
Sekundärkreis Rücklauftemperatur (Wasser) | 0x0106 | / 10 | 2
Druck im Verdampfer | 0x0680 | / 10 | 2
Druck im Kondensator | 0x0682 | / 10 | 2
Temperatur im Verdampfer | 0x0102 | / 10 | 2
Betriebsstunden Verdichter | 0x0580 | Sek | 4
Betriebsstunden Primärquelle | 0x0582 | Sek | 4
Betriebsstunden Sekundärquelle | 0x0584 | Sek | 4
Betriebsstunden Durchlauferh. Stufe1 | 0x0588 | Sek | 4
Betriebsstunden Durchlauferh. Stufe2 | 0x0589 | Sek | 4
Betriebsstunden Heizkreis A1 | 0x058D | Sek | 4
Betriebsstunden Heizkreis M2 | 0x058E | Sek | 4
Betriebsstunden Heizkreis M3 | 0x058F | Sek | 4
Betriebsstunden Zusatzausgang (Z-Pumpe) | 0x0590 | Sek | 4
Betriebsstunden Sammelmeldung | 0x0591 | Sek | 4
Betriebsstunden Umschaltventil Heizen/Warmw. | 0x0594 | Sek | 4
Betriebsstunden Speicherladepumpe | 0x0596 | Sek | 4
Anzahl Einschaltungen Verdichter | 0x0500 |   | 4
Anzahl Einschaltungen Primärquelle | 0x0502 |   | 4
Anzahl Einschaltungen Sekundärquelle | 0x0504 |   | 4
Anzahl Einschaltungen Durchlauferh. Stufe1 | 0x0508 |   | 4
Anzahl Einschaltungen Durchlauferh. Stufe2 | 0x0509 |   | 4
Anzahl Einschaltungen Heizkreis A1 | 0x050D |   | 4
Anzahl Einschaltungen Heizkreis M2 | 0x050E |   | 4
Anzahl Einschaltungen Heizkreis M3 | 0x050F |   | 4
Anzahl Einschaltungen Zusatzausgang (Z-Pumpe) | 0x0510 |   | 4
Anzahl Einschaltungen Sammelmeldung | 0x0511 |   | 4
Anzahl Einschaltungen Umschaltventil Heizen/Warmwasser | 0x0514 |   | 4
Anzahl Einschaltungen Speicherladepumpe | 0x0516 |   | 4
Status Verdichter Ein/Aus | 0x0400 | 0..1 | 1
Status Primärquelle Ein/Aus | 0x0402 | 0..1 | 1
Status Sekundärquelle Ein/Aus | 0x0404 | 0..1 | 1
Status Durchlauferh. Stufe1 Ein/Aus | 0x0408 | 0..1 | 1
Status Durchlauferh. Stufe2 Ein/Aus | 0x0409 | 0..1 | 1
Status Pumpe Heizkreis A1 Ein/Aus | 0x040D | 0..1 | 1
Status Pumpe Heizkreis M2 Ein/Aus | 0x040E | 0..1 | 1
Status Pumpe Heizkreis M3 Ein/Aus | 0x040F | 0..1 | 1
Status Zusatzausgang (Z-Pumpe) Ein/Aus | 0x0410 | 0..1 | 1
Status Sammelmeldung Ein/Aus | 0x0411 | 0..1 | 1
Status Umschaltventil Heizen/Warmwasser Ein/Aus | 0x0414 | 0..1 | 1
Status Speicherladepumpe Ein/Aus | 0x0416 | 0..1 | 1

</body>

</html>

s0170071, 2022-12-11:
Aus https://docplayer.org/221719226-Viessmann-abnahmeprotokoll.html, scheinbar auch mit einigen Ebene2 Parametern: 

<body>

  | 7900 | Freigabe Heizwasser-Durchlauferhitzer
-- | -- | --
  | 7902 | Heizen mit Elektro
  | 2022 | Raumtemperatur im Partybetrieb HeizkreisA1/HK1
  | 2000 | Raumtemperatur Soll Heizkreis A1/HK1
  | 2001 | Red. Raumtemperatur Soll Heizkreis A1/HK1
  |   |  
Bedienbetriebsart Heizkreis A1/HK1 |   |  
  | 2007 | Neigung Heizkennlinie Heizkreis A1/HK1
  | 2006 | Niveau Heizkennlinie Heizkreis A1/HK1
  | 71FE | Freigabe Active Cooling
  | 7111 | Neigung Kühlkennlinie
  | 7110 | Niveau Kühlkennlinie
  |   |  
Warmwasser |   |  
  | 6000 | Warmwassertemperatur-Sollwert 1x WW-Bereitung
  | 6015 | Freigabe Elektroheizung für WW-Bereitung
  | 6009 | Einschaltoptimierung für Warmwasserbereitung
  | 600A | Ausschaltoptimierung für Warmwasserbereitung
  | 600C | Warmwassertemperatur-Sollwert 2
  | 7000 | Anlagenschema
  | 7003 | Temperaturdifferenz für Berechnung der Heizgrenze
  | 7004 | Temperaturdifferenz für Berechnung der Kühlgrenze
  | 7008 | Schwimmbad
  | 700A | Kaskadenansteuerung
  | 700B | Leistung Folge-Wärmepumpe
  | 7010 | Externe Erweiterung
  | 7011 | Anlagenkomponente bei externer Umschaltung
  | 7012 | Betriebsstatus bei externer Umschaltung
  | 7013 | Dauer der externen Umschaltung
  | 7014 | Wirkung externe Anforderung auf Wärmepumpe/Heizkreise
  | 7015 | Wirkung extern Sperren auf Wärmepumpe/Heizkreise
  | 7017 | Vitocom 100
  | 701A | Wirkung extern Sperren auf HKP1
  | 701A | Wirkung extern Sperren auf HKP2
  | 701A | Wirkung extern Sperren auf HKP3
  | 701A | Wirkung extern Sperren auf Pumpen/Verdichter
  | 701A | Wirkung extern Sperren auf SekPumpe
  | 701A | Wirkung extern Sperren auf SLP
  | 701B | Gemeinsamer Vorlaufsensor Anlage
  | 5735 | Anzahl Folge-Wärme-pumpen in Kaskade
  | 7A00 | Typ Solaranlage
  | 5000 | Freigabe Verdichter
  | 5010 | Verdampfertemperatur für Abtauende
  | 5030 | Leistung Verdichterstufe
  | 7B00 | Freigabe Externer Wärmeerzeuger
  | 7B01 | Vorrang externer Wärmeerzeuger/Heizw.-Durchlauferh.
  | 7B02 | Bivalenztemperatur externer Wärmeerzeuger
  | 7B0D | extWE auf Warmwasser
  | 6000 | Warmwassertemperatur-Sollwert
  | 6005 | Min. WW-Temperatur
  | 6006 | Max. WW-Temperatur
  | 6007 | Hysterese WW-Temperatur Wärmepumpe
  | 6008 | Hysterese WW-Temperatur Zusatzheizung
  | 6009 | Einschaltoptimierung für Warmwasserbereitung
  | 600A | Ausschaltoptimierung für Warmwasserbereitung
  | 600C | Warmwassertemperatur-Sollwert 2
  | 600E | Temperatursensor unten im Speicher-Wassererwärmer
  | 6014 | Freigabe Zusatzheizungen für Warmwasserbereitung
  | 6015 | Freigabe Elektroheizung für WW-Bereitung
  | 6016 | Vorrang WW-Bereitung bei Kombispeicher
  | 6017 | Einschaltversuche für WW nach Hochdruckabschaltung
  | 6020 | Betriebsweise Speicherladepumpe
  |   |  
Solar |   |  
  | 7A00 | Typ Solaranlage
[... 313 lines cut ...]
  | 5125 | Verdichter2 Niederdruck Grenzwert
  | 5126 | Verdichter2 Niederdruck Hysterese
  | 511F | Verdichter2 Öffnungsstellung EEV Ventil Startphase
  | 5110 | Verdichter2 Temperatur Abtauende
  | 5119 | Verdichter2 Überhitzungssollwert
  | 5124 | Verdichter2 Überhitzungstemperatur Grenzwert
  | 5100 | Verdichter2 Verdichter Stufe Freigegeben
  | 6007 | Hysterese WW-Temperatur Wärmepumpe
  | 6008 | Hysterese WW-Temperatur Zusatzheizung
  | 600C | Warmwassertemperatur- Sollwert 2
  | 6000 | Warmwassertemperatur-Sollwert
  | 6001 | Zeitprogramm Warmwasserbereitung
  | 600E | Temperatursensor unten im Speicher-Wassererwärmer
  | 7101 | Kühlkreis
  | 7201 | Basisniveau Puffer
  | 7B0D | extWE auf Warmwasser
  | 7B0E | extWE Bivalentbetrieb 20 °C 23 °C
  | 7313 | Hysterese Vorlauf Aus
  | 6016 | Vorrang WW-Bereitung bei Kombispeicher
  | 7208 | Temperaturgrenze Betriebsstatus Festwer für Puffersp.
  | 730D | Freigabe 3-Wege- Umschaltventil Heizen/WWt

</body>

</html>

sude22, 2023-07-13:
Hallo, ich wollte mal fragen ob es jemand geschafft hat für die WO1B die Schaltzeiten für WW und HK1/2 auszulesen und zu schreiben?
Ich habe nur die Adressen ab 0x92E0 (wahrscheinlich WO1C) gefunden. Aber die liefern bei mir nur INT_MAX(4294967295).
Danke im voraus.

ronniebach, 2023-07-26:
Hi 
I have the WO1B

Do you have more information on the operating mode ( betrieb) which should be related to address: 0xB000
I miss information on how to set PARTYMODE / COMFORTMODE ?

Thanks.

s0170071, 2023-07-26:
> 
address: 0xB000
0= Off
1= Water 
2= WW+Heat + Cool
0x42 (decimal 66) =party

ronniebach, 2023-07-26:
Thanks... :-) Where do you have the information from ? Do you have a datasheet/pdf from Viessmann with the address / definitions?

s0170071, 2023-07-26:
google. A lot.  Sometimes it helps if you google known information like e.g. "wo1b 0xb000". This leads you to documents where there are more data points. 
The same goes for the[ expert password](https://www.viessmann-community.com/t5/Lueftung/Kann-man-die-Vitovent-300-W-an-die-Vitotronic-200-WO1C/m-p/82637/highlight/true#M1938). Once you know it, you may use this info to find more references to it. 
The manual is usually a good resource too. That particular 0xB000 is not in there but a lot of the others.  
You get that 66  if you set the party mode on your vitotronic and read the data point afterwards ;-)

ronniebach, 2023-07-27:
Thanks. That's great info.

I now only miss:
-Common system temperature (not sure what the exact wording is)
-Compressor starts and hours. I cant seem to find the address on that.

Perhaps you have them?

Thanks

s0170071, 2023-12-02:
There is no such thing as system tempeature. There are many. 

-Common system temperature (not sure what the exact wording is)
-Compressor starts and hours. I cant seem to find the address on that.

Compressor starts: 0x0504
Runtime compressor: 0x0580

Air intake: 0x0103
Air out: 0x0104
Water forward: 0x0105
Water return: 0x0106

Here is my OpenHab config, you might find other temperatures in there. Look for the DP argument, this ist the data point . Its in german, but deepl.com is your friend.

```

Thing http:url:vitowifi "VitoWifi"   [baseURL="http://192.168.1.151/", commandMethod="POST", delay="500", refresh="45", timeout="500"]   {
    Channels:
   
        Type number : Channel_Vito_Geraetekennung               "Vito_Geraetekennung"               [ stateExtension="read?DP=0x00f8&Type=CountS", stateTransformation="REGEX:(^[-+]?[0-9]+)" ]
   //0
        Type number : Channel_Vito_Betriebsmodus                "Vito_Betriebsmodus"                [ stateExtension="read?DP=0xb000&Type=Mode", stateTransformation="REGEX:(^[-+]?[0-9]+)", commandExtension="write?DP=0xb000&Type=Mode&Value=%2$s" ]
        Type number : Channel_Vito_ManuellerModus               "Vito_ManuellerModus"               [ stateExtension="read?DP=0xb020&Type=Mode", stateTransformation="REGEX:(^[-+]?[0-9]+)", commandExtension="write?DP=0xb020&Type=Mode&Value=%2$s" ]
        Type number : Channel_Vito_Freigabe_WasserElektro       "Vito_Freigabe_WasserElektro"       [ stateExtension="read?DP=0x6015&Type=Mode", stateTransformation="REGEX:(^[-+]?[0-9]+)", commandExtension="write?DP=0x6015&Type=Mode&Value=%2$s" ]
        Type number : Channel_Vito_Freigabe_HeizenElektro       "Vito_Freigabe_HeizenElektro"       [ stateExtension="read?DP=0x7902&Type=Mode", stateTransformation="REGEX:(^[-+]?[0-9]+)", commandExtension="write?DP=0x7902&Type=Mode&Value=%2$s" ]
        Type number : Channel_Vito_Freigabe_Durchlauferh       "Vito_Freigabe_Durchlauferh  "       [ stateExtension="read?DP=0x7900&Type=Mode", stateTransformation="REGEX:(^[-+]?[0-9]+)", commandExtension="write?DP=0x7900&Type=Mode&Value=%2$s" ]
        Type number : Channel_Vito_Freigabe_Kuehlung            "Vito_Freigabe_Kuehlung"            [ stateExtension="read?DP=0x71fe&Type=Mode", stateTransformation="REGEX:(^[-+]?[0-9]+)", commandExtension="write?DP=0x71fe&Type=Mode&Value=%2$s" ]
        Type number : Channel_Vito_Status_Verdichter            "Vito_Status_Verdichter"            [ stateExtension="read?DP=0x0400&Type=Mode", stateTransformation="REGEX:(^[-+]?[0-9]+)" ]
        Type number : Channel_Vito_Status_Ventilatorstufe1      "Vito_Status_Ventilatorstufe1"      [ stateExtension="read?DP=0x0402&Type=Mode", stateTransformation="REGEX:(^[-+]?[0-9]+)" ]
        Type number : Channel_Vito_Status_Ventilatorstufe2      "Vito_Status_Ventilatorstufe2"      [ stateExtension="read?DP=0x0404&Type=Mode", stateTransformation="REGEX:(^[-+]?[0-9]+)" ]
        Type number : Channel_Vito_Status_Durchlauferh_Stufe1   "Vito_Status_Durchlauferh_Stufe1"   [ stateExtension="read?DP=0x0408&Type=Mode", stateTransformation="REGEX:(^[-+]?[0-9]+)" ]
        Type number : Channel_Vito_Status_Durchlauferh_Stufe2   "Vito_Status_Durchlauferh_Stufe2"   [ stateExtension="read?DP=0x0409&Type=Mode", stateTransformation="REGEX:(^[-+]?[0-9]+)" ]
        Type number : Channel_Vito_Status_HK_UmkehrCool         "Vito_Status_UmkehrCool"            [ stateExtension="read?DP=0x041c&Type=Mode", stateTransformation="REGEX:(^[-+]?[0-9]+)" ]
        Type number : Channel_Vito_Status_Pumpe_Heizkreis_A1    "Vito_Status_Pumpe_Heizkreis_A1"    [ stateExtension="read?DP=0x040d&Type=Mode", stateTransformation="REGEX:(^[-+]?[0-9]+)" ]
    //10     
        Type number : Channel_Vito_Status_Zusatzausgang_ZPumpe  "Vito_Status_Zusatzausgang_ZPumpe"  [ stateExtension="read?DP=0x0410&Type=Mode", stateTransformation="REGEX:(^[-+]?[0-9]+)" ]
        Type number : Channel_Vito_Status_Umschaltventil_H_WW   "Vito_Status_Umschaltventil_H_WW"   [ stateExtension="read?DP=0x0414&Type=Mode", stateTransformation="REGEX:(^[-+]?[0-9]+)" ]
        Type number : Channel_Vito_Status_Speicherladepumpe     "Vito_Status_Speicherladepumpe"     [ stateExtension="read?DP=0x0416&Type=Mode", stateTransformation="REGEX:(^[-+]?[0-9]+)" ]
        Type number : Channel_Vito_Status_Kaeltekreisumkehr     "Vito_Status_Kaeltekreisumkehr"     [ stateExtension="read?DP=0x049c&Type=Mode", stateTransformation="REGEX:(^[-+]?[0-9]+)" ]
        Type number : Channel_Vito_Status_Sammelmeldung         "Vito_Status_Sammelmeldung"         [ stateExtension="read?DP=0x0411&Type=Mode", stateTransformation="REGEX:(^[-+]?[0-9]+)" ]
        Type number : Channel_Vito_Aussentemperatur             "Vito_Aussentemperatur"             [ stateExtension="read?DP=0x0101&Type=TempL", stateTransformation="REGEX:(^[-+]?[0-9]+\\.[0-9]+)" ]
        Type number : Channel_Vito_Raumtemperatur_ist           "Vito_Raumtemperatur_ist"           [ stateExtension="read?DP=0x0116&Type=TempL", stateTransformation="REGEX:(^[-+]?[0-9]+\\.[0-9]+)" ]
        Type number : Channel_Vito_AussentemperaturRegelung     "Vito_AussentemperaturRegelung"     [ stateExtension="read?DP=0x0952&Type=TempL", stateTransformation="REGEX:(^[-+]?[0-9]+\\.[0-9]+)" ]
        Type number : Channel_Vito_Brauchwassertemperatur_oben  "Vito_Brauchwassertemperatur_oben"  [ stateExtension="read?DP=0x010D&Type=TempL", stateTransformation="REGEX:(^[-+]?[0-9]+\\.[0-9]+)" ]
[... cut ...]
         Type number : Channel_Vito_Aussentemperatur4             "Vito_Aussentemperatur"             [ stateExtension="read?DP=0x081d&Type=TempL", stateTransformation="REGEX:(^[-+]?[0-9]+\\.[0-9]+)" ]
      Type number : Channel_Vito_Aussentemperatur4             "Vito_Aussentemperatur"             [ stateExtension="read?DP=0x081e&Type=TempL", stateTransformation="REGEX:(^[-+]?[0-9]+\\.[0-9]+)" ]

   
     //   Type number : Channel_Vito_EVUMQTT                     "Vito_EVUMQTTreadback"             [ stateExtension="control?cmd=Status,GPIO,4" , stateTransformation="JSONPATH:$.state" ]
}
*/

```

fesch, 2024-10-12:
Ich habe auch einen V200WO1B und kann sowei alles auslesen, bis auf dir Error-Codes. Sobald ich diese versuche aus zu lesen bekomme ich einen CRC Error und dann sind auch alle aneren Datepunkte mit einem CRC Error behaftet.

`  VitoWiFi::Datapoint("ecnsysFehlerhistorie0" , 0x7507, 9, VitoWiFi::noconv),
  VitoWiFi::Datapoint("ecnsysFehlerhistorie1" , 0x7510, 9, VitoWiFi::noconv),
  VitoWiFi::Datapoint("ecnsysFehlerhistorie2" , 0x7519, 9, VitoWiFi::noconv),
  VitoWiFi::Datapoint("ecnsysFehlerhistorie3" , 0x7522, 9, VitoWiFi::noconv),
  VitoWiFi::Datapoint("ecnsysFehlerhistorie4" , 0x752B, 9, VitoWiFi::noconv),
  VitoWiFi::Datapoint("ecnsysFehlerhistorie5" , 0x7534, 9, VitoWiFi::noconv),
  VitoWiFi::Datapoint("ecnsysFehlerhistorie6" , 0x753D, 9, VitoWiFi::noconv),
  VitoWiFi::Datapoint("ecnsysFehlerhistorie7" , 0x7546, 9, VitoWiFi::noconv),
  VitoWiFi::Datapoint("ecnsysFehlerhistorie8" , 0x754F, 9, VitoWiFi::noconv),
  VitoWiFi::Datapoint("ecnsysFehlerhistorie9" , 0x7558, 9, VitoWiFi::noconv),`
  
  Irgend ein Tipp warum diese nicht auslesbar sind?

stevstrong, 2025-02-16:
Hallo,
ich habe einen WO1B.
Weißt jemand wie man 1xWarmwasserbereitung einschalten kann? Parameter und Wert?
Danke im voraus.

sude22, 2025-02-16:
Moin, ist bei mir 0xB020 (1Byte) mit Wert 0 oder 1

s0170071, 2025-02-16:
@stevstrong 
**Ich schreib ne 2 nach 0xb020**. 

Die WO1C nuutzt hier wohl das gleiche Register.  Sie nutzt 2 für 1xWW und eine 1 für "sofort Heizen". 
Siehe hier: https://community.viessmann.de/t5/Waermepumpe-Hybridsysteme/1xWW-Bereitung-WO1C/td-p/54507

Es gibt den Automatikmodus (aus / WW / heizen+ww), oben in der Liste Betriebsmodus 0xb000
val int VITO_Auto_OFF = 0
val int VITO_Auto_WW = 1
val int VITO_Auto_wwheat = 2
 
Der Manuelle Modus übersteuert den aktuellen Automatikmodus. Adresse ist  0xb020
val int VITO_ManuellerModus_OFF = 0
val int VITO_Vito_ManuellerModus_HEATER_ON = 1
val int VITO_Vito_ManuellerModus_WATER_ON = 2
val int VITO_ManuellerModus_HeizenParty = 66

stevstrong, 2025-02-17:
@sude22 , @s0170071  danke für euere Antworten, ich werde zuerst Wert 1 ausprobieren, und wenn nicht geht dann Wert 2.

sude22, 2025-02-17:
Die Info von @s0170071 ist wohl korrekt. Habe das scheinbar nie mehr richtig getestet..
Bei Wert = 1 startet bei mir die Aufheizung Heizung-Pufferspeicher.
Bei Wert = 2 passiert scheinbar nichts, auch nach längerer Wartezeit. Ich habe dann die WW-Aufbereitung über das Menü gestartet am Gerät, auch da wird der Wert auf 2 gesetzt, aber nichts passiert.

stevstrong, 2025-02-17:
Wenn nichts passiert das könnte bedeuten dass die Tempereatur bereits hoch genug ist, weniger als (Ist + hysterese)

CaneTLOTW, 2025-06-27:
Gefunden:

0x160D 3 byte lang. 
Es wird jedoch nur das erste byte benötigt --> 'b:0:0'

(4,"COMMON.gedämpfte.Aussentemperatur (160D)", 0x160D, 3,'b:0:0', 0.1, False),

CaneTLOTW, 2025-07-04:
es sind doch die ersten beiden bytes b:0:1

Mit 3 byte lesen und nur die beiden Ersten nutzen gibt es noch weitere.

Stefan1357, 2025-12-06:
[image]

Ich verfolge die Diskussion hier mit Begeisterung und lese meine Vitocal 200s (204d) mit openv und ESPhome aus. Das klappt auch alles super. 
Lediglich die Daten des Kältekreises bekomme ich nicht rein. 

Beispielsweise kann ich zwar 0x0103 und 0x0104 auslesen, aber die Werte sind immer 0 grad. 
Kann es sein, dass die Steuerung die Daten gar nicht hat und lediglich zur Diagnose des Kältekreises bei der Außeneinheit abruft, aber selbst nicht speichert? Scheinbar werden diese auch nur ohne Nachkommastelle übermittelt (siehe Foto). 

0x0101 ist die Außentemperatur, sodass 0x0103 eigentlich schon richtig sein sollte, oder nicht?