openv/openv #424: Außentemperatur setzen?

fotter, 2018-10-01:
Ich hoffe, ich bin hier nicht völlig falsch.
Unser Außentemperatursensor wurde vom Fachmann aufgrund "baulicher Gegebenheiten" in einem Lichtschacht auf der Südeseite des Gebäudes versteckt...
Gibt es eine Möglichkeit, der Heizung einen bestimmte Wert vorzugauckeln (z.Bsp. von Raspi aus dem Internet geladene Temperatur)?

Danke im Voraus!

speters, 2018-10-08:
Ohne Gewähr:

Je nach Regelung gibt es an Adresse `0x7797` ein Konfigurationsbyte mit 0=Aussentemperatur vom Sensor übernehmen, 1=vom LON uebernehmen, 2=auf LON senden.

Probiere mal, dieses Byte auf 1 zu setzen, so dass der Aussentemperatursensor nicht gelesen wird.

Auf Adresse `0x0952` findet sich die für die Regelung verwandte Aussentemperatur, die du nun schreibst als 2 Byte mit dem Wert Temperatur in °C / 10. Also 17.2°C = 0xAC 0x00.

Berichte mal, ob das klappt...

w353r, 2018-11-12:
Ich bin zwar nicht der Thread-Ersteller, habe aber auch Interesse die Außentemperatur per Optolink zu schreiben. Leider lässt sich anscheinend die Adresse 0x0952 anscheinend nicht beschreiben:

7797 auf 1 setzen:
```
DEBUG:Mon Nov 12 20:01:06 2018 : Command: setTempSensSrc FROM_LON
DEBUG:Mon Nov 12 20:01:06 2018 : FROM_LON -> 01
DEBUG:Mon Nov 12 20:01:06 2018 : >SEND: 41
DEBUG:Mon Nov 12 20:01:06 2018 : >SEND: 06
DEBUG:Mon Nov 12 20:01:06 2018 : >SEND: 00
DEBUG:Mon Nov 12 20:01:06 2018 : >SEND: 02
DEBUG:Mon Nov 12 20:01:06 2018 : >SEND: 77
DEBUG:Mon Nov 12 20:01:06 2018 : >SEND: 97
DEBUG:Mon Nov 12 20:01:06 2018 : >SEND: 01
DEBUG:Mon Nov 12 20:01:06 2018 : >SEND: 01
DEBUG:Mon Nov 12 20:01:06 2018 : >SEND: 18
DEBUG:Mon Nov 12 20:01:06 2018 : <RECV: len=1 06 (90.0 ms)
DEBUG:Mon Nov 12 20:01:06 2018 : <RECV: received 06
DEBUG:Mon Nov 12 20:01:06 2018 : >FRAMER: framer_set_actaddr framer_current_addr = 9777 (was FFFF)
DEBUG:Mon Nov 12 20:01:06 2018 : >FRAMER: Command send
DEBUG:Mon Nov 12 20:01:06 2018 : >FRAMER: no preset result
DEBUG:Mon Nov 12 20:01:06 2018 : <RECV: len=1 41 (0.0 ms)
DEBUG:Mon Nov 12 20:01:06 2018 : <RECV: received 41
DEBUG:Mon Nov 12 20:01:06 2018 : <RECV: len=1 05 (0.0 ms)
DEBUG:Mon Nov 12 20:01:06 2018 : <RECV: received 05
DEBUG:Mon Nov 12 20:01:06 2018 : <RECV: len=1 01 (0.0 ms)
DEBUG:Mon Nov 12 20:01:06 2018 : <RECV: len=5 02 (10.0 ms)
DEBUG:Mon Nov 12 20:01:06 2018 : <RECV: received 01 02 77 97 01 17
DEBUG:Mon Nov 12 20:01:06 2018 : >FRAMER: framer_reset_actaddr framer_current_addr = FRAMER_NO_ADDR (was 9777)
DEBUG:Mon Nov 12 20:01:06 2018 : 00 -> OK
DEBUG:Mon Nov 12 20:01:06 2018 : OK
```

0x952 schreiben:
```
DEBUG:Mon Nov 12 20:02:50 2018 : Command: setTempA 20.0
DEBUG:Mon Nov 12 20:02:50 2018 : Send Exp: V*10 [V=20.000000]
DEBUG:Mon Nov 12 20:02:50 2018 : Type: short (bytes: C8 )
DEBUG:Mon Nov 12 20:02:50 2018 : >SEND: 41
DEBUG:Mon Nov 12 20:02:50 2018 : >SEND: 07
DEBUG:Mon Nov 12 20:02:50 2018 : >SEND: 00
DEBUG:Mon Nov 12 20:02:50 2018 : >SEND: 02
DEBUG:Mon Nov 12 20:02:50 2018 : >SEND: 09
DEBUG:Mon Nov 12 20:02:50 2018 : >SEND: 52
DEBUG:Mon Nov 12 20:02:50 2018 : >SEND: 02
DEBUG:Mon Nov 12 20:02:50 2018 : >SEND: C8
DEBUG:Mon Nov 12 20:02:50 2018 : >SEND: 00
DEBUG:Mon Nov 12 20:02:50 2018 : >SEND: 2E
DEBUG:Mon Nov 12 20:02:50 2018 : <RECV: len=1 06 (30.0 ms)
DEBUG:Mon Nov 12 20:02:50 2018 : <RECV: received 06
DEBUG:Mon Nov 12 20:02:50 2018 : >FRAMER: framer_set_actaddr framer_current_addr = 5209 (was FFFF)
DEBUG:Mon Nov 12 20:02:50 2018 : >FRAMER: Command send
DEBUG:Mon Nov 12 20:02:50 2018 : >FRAMER: no preset result
DEBUG:Mon Nov 12 20:02:50 2018 : <RECV: len=1 41 (0.0 ms)
DEBUG:Mon Nov 12 20:02:50 2018 : <RECV: received 41
DEBUG:Mon Nov 12 20:02:50 2018 : <RECV: len=1 06 (20.0 ms)
DEBUG:Mon Nov 12 20:02:50 2018 : <RECV: received 06
DEBUG:Mon Nov 12 20:02:50 2018 : <RECV: len=6 03 (0.0 ms)
DEBUG:Mon Nov 12 20:02:50 2018 : <RECV: len=1 68 (20.0 ms)
DEBUG:Mon Nov 12 20:02:50 2018 : <RECV: received 03 02 09 52 01 01 68
DEBUG:Mon Nov 12 20:02:50 2018 : >FRAMER: framer_reset_actaddr framer_current_addr = FRAMER_NO_ADDR (was 5209)
DEBUG:Mon Nov 12 20:02:50 2018 : >FRAMER: ERROR address 0952 code 1
DEBUG:Mon Nov 12 20:02:50 2018 : Error in recv, terminating
DEBUG:Mon Nov 12 20:02:50 2018 : Error executing setTempA 20.0
ERR: >FRAMER: ERROR address 0952 code 1
Error in recv, terminating
Error executing setTempA 20.0
```

Noch andere Ideen, die ich ausprobieren könnte? Bei meinem Gerät handelt es sich um eine Vitodens 300W (ID=20CB).