openv/openv #425: VitoConnect 100 mit Raspberry PI als Proxy

muu55, 2018-10-05:
Meine neue Heizung (Vitodens-222 F) kann ich mittels der ViCare App "fernbedienen", leider ist der Funktionsumfang  dieser App, na sagen wir mal, "begrenzt".  Das schreit nach einer Verbesserung, wobei ich die wesentlichen Komponenten ja bereits habe und auch in ihrer bisherigen Funktion erhalten möchte.

Am schönsten wäre natürlich, ich könnte mich auf der VitoConnect anloggen und direkt
(also ohne das Internet) meine Heizung abfragen und steuern. Das wüde den Nutzen so 
einer Box erheblich steigern, ist vom Hersteller aber wohl nicht vorgesehen (?).
Also hab ich mir etwas ausgedacht.

Benötigte Bauteile: Eine Viessman Heizumg mit VitoConnect 100 Typ OPTO1, 
ein Raspberry PI Zero WH und ein CP2102 USB-Seriell Adapter.

Um sowohl zu verstehen, was die VitoConnect 100 mit meiner Heizung redet, als auch eigene Kommandos zu senden, habe ich zwischen das original OPTO-Link Kabel und die VitoConnect 100 einen Raspberry PI Zero gehängt, den ich um einen CP2102 USB-Seriell-Adapter erweitert habe.

Das Optolink Kabel verbinde ich über ein "USB-A Buchse auf Micro-USB Stecker"-Kabel mit der USB-Buchse vom PI-Zero. Den CP2102 Adapter verbinde ich über drei Leitung (GND, Tx, Rx) mit dem seriellen Port des PI Zero auf der GPIO Pinleiste und auf der USB-Seite mit der Vitoconnect 100.
Fertig ist die Hardware. Dazu noch ein kleines Programm, das die beiden seriellen Schnitstellen miteinander verbindet und die Daten beim Durchreichen mitprotokolliert.
Schon hat man einen einfachen Proxy mit man-in-the-middle Option.

Erste Erkenntnis: Die VitoConnect 100 redet ununterbrochen mit der Heizung.
Also nicht nur, wenn ich mit der ViCare App eine Verbindung aufbaue oder etwas abfrage.
Offenbar wird dabei das Protokoll 300 (bzw. eine erweiterte Variante) verwendet.

Ich sammle erstmal noch Daten. Sobald ich mehr davon verstehe, versuche ich auch eigene Kommandos auszuführen. Wer so einen Proxy nachbauen will oder wem diese kurze Beschreibung nicht konkret genug war, bitte einfach melden.

GolterU, 2018-10-05:
Zum Thema Mitschneiden des Datenverkehrs der vitoconnect 100 s.a. [Bauanleitung USB_Duo Sniffer](https://github.com/openv/openv/wiki/Bauanleitung-USB_Duo-Sniffer). Dort ist eine alternative Hardware dafür beschrieben und es sind ein paar daraus gewonnene Erkenntnisse geschildert.

muu55, 2018-10-05:
Ja, diese Bauanleitung habe ich gesehen und gelesen, da hatte ich meine Hardware schon fertig. 
Schaue ich mein Logfile an, kann ich die dort berichteten Ergebnisse bestätigen. Wenn ich die Verbindung zwischen VitoConnect100 und der Heizung im Proxy abschalte, dann "pingt" die Vitoconnect nach der Heizung mit 0x04 0x16 0x00 0x00. Dauert die Auszeit nicht zu lang, dann findet die VitoConnect die Heizung wieder und plappert erneut drauf los.

muu55, 2018-10-05:
Ich schau mal, ob ich das vcontrold auf dem PI an meinen Proxy "anflanschen" kann.
Sobald das klappt, poste ich wie es geht.

DivisonByZero, 2018-10-07:
Hallo muu55,

ich habe einen ähnlichen Aufbau mit meinem Raspi3 versucht:
- Optolink-Kabel an /ttyUSB0
- USB-UART-Adapter1 an /ttyUSB1 
- VitoConnect mit zweitem USB-UART-Adapter2
- die beiden USB-UARTs mit GND und gekreuzten TX und RX verbunden

Mit python ein paar zeilen Code geschrieben: 
- beide serielle Schnittstellen werden mit ( 4800, 8, 2, E) geöffnet 
- in zwei Threads wird jeweils die eine Schnittstelle gelesen und die Daten an die andere gesendet

Bisher kann ich sehen, dass vom Optrolink periodisch 0x05 geschickt wird, was auch an den VitoConnect rausgeht - dieser reagiert aber garnicht (sendet auch nichts von sich aus ...)

Habe ich etwas vergesen? Hast Du vielleicht eine Idee oder Dein Beispiel-Code? 

P.S.: Losgelöst von diesem Versuch funktioniert die ViCare-App und ich kann auch einige Daten manuell auslesen.

muu55, 2018-10-07:
Also ich vermute, die VitoConnect erwartet an ihrem USB-Port einen USB-Seriell Adapter mit CP210x Chip, der dieselbe Vendor:Chip ID wie das Optolink-Kabel hat.

muu55, 2018-10-07:
Das kleine C-Programm findet sich hier: https://github.com/muu55/vitolog.git

muu55, 2018-10-07:
Hier die Addressen+Länge, die von "meiner" VitoConnect100 laufend abgefragt werden:

 00 F0 02    
 00 F8 01 
 00 F8 08    GWG_SystemIdent
 00 F9 01    GWG_Kennung
 00 FA 01    WPR_Hardware_Index
 00 FB 01    SystemIdent_SW1
 00 FC 01    WPR_Protokollversion_LDAP
 00 FD 01    WPR_Protokollversion_RDAP
 00 FE 01    WPR_Software_Version_Byte1
 00 FF 01    WPR_Software_Version_Byte2

 08 10 04    TiefpassTemperaturwert_KTS
 08 3A 01    TemperaturFehler_ATS
 08 3B 02    TemperaturFehler_KTS
 08 8E 08    Uhrzeit
 08 96 06    TiefpassTemperatur_RTS_A1M1
 08 9C 03    NRF_TemperaturFehler_RTS_M1

 20 00 30    WPR_HK1_Normaltemperatur
 20 30 08    <xxx>Durchfluss
 21 00 30 
 21 30 08    mBusNRZ012130Stichtag02
 22 00 30 
 22 30 08 
 23 06 2C    BedienRTSolltemperaturA1M1
 23 09 19    FerienBeginnA1M1
 25 00 16    HK_RaumsolltemperaturaktuellA1M1
 25 35 15    HK_FerienbetriebA1M1
 27 D3 02    KD3_KonfiNeigungA1
 29 00 02    Vorlauftemperatur
 29 04 01    Vorlauftemperaturstatus
 30 00 30    WPR_HK2_Normaltemperatur
 30 30 08 
 31 00 30 
 31 30 08 
 32 00 30 
 32 30 08 
 33 06 2C    BedienRTSolltemperaturM2
 33 09 19    FerienBeginnM2
 35 00 36    HK_AktuelleBetriebsartM2
 35 49 01    ExtBetriebsartenumschaltung_M2
 37 D3 02    KD3_KonfiNeigungM2D
 39 00 02    VorlauftemperaturM2
 39 04 01    Status_2_M2
 40 00 30    WPR_HK3_Normaltemperatur
 40 30 08 
 41 00 30 
 41 30 08 
 42 00 30 
 42 30 08 
 43 06 2C    BedienRTSolltemperaturM3
 43 09 10    FerienBeginnM3
 45 00 16 
 45 35 01 
 45 49 01 
 47 D3 02 
 49 00 02 
 49 04 01    Status_2_M3
 55 25 02    TiefpassTemperaturwert_ATS
 55 D3 09    GWG_Flamme
 57 21 03    K21_KonfiBetriebsstdBrenner_Wartung
 57 24 01    K24_KonfiStatusWartung
 57 38 01    K38_KonfiFehlerByteGFA
 63 00 01    Bedien_WW_Solltemperatur
 75 07 36    Fehlerhistorie
 75 3D 24 
 75 61 0A 
 75 6C 08    LetzteIntervallWartung
 77 00 01 
 77 7F 01 
 F0 00 20

DivisionDurchNull, 2018-10-07:
Danke für den Hinweis.
Ich habe 3 verschiedene USB-UARTs ... leider alle mit einem PL2303-Chip  :(
Mal sehen wann die Bestellung vom CP2102 eintrudelt (gab genau 1 Angebot aus DE auf ebay ...). Dann setze ich mich wieder dran ...

speters, 2018-10-08:
Hast du @muu55 eine Möglichkeit, die Kommunikation der VitoConnect mit dem Viessmann-Server mitzuschneiden (z.B. [mitmproxy](https://mitmproxy.org/)?
Ggf. liesse sich das API des Viessmann-Servers dann zu weiteren Erkenntnisgewinnen über Featuresets/Adressen/EventTypes anderer Anlagen nutzen. Die XML-Dateien der VitoSoft-Demo sind zwar schon sehr umfassend, allerdings auch relativ alt und nicht fehlerfrei.

Kannst du in den Logs Kommunikation finden, die nicht ReadData/0x01/Virtual_Read oder WriteData/0x02/Virtual_Write, sondern z.B. Funktion Calls/0x07/Remote_Procedure_Call sind? (wg. Ergänzung der Dokumentation von [Protokoll 300](https://github.com/openv/openv/wiki/Protokoll-300))

---- 

##### Falls du die XML-Dateien der VitoSoft mal angeschaut hast:
Es gibt EventTypes, mit der gleichen Adresse, aber unterschiedlicher BytePosition (quasi Offset). Kannst du erkennen, ob der Zugriff dann auf Adresse+BytePosition in ByteLength Länge stattfindet oder ob die Adresse in voller BlockLength abgefragt wird? Und wie sieht das beim Schreiben von EventTypes aus, die eine BitLength > 0 haben?
Hintergrund ist der, dass ich zunächst dachte, der Adressraum wäre linear und direkt adressierbar (was z.B. 0x00F8 ff. nahelegt), bis ich dann herausbekommen habe, dass das nicht so ist (konkret war das bei der Fehlerhistorie, die ich mal nicht in Vielfachen von (BlockLength/BlockFactor) ausgelesen hatte).

GolterU, 2018-10-08:
Ich habe längere Zeit die Daten mit dem [Sniffer](https://github.com/openv/openv/wiki/Bauanleitung-USB_Duo-Sniffer) aufgezeichnet, die Vitoconnect mit der Heizung (also auf der seriellen Ebene) austauscht. Zum Viessmann-Server sollte er dann wohl nichts anderes schicken können, als er von der Heizung geliefert bekommt. Auszug siehe Excel-Datei [hier](https://github.com/openv/openv/wiki/Bauanleitung-USB_Duo-Sniffer#datenverkehr-der-vitoconnect-100). In dieser Datei [Vitoconnect-Output.zip](https://github.com/openv/openv/files/2455842/Vitoconnect-Output.zip)
 sind noch mehr Daten gespeichert, z.T. mit grafischer Auswertung, um 'rauszufinden, wann und wie häufig die Adressen abgefragt werden. Darin sieht man zwar die zyklischen Abfragen, aber auch, dass die Abfragen variiert werden und sich nicht 100% gleichartig wiederholen.

Was den beschriebenen Adressraum anbelangt: Ich glaube jetzt, dass die Adressen eher als Befehle zu verstehen sind, die die Heizungssteuerung mit unterschiedlich langen Datensätzen beantwortet. Damit werden nicht einfach Speicherplätze im Bereich von 0x0000 bis 0xFFFF adressiert und das dort gespeicherte Byte zurückgegeben.
Beispiel:
Die Abfrage der Adresse 0x2500 (Status Heizkreis A1M1) mit 

41 | 05 | 00 | 41 | 25 | 00 | 16 | 81
-- | -- | -- | -- | -- | -- | -- | --

 liefert 

xx | xx |xx |xx | xx | xx | xx | 00 | 01 | 02 | 03 | 04 | 05 | 06 | 07 | 08 | 09 | 10 | 11 | 12 | 13 | 14 | 15 | 16| 17| 18| 19| 20| 21| xx 
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
41 | 1B | 01 | 41 | 25 | 00 | 16 | 02 | 02 | 53 | 25 | 04 | 00 | 47 | 01 | 01 | 00 | 00 | 01 | C8 | 00 | 00 | 00 | 00 | 47 | 01 | 00 | C8 | 00 | 3A

Im Antwortdatensatz mit 22 Byte Nutzdaten stecken laut "ecnEventType.xml" (aus der Vitosoft-Demo) 
an der Byteposition 1 (wird ab 0 gezählt) mit Länge 1 der Heizungsstatus als Integer,
ab Pos. 2 mit 4 Byte Länge der Aufheiztimer als Double in Sekunden,
ab Pos. 6 mit 2 Byte Länge die Vorlauf-Solltemperatur als Double (hier 32,7°C, Berechnung wie [hier](https://github.com/openv/openv/wiki/Protokoll-300)),
an Pos. 8 mit 1 Byte Länge der Heizkreisfreigabe-Status als Integer,
an Pos. 10 mit 1 Byte Länge der Status der Heizkreispumpe als Integer,
an Pos. 11 mit 1 Byte Länge die RSWS-Betriebsart (was immer das ist) als Integer,
ab Pos. 12 mit 2 Byte Länge die Raumsolltemperatur als Double (hier 20°C),
an Pos. 16 mit 1 Byte Länge der Zustand Frostgefahr.

Was mich irritiert: An der gleichen Adresse kann auch ein Datensatz mit 17 Byte Datensatzlänge abgefragt werden kann. Laut der obigen XML-Datei sind dann andere Daten unter den genannten Bytepositionen/Offsets verfügbar. Z.B. an Pos. 2 steht anstelle "HK_AufheiztimerA1M1" der Wert von "NRF_HK_VT_Solltemperatur_A1M1".

speters, 2018-10-08:
Danke für [dein Log](https://github.com/openv/openv/files/2455842/Vitoconnect-Output.zip).
Habe gerade erst gesehen, dass da ja ganze 19 Sheets mit Daten drin sind.

Im 4. Byte der Antwort wo der klassischen P300 Protokolldefinition nach nur 0x01, 0x02, 0x07 vorkommen dürfte, sind Bits 7,6,5 offenbar als Zähler (0..7) verwendet worden (da kann man sich Anwendungsmöglichkeiten ausdenken: Erkennung fehlender Pakete,  Zuordnen von Anfragen zu Antworten, ...).
Den Daten nach hast Du eine Systemkennung 0x20CB0009. Also 20CB Hardwarereindex 0, Softwarereindex 9. Bei meiner 0x20CB000B (ähnlich, nur Softwarerevision 11) kann ich diese Zählersache nicht beobachten.

Hast du ein Log, wo die Anfragen der Vitoconnect auch auftauchen?
Ein wenig stutzig macht mich die Logzeile
`06 | 41 | 06 | 03 | 21 | 00 | F0 | 01 | 01 | 1C`
Was für eine Anfrage produziert diesen Fehler? Die Adresse 0x00F0 (ecnsysDeviceIdentF0) und die Länge 1 müsste doch valide sein.

PS:
Ich kann auf die Schnelle nix sehen, aber hast du irgendwo geloggt, wie die Fehlerhistorie und die Fehlerhistorie des Feuerungsautomaten gelöscht werden?

GolterU, 2018-10-08:
> Danke für [dein Log](https://github.com/openv/openv/files/2455842/Vitoconnect-Output.zip).
> Habe gerade erst gesehen, dass da ja ganze 19 Sheets mit Daten drin sind.

Hallo Sönke,
die Logs in den ersten Blättern obiger Datei sind während der Zeit entstanden, als ich mir den [USB-Sniffer](https://github.com/openv/openv/wiki/Bauanleitung-USB_Duo-Sniffer) ausdachte. Am Anfang war die Hardware so, dass nur die Antwortdaten der Heizungsregelung an die Vitoconnect aufgezeichnet wurden. Die Erläuterungen und Kommentare auf den einzelnen Blättern sind wahrscheinlich auch nicht sehr aufschlussreich. Es war halt nicht zur Veröffentlichung gedacht.  Ab Tabellenblatt T8 sind auch Sendedaten enthalten. Jetzt (so auch [hier](https://www.godemann.de/test/git/Vitoconnect-Output-public.xlsm)) ist standardmäßig der Datenverkehr in beiden Richtungen gespeichert.

> Im 4. Byte der Antwort wo der klassischen P300 Protokolldefinition nach nur 0x01, 0x02, 0x07 vorkommen dürfte, sind Bits 7,6,5 offenbar als Zähler (0..7) verwendet worden (da kann man sich [...]

Meine Vermutungen zu diesen drei ominösen Bits (und was mir sonst noch so aufgefallen ist) hatte ich [hier](https://github.com/openv/openv/wiki/Bauanleitung-USB_Duo-Sniffer#datenverkehr-der-vitoconnect-100) geschildert. 
Da die drei Bits schon in den Anfragen an die Heizung hin und wieder als gesetzt auftauchen, kann es eigentlich nicht mit dem Softwarestand der Regelung zusammenhängen. Die scheint in ihren Antworten dieses Byte einfach aus der Anfrage zu wiederholen. Logs dazu siehe ebenfalls [hier](https://www.godemann.de/test/git/Vitoconnect-Output-public.xlsm). Eine Systematik bezüglich der Werte dieser drei Bits habe ich bislang nicht erkannt.

> Ein wenig stutzig macht mich die Logzeile
> `06 | 41 | 06 | 03 | 21 | 00 | F0 | 01 | 01 | 1C`
> Was für eine Anfrage produziert diesen Fehler? Die Adresse 0x00F0 (ecnsysDeviceIdentF0) und die Länge 1 müsste doch valide sein.

Die Adresse 0x00F0 erscheint in der "ecnDataPointType.xml" (aus der Vitosoft-Demo) nicht, jedenfalls nicht unter Device-ID "VScotHO1_4" - die hier zutreffen sollte. In der "ecnEventType.xml" (auch aus der Vitosoft-Demo) steht sie mit der Bezeichnung "KBUS_V300_T03_VirtKanal_07" und der Länge von zwei Byte. 
Ich vermute, Du hast die Zeile aus dem Blatt T1 der "Vitoconnect-Output.xlsm". Dort hatte ich einfach mal den Beginn des Datenverkehrs zwischen Vitoconnect und Heizung aufgezeichnet, allerdings wie gesagt, nur die Antwortdaten. Warum Vitoconnect die Adresse benutzt, bevor der Typ der Steuerung abgefragt wird, erschließt sich mir nicht. Jedenfalls ist die Adresse 0x00F0 bei mir ungültig. Ich hatte mal die Heizung mit Hilfe des Programms "Vies-sion" mit allen Adressen 0...0xFFFF traktiert, dabei gibt es bei 0x00F0 auch einen Fehler.
Wenn es Dich interessiert, kann ich nochmal ein Log vom Startvorgang der Vitoconnect erzeugen, diesmal mit den Daten aus beiden Richtungen.

> 
> PS:
> Ich kann auf die Schnelle nix sehen, aber hast du irgendwo geloggt, wie die Fehlerhistorie und die Fehlerhistorie des Feuerungsautomaten gelöscht werden?

Bislang habe ich nur ganz wenige Schreibvorgänge manuell ausgelöst und mitgeloggt. Die Fehlerhistorie (an 0x7507) habe ich nicht angefasst. Da dieser Datenpunkt als read/write ausgewiesen ist, kann man den Inhalt evtl. einfach auf Null setzen??

Gruß
Uwe aka Golter

hendrikland, 2018-12-13:
> * Optolink-Kabel an /ttyUSB0
> * USB-UART-Adapter1 an /ttyUSB1
> * VitoConnect mit zweitem USB-UART-Adapter2
> * die beiden USB-UARTs mit GND und gekreuzten TX und RX verbunden
>
> Mit python ein paar zeilen [...]

Könntest du den Python-Code sharen? Ich habe einen ähnlichen Aufbau vor, es geht erstmal nur darum die Daten mitzulesen. In meinem Fall ein Pellet-Kessel (Vitoligno 300-C), bei dem es nützlich wäre einige Werte zu erfassen (Anzahl Zündvorgänge, Brenndauer, Pelletverbrauch,...). Leider bietet die App nichts davon an, ich vermute aber das das Vitoconnect die Werte ausliest. Ziel wäre es, dieses auf dem Raspi abzuzweigen und separat wegzuschreiben (z.B. Grafana).

DivisionDurchNull, 2018-12-14:
Hallo hendrikland,

hier mein Beispiel. Aufruf ist dann z.B. 
`python3 "programm.py" > "logfile"`

Und es ist ein UART mit dem Chip CP2102 notwendig (PL2303 geht definitiv nicht) .

```

#!/usr/bin/python
#!/usr/bin/env python
import serial
import time
import threading

__ready  = b'\x05'

serv = serial.Serial("/dev/ttyUSB1",
                     baudrate=4800,
                     parity=serial.PARITY_EVEN,
                     stopbits=serial.STOPBITS_TWO,
                     bytesize=serial.EIGHTBITS,
                     timeout=0.3)

sero = serial.Serial("/dev/ttyUSB0",
                     baudrate=4800,
                     parity=serial.PARITY_EVEN,
                     stopbits=serial.STOPBITS_TWO,
                     bytesize=serial.EIGHTBITS,
                     timeout=0.3)

def serial_read(s,t, info):
    while True:
        msg = s.read(24)
        if msg:
            start_time = time.strftime("%d.%m.%Y %H:%M:%S")
            print(info,  end="")
            print(start_time, end="")
            print(" ", end="")
            for c in msg:
                print("%02x " % c, end="")
            print()
 
            t.write(msg)

#serial_read(sero, serv)

thread1 = threading.Thread(target=serial_read, args=(serv, sero, 'S ',),)
thread2 = threading.Thread(target=serial_read, args=(sero, serv,'R ',),)
thread2.start()
thread1.start()

```

hydrax0, 2020-04-08:
@hendrikland 
Hi. Ich habe auch eine Vitoligno 300-C und suche verzweifelt nach Adressen, vor allem den FehlerStatus. Ich bekomme korrekte Antworten bei der Fehlerhistorie der letzten 10 Fehler, aber der Status selbst funktioniert noch nicht. 
Hattest du schon Erfolg und würdest du die Adressen mit mir teilen?
Grüsse Richard

dima-357, 2020-11-22:
Please tell me the request address V100
boiler pump in operation or not /
It is not in the list of addresses of Viessmann-adresses

TheCutter, 2022-05-10:
> Könntest du den Python-Code sharen? Ich habe einen ähnlichen Aufbau vor, es geht erstmal nur darum die Daten mitzulesen. In meinem Fall ein Pellet-Kessel (Vitoligno 300-C), bei dem es nützlich wäre [...]

@hendrikland Bist du mit dem Vitoligno 300-C weiter gekommen? Ich würde meinen auch gerne anbinden und die Werte von dir interessieren mich auch. Danke und Gruß

idstein, 2022-07-23:
Wir standen vor dem gleichen Problem und wollten die VitoConnect 100 noch weiter laufen lassen. Da mir ein Raspberry für die Arbeit zu schade war und es mittlerweile für <10 € einen ESP32 DevBoard (mit CP2102 USB / UART Chip) gibt, wurde es dann eben eine minimale ESPHome Lösung.

Netterweise kann man den ESP direkt in den USB Port der VitoConnect 100 stecken und fleißig als Proxy den Traffic verfolgen oder eben weitere Requests einfließen lassen. Kleiner Bonus: Es läuft demnächst auch auf dem ESP ein kleiner Modbus Slave der unserer VitoCal 200-S die frei "verfügbare" PV-Strom Messwerte übergibt.
[image]

Für Viessmann ist diese Variante ohne größeren Aufwand und bei ausschließlich mitlesendem Gebrauch quasi nicht detektierbar.

Wichtig! Es reicht kein ESP8266 und kein Software UART, da die Kommunikation doch schon sehr (A) zeitkritisch erfolgen muss und (B) doch einiges von der VitoConnect 100 abgerufen wird.

h0nIg, 2022-09-04:
@idstein wärst du bitte so nett und teilst kurz dein ESP script/aufbau und eine kurze Beschreibung was du mit deinem DEV Board gemacht hast?

h0nIg, 2022-09-11:
ich hab mir jetzt eine Bridge gebaut, womit man vcontrold und vitoconnect parallel betreiben kann und die fehlenden Werte kriegt: https://github.com/h0nIg/vcontrol-vitoconnect-bridge/

m1k3f15h, 2022-09-27:
Hy @idstein

Echt cooles Projekt :)
würd auch gerne mehr darüber erfahren :)

mfg
Mike

idstein, 2022-10-07:
> @idstein wärst du bitte so nett und teilst kurz dein ESP Sources/Aufbau und eine kurze Beschreibung was du mit deinem DEV Board gemacht hast? Braucht keine große Beschreibung, vorallem interessiert mich wie du den USB Traffic forwardest / interceptest

Aktueller Stand baut das ganze auf ESPHome bei mir lokal auf fürs Management, Updates, ein bisschen Security usw. Die Funktionalität die Werte konkret auszulesen habe ich noch nicht finalisiert, ich hoffe aber demnächst hier weiter zu kommen.

Funktion
- VitoConnect Passthrough (ohne Sniffing)
- ALE Energy Meter Simulator (Photovoltaik)

Roadmap
- Parse data points already accessed via VitoConnect
- Allow asking for specific data points independent of VitoConnect

Hardware
- ESP32 Dev Kit C V4 [Pinout]((https://github.com/openv/openv/files/9732491/ESP32_DevKit_C_V4_Pinout.pdf)
- Kabel Micro-USB
- IR-Emitter SFH4556 OSRAM 860nm 5mm 20° (https://www.ebay.de/itm/333428475405)
- IR-Receiver No-Name 5mm 940nm? Diode (https://www.amazon.de/dp/B08WCT6ZJ4)

Software
- ESPHome
- Fork VitoWifi (https://github.com/bertmelis/VitoWifi)

```
substitutions:
  # Unique device ID
  device_id: "heating01"
  device_name: "Heizung Optolink"

esphome:
  name: ${device_id}
  includes:
    - viessmann-vitoconnect/optolink.h
    - viessmann-vitoconnect/optolink.cpp

external_components:
  - source: github://epiclabs-io/esphome-modbus-server@master
    refresh: 60s
    components:
      - modbus_server

esp32:
  board: az-delivery-devkit-v4

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
    password: !secret wifi_ap_password

# Enable Home Assistant API
api:
  password: !secret api_password
  encryption:
    key: !secret api_encryption
# Enable over-the-air updates
ota:
  password: !secret ota_password

# Enable Web server
web_server:
  version: 2
  local: true
  port: 8080
  auth:
    username: !secret web_user
    password: !secret web_password

logger:
  baud_rate: 0 #disable logging over uart

[... 142 lines cut ...]
            return 0;
          return (id(meter_l1_power_supply)->state - id(heatpump_power_consumption)->state / 3) / 10;
      - start_address: 38 # REG_PHASE_1_REACTIVE_POWER
      - start_address: 39 # REG_PHASE_1_COS_PHI
      - start_address: 40 # REG_PHASE_2_VOLTAGE
      - start_address: 41 # REG_PHASE_2_CURRENT
      - start_address: 42 # REG_PHASE_2_POWER 1545 = 15,45 kVA
        on_read: |
          if (!api_apiserver->is_connected() || !id(meter_l2_power_supply)->has_state() || !id(heatpump_power_consumption)->has_state())
            return 0;
          return (id(meter_l2_power_supply)->state - id(heatpump_power_consumption)->state / 3) / 10;
      - start_address: 43 # REG_PHASE_2_REACTIVE_POWER
      - start_address: 44 # REG_PHASE_2_COS_PHI
      - start_address: 45 # REG_PHASE_3_VOLTAGE
      - start_address: 46 # REG_PHASE_3_CURRENT
      - start_address: 47 # REG_PHASE_3_POWER 1545 = 15,45 kVA
        on_read: |
          if (!api_apiserver->is_connected() || !id(meter_l3_power_supply)->has_state() || !id(heatpump_power_consumption)->has_state())
            return 0;
          return (id(meter_l3_power_supply)->state - id(heatpump_power_consumption)->state / 3) / 10;
      - start_address: 48 # REG_PHASE_3_REACTIVE_POWER
      - start_address: 49 # REG_PHASE_3_COS_PHI
      - start_address: 50 # REG_TOTAL_POWER
      - start_address: 51 # REG_TOTAL_REACTIVE_POWER
```

m1k3f15h, 2022-10-13:
Hy @idstein 

Echt ein MEGA-COOLES Projekt :)
Werde das gespannt weiterverfolgen und hoffentlich hab ich bald mal Zeit das "Nachzubauen".
Erstmals werde ich das ohne dem "Modbus-Energymeter-Simulator" machen... Soll aber mit der PV dann auch noch folgen.

Danke, dass du das Projekt mit uns teilst 😄 

Mfg
Mike

challo2018, 2022-12-07:
Hi @idstein, 
cooles Projekt ich verwende momentan die Viessmann API um über das Vitoconnect100 die Daten in mein Loxone Smarthome zu bekommen. Allerdings ist die API sehr eingeschränkt und teilweise jetzt auch kostenpflichtig. Wegen der Garantieverlängerung würde ich aber gerne das Vitoconnect weiter in betrieb lassen. Daher wäre es echt cool wenn man mit deiner einfachen Lösung das Vitoconnect weitervwenden kann und zusätzlich Daten lokal abfragen und auch setzen kann.  Bin mit ESP Home bisher noch nicht ganz firm. Kannst du kurz den aktuellen Stand nochmal etwas detaillierter beschreiben? 
Ich habe verstanden, dass die Daten zwischen dem IR Modul und dem Vitoconnect in beide Richtungen durchgereicht werden. 
Zusätzlich schleifst du PV Daten hier ein. 
Können auch Daten aus dem Traffic ausgelesen werden ? Vorlauftemperatur oder so? 
Wie läuft die Installation auf dem ESP32. ESPHome flashen und dann zusätzlich VitoWifi?

Vielen Dank!

m1k3f15h, 2023-01-16:
Hi @idstein,
Gibts zu dem Projekt schon was neues?

mfg
Mike

philippoo66, 2023-01-27:
Moin ihr!
hab ich das richtig verstanden - ihr habt eine Wärmepumpe mit Optolink? Wenn dem so ist, könnt ihr mir mal die 4 Bytes ab Adr. F8 und ein Byte an Adr. F0 sagen?
danke & beste Grüsse!
Phil

m1k3f15h, 2023-01-27:
Morgen,
Kann ich leider nicht, da bei mir noch alles original ist und ich nichts auslese.

mfg
mike

philippoo66, 2023-01-27:
aber da is eine Vitotronic drin, oder? da könnte dan im Menü unter 'Service' ganz hinten ein Punkt 'Software Information' oder so sein. Das Passwort dafür ist 'vimaster'

[image]

HW/SW Index sind das 3. und 4. Byte von F8, ich glaube die Zahl vor dem Punkt ist F0. Ich gehe davon aus, dass bei einer Vitotronic die ersten beiden F8 Bytes 20 CB sind...

philippoo66, 2023-01-27:
könntest du das mal checken beizeiten? ;-)

philippoo66, 2023-02-06:
wenn du den vom Jul 23, 2022 meinst - das ist eine HO2B. Meine ist eine HO2C. Die beiden Steuerungen unterscheiden sich nur minimal in der grafischen Darstellung. Diesen Bildschirm wirst du auch kennen?

[image]

Wenn man da nach rechts scrollt, gibt es 'Servicefunktionen', dahinter dann wieder rechts gibt es den Punkt Softwareinformation... 

Ist aber auch nicht so wichtig, hätte mich nur interessiert. Ich war nämlich etwas verwundert, dass ich in dem alten Vitosoft die DP Adresslisten für Wärmepumpen gefunden hatte, aber scheibar ist das ja erklärbar. Und wenn du die API benutzt, hast du wohl auch keine Verwendung für Optolink Adressen. Ich dachte nur, weil du den eingeschränkten Umfang bei der API moniertest....

m1k3f15h, 2023-02-07:
Nein den bildschirm kenne ich ned xD
Sieht anders aus bei mir...

[image]

Aktuell mach ich alles über die sche.... API.. würd aber gern auf einen esp32 als "optolink-proxy" wechseln, der dann auch genauer und hochauflösender ausließt

Mfg
Mike

challo2018, 2023-02-07:
@m1k3f15h : so würde ich es auch gerne machen da ich wegen der Garantie dass vitoconnect weiterlaufen lassen möchte....

m1k3f15h, 2023-02-07:
@challo2018 : Ja leider meldet sich @idstein aktuell nicht mehr.. Der hat ja da schon was zusammen gebracht :)
allerdings wär mir das ganze auf Tasmota-Basis lieber :D

mattjes04, 2023-03-23:
> ich hab mir jetzt eine Bridge gebaut, womit man vcontrold und vitoconnect parallel betreiben kann und die fehlenden Werte kriegt: https://github.com/h0nIg/vcontrol-vitoconnect-bridge/

Amazing work! I'm using a rpi1 with the same setup @DivisionDurchNull mentioned in his [post](https://github.com/openv/openv/issues/425#issuecomment-427666654). But as proxy script I copied your [bridge.py](https://github.com/h0nIg/vcontrol-vitoconnect-bridge/blob/c766bc3218bf944cf41bb30abfde5ce259d99800/bridge.py) and just changed the serial targets towards my setup. Vcontrol-Service also runs on the raspberry and queries the tcp-socket opened by the bridge.py. Vcontrol is then queried by the [Vcontrold-Addon](https://github.com/Alexandre-io/homeassistant-vcontrol) in Homeassistant which pushes the values to mqtt an makes them available in homeassistant. So far everything is working flawless. ViCare-App seems not to notice any interruption. Kudos to @h0nIg & @DivisionDurchNull

challo2018, 2023-03-23:
@mattjes04 : does this solution also allow to set values with vcontrold?

mattjes04, 2023-03-23:
> @mattjes04 : does this solution also allow to set values with vcontrold?

Jup, Warmwasserbereitung auf Anforderung funktioniert einwandfrei.

challo2018, 2023-04-20:
> > @mattjes04 : does this solution also allow to set values with vcontrold?
> 
> Jup, Warmwasserbereitung auf Anforderung funktioniert einwandfrei.

Hört sich auch nach einer guten Lösung an, allerdings wäre man hier dann nochmal auf rasperry basis.
Bei der Lösung von @idstein, läuft das ganze nur über ein ESP. Ich habe allerdings kein homeassistant, daher kann ich mit dem yaml file für espHome nicht so viel anfangen. Wenn man auf Basis von VitoWifi auf dem ESp32 noch ein passthrough für das vitoconnect einbauen würde, hätte man eine coole lösung. Vielleicht kann @idstein doch nochmal etwas unterstützen?

ckoeber83, 2023-08-30:
Hi,
suche auch eine Lösung die VControld und Vitoconnect im Parallelbetrieb zu betreiben und die Werte in den ioBroker zu bringen.
Hat da schon jemand was und kann eine Step by Step Anleitung machen, mit Hardware- und Software-Seite?

THX!

philippoo66, 2023-08-31:
Moin!

verstehe ich das richtig, dass du per Vitoconnect Viessmann deine Daten auslesen lassen möchtest, das aber mit vcontrold 'mitlesen' und auf MQTT posten?

bist du der koeber aus dem Viessmann Forum?

Grüsse!
Phil

ckoeber83, 2023-08-31:
```
Von: Phil <notifications@github.com> 
Gesendet: Donnerstag, 31. August 2023 03:49
An: openv/openv <openv@noreply.github.com>
Betreff: Re: [openv/openv] VitoConnect 100 mit Raspberry PI als Proxy (#425)

Moin!
verstehe ich das richtig, dass du per Vitoconnect Viessmann deine Daten auslesen lassen möchtest, das aber mit vcontrold 'mitlesen' und auf MQTT posten?
bist du der koeber aus dem Viessmann Forum?
Grüsse!
Phil

```
@philippoo66 ja genau der 😊

Ja ich will die Serielle Kommunikation mitschneiden und dann via vcontrold den [ioBroker.viessmann](https://github.com/misanorot/ioBroker.viessmann) Adapter anflanschen, um die Daten im ioBroker als Datenpunkte zu erhalten, zur Visualisierung in Grafana.

Mein bisheriges Grafana Dashboard mit Daten aus der [Viessmann API](https://developer.viessmann.com/start.html)

[image]

ggrote, 2023-08-31:
Wie hast du die Bedeutung und IDs der Datenpunkte rausgefunden? Das frage ich mich schon länger. Habe eine Pelletheizung von Viessmann und bisher noch keine Infos dazu gefunden. Würde aber auch ungern extra das Vitoconnect dafür kaufen und in Betrieb nehmen.
Dein Dashboard sieht super aus! Habe auch Grafana am Start, aber mit direkter Verbindung zu InfluxDB, wo über Node-Red Daten reingeschrieben werden.

ckoeber83, 2023-08-31:
@ggrote meine Dashboard läuft mit der [Viessmann API](https://developer.viessmann.com/start.html) und die Vitoconnect übermittelt die Daten in die API. 
Das ist noch keine lokale Geschichte mit [vcontrold ](https://github.com/openv/vcontrold) oder so.
Da man mit [vcontrold ](https://github.com/openv/vcontrold) aber mehr Datenpunkte auslesen kann, als über die [Viessmann API](https://developer.viessmann.com/start.html) und ich aber die Vitoconnect nicht missen möchte, ist der Wunsch nach einem Proxy, der die Serielle Verbindung zwischen WMP und Vitoconnect mitschneidet und mittels [vcontrold ](https://github.com/openv/vcontrold) und [ioBroker.viessmann](https://github.com/misanorot/ioBroker.viessmann) Adapter in den ioBroker bringt zur Weiterverarbeitung.

Die Unterstützen API Datenpunkte kann man im ioBroker schön mittels Objektbaum auslesen. ([Meine genutzten von Vitocal 333G](https://github.com/openv/openv/files/12483429/used_datapoints_Vitocal-333G.txt))
Bzw. stehen auch Data Points im [Viessmann Dev Portal](https://documentation.viessmann.com/static/iot/data-points)

Bzgl. Node-Red
Der Christoph Krzikalla hat mal im [Viessmann Forum](https://www.viessmann-community.com/t5/The-Viessmann-API/Using-Node-Red-to-visualize-ViCare-data/td-p/195606) auf seinem [Blog ](https://www.rustimation.eu/index.php/a-viessmann-api-und-node-red/)eine geile Viessmann API und Node-Red Anleitung gemacht. 👍

Auch ein schöner Weg 😊

ggrote, 2023-08-31:
Ich bevorzuge halt Cloudfreie Lösungen und habe gerne alles lokal in meinem Netzwerk, daher kommt für mich die Viessmann Api mit Hilfe von Vitoconnect nicht in Frage. Aber danke für den Hinweis mit dem Dev Portal. Muss ich mich mal anmelden, ggf. finde ich dort ein paar Datenpunkte für meinen Vitoligno 300-C Pelletkessel.

Node-Red ist wirklich super und mega vielseitig. Zusammen mit MQTT, Influx und Grafana meiner Meinung nach unschlagbar. Aber jeder wie er mag. ;)

philippoo66, 2023-08-31:
Moin ihr! :-)

@ggrote die Datenpunkte findest du mit aller Wahrscheinlichkeit so wie hier beschrieben https://github.com/philippoo66/ViessData21/blob/master/DataPoints_ReadMe.txt und schau auch noch was dazu unter https://github.com/philippoo66/ViessData21 steht, wie man möglicherweise auch noch an welche kommt, die evtl. nicht in der Liste sind. Dann noch einen Optolink Adapter gebastelt oder gekauft, und du hast die volle Bandbreite Datenpunkte ohne blöde Cloud und Heizung im Internet und ohne teures Vitoconnect.

@ckoeber83 die Lösung deines Anliegens steht doch eigentlich schon oben. [bridge.py](https://github.com/h0nIg/vcontrol-vitoconnect-bridge/blob/c766bc3218bf944cf41bb30abfde5ce259d99800/bridge.py) ist realisiert ja das, was ich im V Forum beschrieben hatte, und es scheint ja doch nicht zu Timeout zu führen, was ich befürchtet hatte.

beste Grüsse!

ggrote, 2023-08-31:
> @ggrote die Datenpunkte findest du mit aller Wahrscheinlichkeit so wie hier beschrieben https://github.com/philippoo66/ViessData21/blob/master/DataPoints_ReadMe.txt und schau auch noch was dazu [...]

Wahnsinn! Danke! Hoffe da ist auch was für die Vitoligno 300-C bei.

philippoo66, 2023-08-31:
um die Gerätekennung und ggf. ecnsysDeviceIdent auszulesen brauchst du natürlich schon einen Kommunikationsadapter... hast du einen?

ggrote, 2023-08-31:
Noch nicht, aber alles für da. Muss quasi nur noch zusammengebaut werden. Dafür fehlt mir bei gefühlt 500 anderen Projekten im und ums Haus aber die Zeit. Trotzdem verfolge ich hier alles.

ckoeber83, 2023-09-04:
Hi,

ich kann die [vcontrol-vitoconnect-bridge](https://github.com/h0nIg/vcontrol-vitoconnect-bridge/tree/main) nicht öffnen.
Ich erhalte:
```
./bridge.py", line 5, in <module>
    import serial
ModuleNotFoundError: No module named 'serial'

```

Serial devices habe ich aber euf meine entsprechenden angepasst:
**usb 1-1.3: cp210x converter now attached to ttyUSB0 > vitoconnect
usb 1-1.4: cp210x converter now attached to ttyUSB1 > Vitotronic 200, Typ WO1C Regelung**

```
## open devices
ser_heating = serial.Serial("/dev/ttyUSB1",
                     baudrate=4800,
                     parity=serial.PARITY_EVEN,
                     stopbits=serial.STOPBITS_TWO,
                     bytesize=serial.EIGHTBITS,
                     timeout=1)
ser_vitoconnect = serial.Serial("/dev/ttyUSB0",
                     baudrate=4800,
                     parity=serial.PARITY_EVEN,
                     stopbits=serial.STOPBITS_TWO,
                     bytesize=serial.EIGHTBITS,
                     timeout=1)

```

Fehlt mir hier ein Modul in phyton?
THX!