openv/openv #426: delete errors / set reset via vcontrold

passuff, 2018-10-05:
I'm using vcontrold for my vitodens 200 (w2b2). I was wondering if someone has a command to 

1. delete the error messages
2. reset the device after an error appeared 

My device id is 20CB // protocol is P300

DBa2016, 2019-02-09:
Do you have by any chance the config files for this setup?

richdyer2000, 2020-10-12:
Any joy on this one? We tend to get a drop in Gas Pressure in our neighbourhood early in the morning when everyone's boilers come on. This causes a fault when our boiler tries to ignite. I've implemented some tricks to make a boiler activation in this time window unlikely, but I'd really like to do something a bit more robust like a single retry say 3 hours later.