openv/openv #428: V200KW2 Probleme mit setTimerM

kleinemeise, 2018-10-23:
Hallo,

kann mir jemand sagen, warum meine Timer nicht richtig geschrieben werden:

```
vctrld>setTimerM2Di "05:00  07:00 14:00 16:00"
OK
vctrld>getTimerM2Di
1:An:00:00  Aus:07:00
2:An:14:00  Aus:16:00
3:An:--     Aus:--
4:An:--     Aus:--
```

Auslesen per getTimer geht sauber:

```
vctrld>getTimerM2Di
1:An:05:00  Aus:07:00
2:An:15:00  Aus:22:00
3:An:--     Aus:--
4:An:--     Aus:--
```
Was mache ich hier falsch? Danke für eure Hilfe

drtp, 2019-01-02:
Versuche es mal mit 

vctrld>setTimerM2Di "05:00  07:00 14:00 16:00 00:00"

Sofern du nicht sämtliche Timer-Werte setzte, musst du die Eingabe mit "00:00" abschließen.

kleinemeise, 2019-01-03:
Hallo drtp,

das klappt für den 2 Timer Wert super. Beim ersten gibt es aber noch ein Problem:

`vctrld>setTimerM2Di "05:00 07:00 14:00 16:00 00:00"
OK
vctrld>getTimerM2Di
1:An:00:00  Aus:07:00
2:An:14:00  Aus:16:00
3:An:--     Aus:--
4:An:--     Aus:--
`
Hast du noch eine Idee? Ich benötige eigentlich immer nur 2 Timer

drtp, 2019-01-03:
Hast du den Zeitpunkt "00:00" im ersten Timer auch mit "0:00" übergeben? Hier unbedingt nur eine einzige Null für die tatsächliche Zeitübergabe verwenden. "00:00" dient dann zum Terminieren, wenn nicht alle acht Timerwerte gesetzt werden. Zeiten, wie z.B. "05:00" müssen dagegen immer mit vorangestellter Null geschrieben werden.

drtp, 2019-01-03:
Es kommt übrigens immer eine Fehlermeldung "FRAMER: unexpected length 8 08" bei der Übergabe, aber die kann ingnoriert werden, da die Zeitpunkte korrekt übernommen werden.

drtp, 2019-01-03:
Sorry, hab nen kleinen Fehler gemacht. Uhrzeiten vor 10:00 Uhr dürfen nie mit einer vorangestellten Null gesetzt werden. Versuch's mal bitte mit

setTimerM2Di "5:00 7:00 14:00 16:00 00:00"

Gruß,

Thorsten

kleinemeise, 2019-01-03:
traumhaft - ohne Anführungszeichen funktioniert es :-)

```
vctrld>setTimerM2Di 5:00 7:00 14:00 16:00 00:00
OK
vctrld>getTimerM2Di
1:An:05:00  Aus:07:00
2:An:14:00  Aus:16:00
3:An:--     Aus:--
4:An:--     Aus:--
```

drtp, 2019-01-03:
Freut mich. Jetzt suche ich noch eine Möglichkeit zum Setzen der Systemzeit. Alles, was ich bisher diesbezüglich probiert habe, hat leider nicht funktioniert.

B4n4n3, 2019-01-04:
Hallo drtp,

vielen Dank für die Hilfe! So hat es bei mir nun auch funktioniert!

gruß B4n4n3