openv/openv #429: Installation auf Raspi

micklafisch, 2018-11-10:
Hallo zusammen,
ich versuche gerade anhand der Anleitung (https://github.com/openv/openv/wiki/vcontrold-mit-Raspberry-Pi) auf meinem ZeroW vcontrold zu installieren. Die Anleitung wurde Punkt für Punkt abgearbeitet (auf aktuellem Rasbian strech lite), aber bei Punkt "make" kommt als Ausgabe 

> make: Nothing to be done for 'all'

Weiter kommt dann bei "make install" folgendes zurück:

> make[1]: Entering directory '/home/pi/openv/vcontrold-code/vcontrold'
>  /bin/mkdir -p '/usr/local/bin'
>   /usr/bin/install -c vclient vsim '/usr/local/bin'
> /usr/bin/install: cannot remove '/usr/local/bin/vclient': Permission denied
> /usr/bin/install: cannot remove '/usr/local/bin/vsim': Permission denied
> Makefile:321: recipe for target 'install-binPROGRAMS' failed
> make[1]: *** [install-binPROGRAMS] Error 1
> make[1]: Leaving directory '/home/pi/openv/vcontrold-code/vcontrold'
> Makefile:664: recipe for target 'install-am' failed
> make: *** [install-am] Error 2

Ich komme leider nicht weiter, hat das jemand schonmal gehabt?
Danke und Gruß 
Michael

Jojo-A, 2018-11-14:
Moin,
Besteht das Problem noch?

micklafisch, 2018-11-14:
Hi,
leider ja. Ich hab das System nochmal komplett neu aufgesetzt, leider ohne Erfolg.
Aktuell „ruht“ es, da ich keine Idee habe.

micklafisch, 2018-11-14:
Hallo Jojo,
ich kann es nicht glauben. Scheinbar hatte die SD Karte nen Hau weg. Neue SD Karte, gleiches Prozedere und zack, vcontrold läuft. Danke für die Ideen.
Gruß

closed 2018-11-14 by micklafisch