openv/openv #432: vitotronic 200 KW4

sanktnelson, 2018-12-08:
Hallo zusammen,
ich habe mir das Optolink-Kabel gekauft, da ich mit der o.g. Heizungssteuerung reden wollte. Inzwischen weiss ich auch, dass sie nicht unter den unterstuetzten Geraeten ist :( 
Dementsprechend bekomme ich keinerlei link mit vcontrold Zustande.
kann die v200KW4 ueberhaupt per Optolink gesteuert werden? Wenn ja, was muesste ich dafuer tun?
gruss
Felix

Nockapatzl, 2018-12-12:
Hallo Felix!

Ich bin Anfangs mit den Programmen nicht ganz zurecht gekommen. Mittlerweile habe ich das Wiki einigermaßen durchgeschmökert, folgendes hat bei mir funktioniert:
Treiber für das Optolink-Kabel installieren (normalerweise bekommt man dann einen Eintrag im Gerätemanager unter COM/LPT "CP210X USB To UART Bridge - COM7) .

Nach der Installation ein Terminalprogramm installieren (z.B. HTerm), und einfach mal am Port mit den angengebenen Einstellungen lauschen (4200, 8 Datenbits, Even Parity, 2 Stoppbits, ohne Handshake)-->es sollte zyklisch 0x05 empfangen werden.
Weitere Details findest Du hier: https://github.com/openv/openv/wiki/Protokoll-300 
ACHTUNG: Bei den beiden Beispielbefehen ist meiner Meinung nach die Checksumme falsch (die richtigen habe ich gerade nicht parat, kann ich aber nachliefern)!

Ob die Anlage per Optolink gesteuert werden kann, weis ich nicht. Werte (Außentemp, etc.) abfragen sollte auf alle Fälle gehen.

Ahja, Liste mit bisher bekannten Befehlen, nach Steuerung gegliedert:
https://github.com/openv/openv/wiki/Adressen (V200KW4 ist nicht dabei, soweit ich das sehen konnte).

Gruß Bernhard

sanktnelson, 2018-12-15:
Vielen Dank erstmal für die tips, besonders die Linksammlung ist Gold wert!
v200KW4 ist voll steuerbar, man sollte nur im vcontrold auch das KW-protokoll einstellen - die default-einstellung ist GWG.