openv/openv #434: OptoPi v2

thoschi, 2018-12-29:
Hallo zusammen,
ich habe in den letzten Jahren ein altes Haus renoviert und mit ziemlich viel Elektronik vollgestopft. Heizung wird befeuert von einer Viessmann Vitodens 200. 

Erstmal finde ih es toll, dass es eine offene Alternative zum doch etwas zweifelhaften (WLAN?!) herstellereigenen Anbindungskonzept gibt.

Ich finde die OptoPi-Variante im Moment reizvoll (und würde ihn über einen per PoE betriebenen Raspberry Pi per Kabel an unser Hausnetz anschließen).

Gibt es eine Möglichkeit für eine (Sammel)Bestellung? Es darf ruhig etwas kosten - aber wenn ich das selber versuche zu löten, wird das vermutlich nichts - gerade mit SMD hab ich keinerlei Erfahrung.

Für jede Hilfe vielen Dank!

Thomas

Arty68, 2019-09-16:
Hallo thoschi.
Falls du noch interresse hast. 
Bin ich auch interessiert für  eine SammelBestellung.
Sorry für mein Schlechte Deutsch (ich bin Elsassisch (aus Elsass (Frankreich)) 
Aber ich wohne nicht Weit von Deutschland (so kann ich auch in Deutschland liefern lassen)
Gruss
MFG
Arty68

usmansuleman, 2019-10-25:
Hello:
I would also like to get an Optopi v2.
How do I do it?
Regards,
...Usman from NY USA