openv/openv #436: Trimatic MC / Stecker 58

funkyluke, 2019-01-13:
Hallo,

ich versuche im Moment meine steinalte Trimatik MC ins IoT Zeitalter zu holen und bin auf dieses Projekt gestoßen.
Ich habe hier eine Comfortrol FB, die über Stecker 58 angebunden werden kann und die ich mal belauscht habe -> das sieht dann so aus:
https://imgur.com/a/ioTpYsd
Das Protokoll scheint mit 1200 Baud zu laufen.
Wird am Stecker 58 auf eine ähnliche weise wie über Optolink (hab ich nicht) kommuniziert, sprich kann man die Informationen auf diese Schnittstelle übertragen?
Falls nicht, gibt es zu diesem Protokoll bereits Informationen?

Grüße,
Funky Luuke

qwert089, 2019-06-10:
wenn man mal beim Nachfolgemodell untersucht  wie die IR Schnittstelle an der Platine der Vitotronic x00 KB1 hängt. 
vielleicht könnte man an der Trimatik die IR Leiterplatte test-weise  nachrüsten??
VG

mbr1989, 2021-01-29:
Push ... Trimatik B ... Bedieneineheit S ... 7450155