openv/openv #439: Vitoconnect 100 Opto1 Kabel Orig. an USB_Duo Sniffer betreiben

Camel2k, 2019-01-31:
Hallo,

laut der Doku für den USB Sniffer habt Ihr ein selbst gebautes OPTO Kabel (angelötet) benutzt.
Ich würde gerne mein Original (USB-A) benutzen, wie ist da die Belegung des USB Steckers.

Lg
Camel2k

GolterU, 2019-02-04:
Ich verstehe die Frage nicht ganz: 
Der USB-Sniffer ist genauso aufgebaut wie ein Viessmann-Original-Optolink-Adapter. Er hat halt nur zwei USB-Schnittstellen in der Form von USB-Mikro-Buchsen. Steckt man dort ein Standard-Kabel an (mit USB-A-Stecker auf der einen und USB-Mikro-B-Stecker auf der anderen Seite), verhält sich der Adapter wie der Original-Adapter von Viessmann. 
Siehe dazu auch [hier](https://github.com/openv/openv/wiki/Bauanleitung-USB_Duo-Sniffer#aufbau).

Gruß
Uwe

Camel2k, 2019-02-04:
Moin Uwe,
ich meine den Connector XS3, dort habt Ihr eine Leitung angelötet oder verbunden die zur Therme geht. Dort wo Ihr die Opto-Platine 3pol. angeschlossen habt, da muss ich doch das Original Viessmann Kabel "V" anschließen müssen?!  Das Original Kabel hat aber doch einen USB Stecker. Und das ist die Frage wie ich das angeschlossen bekomme. 
An den beiden USB-Mikro (Modul1 u. Modul2) würde ich dann den Raspi und das Original Vitoconnect 100 angeschlossen.
Oder hab ich da was falsch verstanden?

Gruß
Kai

GolterU, 2019-02-05:
Hallo Kai,
hier liegt tatsächlich ein Missverständnis vor:
Das Original-Viessmann-Kabel entspricht dem **kompletten** USB-Adapter Duo/Sniffer, aber natürlich nur mit **einem** USB-Anschluss. Am Steckverbinder XS3 meines Adapters werden nur die optoelektronischen Bauelemente angeschlossen, das ist kein USB-Anschluss. Die Belegung des XS3 ist im [Stromlaufplan](https://godemann.de/test/git/V3/OptolinkDuo_SCH.pdf) dargestellt.
Es gibt daher zwei Alternativen:

- Das Original-Viessmann-Kabel mit dem USB-A-Stecker entweder an Vitoconnect oder Raspi anschließen, das andere Ende mit dem "V" an die Heizung stecken.

- Der USB-Adapter Duo/Sniffer hat zwei USB-Anschlüsse, also einen für Vitoconnect (um die Heizung mit dessen Daten zu füttern und die Antworten aufzunehmen), den anderen für den Raspi (um die gesendeten Daten an die Heizung und/oder die zurücklaufenden Daten mitzuschneiden). An XS3 hängt über das dreipolige Kabel ein nachempfundenes Viessmann-V für den Betrieb an der Heizung (siehe [hier](https://camo.githubusercontent.com/14953a796b586366cf4de61cf18078ec15f467c5/68747470733a2f2f676f64656d616e6e2e64652f746573742f6769742f56332f4f70746f6c696e6b5f44756f5f4f70746f706c6174696e652e4a5047)).

Zusammengefasst heißt das, Du brauchst den kompletten USB-Adapter Duo/Sniffer, um den Datenverkehr wie beschrieben mitschneiden zu können.

Gruß
Uwe

Camel2k, 2019-02-05:
Hallo Uwe,
vielen Dank für die ausführliche Antwort. Nun ist Licht im dunkeln ;)
Alles klar dann werde ich mir das Optokabel mal nachbauen, die CJMCU-Module habe ich schon.

Lieben Dank 
Kai

closed 2019-02-05 by Camel2k