openv/openv #442: Vitodens 200 WB2 Adressen und Issues

rbike, 2019-02-19:
Nachdem unsere Vitodens 200 WB2 die 2003 installiert wurden unregelmäßig alle paar Tage auf Störung ging habe ich nach den Anleitungen hier eine Datenlogger über die Optlink Schnittstelle gebaut. Danke an alle die hier beitragen.
Die Addressen für die Vitodens 200 WB2 mit Comfort control habe ich aus verschiedenen Quellen zusammen gesucht und auch selbst einige neue Addressen ausgetestet. Die sind im beiligenden Excel File aufgelistet. Einige sind noch in rot, daher ich sie bisher noch nicht überprüft habe.

[addr_2053.txt](https://github.com/openv/openv/files/2880236/addr_2053.txt)

Das GWG Prookoll unterscheidet zwischen verschiedenen Addressbereichen (Phyiscial, Virtual, Port, EEPROM, XRAM, KM BUS, Bedieneinheit), die aber teilweise bei der Vitodens200 WB2 nicht alle benutzt werden wie die Bedieneinheit Addressen (siehe Excel File). Ich dachte, dass man darüber auf die Comfort Control zugreifen kann, aber das ist nicht der Fall.

Ich habe auch versucht auf Addressen zu schreiben (Physical Addressbereich). Wenn ich den Wert direkt nach dem Schreiben ausgelesen habe war die Änderung auch sichtbar, aber nach wenigen Sekunden wurde er wieder auf den alten Wert zurück gesetzt. Für Werte wie die Tages- und Nahtsolltemperatur die über Drehregler eingestellt werden kann ich das verstehen, aber nicht bei Neigung und Niveau der Heizkurve die über die Tastatur eingegeben werden. Hat hier jemand ähnliche Erfahrungen gemacht?

DBa2016, 2019-02-28:
Da ich bisher die Werte noch gar nicht erst auslesen kann... Kannst Du evtl. die XML-Dateien hochladen?

rbike, 2019-02-28:
Hier die vito.xml und vcontrol.xml angepasst nach der obigen Adressliste. Ich habe hier Einträge für andere Modelle ud Protokolle entfernt um übersichtlichere Dateien zu haben.
[vcontrold-xml.zip](https://github.com/openv/openv/files/2914982/vcontrold-xml.zip)

Arty68, 2020-01-02:
Vielen Dank das funktionniert auch für die Heizung vitodens 100 wb1.
Es werde auch ein bisschen "scan machen"

schwatter, 2020-10-10:
> Hier die vito.xml und vcontrol.xml angepasst nach der obigen Adressliste. Ich habe hier Einträge für andere Modelle ud Protokolle entfernt um übersichtlichere Dateien zu haben.
> [vcontrold-xml.zip](https://github.com/openv/openv/files/2914982/vcontrold-xml.zip)

Vielen Dank! Damit habe ich eine Grundconfig für meine Vitodens 222 WS2 aus 2003 auf der ich aufbauen kann.
Temps auslesen klappt. Jetzt kann ich alles in Ruhe durchgehen.