openv/openv #443: ESP32 Feather Huzzah32 Proto Wing: "druckbare" vito.stl auch im Form Factor Proto Wing?

achill-es, 2019-03-10:
Ich habe "https://github.com/openv/openv/wiki/Bauanleitung-ESP32-Adafruit-Feather-Huzzah32-and-Proto-Wing" nachgebaut und konnte auch die "https://github.com/openv/openv/wiki/files/vito.stl" wunderbar ausdrucken.
Nur bin ich mit dem Form Factor nicht zufrieden: Warum kreisrund und nicht passend zum Proto Wing (also rechteckig)?
Dann hätte man es direkt auf das Proto Wing schrauben können (natürlich mit passenden Abstandshaltern).
Noch besser wäre ein passendes Gehäuse für die gesamte Konstruktion wie auf dem Bild https://github.com/openv/openv/wiki/files/IMG_4211.jpg

Schnup89, 2019-03-15:
Eventuell nicht ganz das was du suchst, aber ggf. eine Grundlage:
https://raw.githubusercontent.com/Schnup89/OpenV_NodeMCU/master/OptoLink-Adapter_small.jpg

Ich suche morgen die STL und stelle diese hier online:
https://github.com/openv/openv/wiki/Bauanleitung-NodeMCU-WIFI---MQTT---HomeAssistant

callimero, 2019-03-16:
Meine Variante

[image]

[STL.zip](https://github.com/openv/openv/files/2973573/STL.zip)