openv/openv #445: neue version im SVN für Raspberry PI

XerXes777, 2019-04-09:
BITTE BITTE BITTE mach eine neue version im SVN !!!

Ich brauche die neuste Version für einen Raspberry PI.

Wenn ich die aus der Anleitung nehme:
svn checkout svn://svn.code.sf.net/p/vcontrold/code/trunk vcontrold-code
Läuft zwar stabil, aber ich kann nicht auf eine 2te BYTE POSI abfragen und dies als ein ENUM korrekt wieder geben!!! Da mein Ergebins immer eine 2 BYTE ausgabe ist und das 2te BYTE immer anders ist.

Wenn ich nach der Anleitung gehe und danach nur die beiden Dateien:
vclient und vcontrold tausche aus dem Archive von:
https://github.com/openv/vcontrold/releases
vcontrold_0.98.7-99_armhf.zip
Bekomme ich zwar eine neuere Version,
mit dem ich dann besser arbeiten kann,
aber diese ist sehr instabil und stürzt oft ab!!!

Ich würde gerne die neuste Version für meinen PI Kompilieren,
ABER ich weiß nicht wie!!

BITTE BITTE BITTE brauche eine genaue Anleitung!!!

philverh, 2019-04-30:
From memory, here is how to get the latest build compiled on your PI.
In a terminal window on your PI:

0) Make sure your PI is up to date by issuing (in this order)
	`sudo apt-get update`
	`sudo apt-get upgrade`

1) On the PI, create a directory to hold the openv project. Let's assume you are in /home/pi
	`mkdir openv` 
	`cd openv`
2) get latest code
	Use Git, or checkout with SVN, using the URL https://github.com/openv/vcontrold.git, to get a copy of the code in a directory. I used "vcontrold" in the example below.
		 `svn checkout https://github.com/openv/vcontrold.git vcontrold `
3) Build the code
	Refer to the build instructions for Linux at https://github.com/openv/vcontrold/blob/master/doc/INSTALL.md
	If one of the required packages is missing on your PI, a quick
 `sudo apt-get install missing_package_name` is your friend.
	In summary, once all the pre-requisites are available, the build is quite straightforward. Start in the directory where you downloaded the code in the previous step :
		`mkdir build` (this has to be done only once when you first try to build)
		`cd build`
		`cmake ..`
		`make`
		`sudo make install`
	
That's it. Assuming you did not get errors during the build, your new code is ready to go. If you already had a version of vcontrold running, dont forget to ensure the latest code is running. You may want to `sudo systemctl restart vcontrold` 
When you issue `vcontrold -version` you will get back the running version.

speters, 2019-06-21:
The vcontrold project on Sourceforge is not maintained by the openv team.

Please use https://github.com/openv/vcontrold/ repository for up-to-date code and documentation.
There are also binary releases available at https://github.com/openv/vcontrold/releases

---

Das vcontrold Projekt wird vom OpenV team nicht gepflegt.

Bitte aktueller Code und Dokumentation ist unter https://github.com/openv/vcontrold/ zu finden.
Dort werden auch Binärpakete unter https://github.com/openv/vcontrold/releases veröffentlicht.

closed 2019-06-21 by speters