openv/openv #447: getPumpeDrehzahlSolar always returns 0% on Vitodens 200

3s1d, 2019-05-23:
[Not sure whether German or English is preferred. Feel free to answer my in one of them...]

Hi,
i got a new Vitodens 200 with a solar collector on the roof. Device ID 20CB seems to work. Everything is fine, except one thing. As a measure how good the solar collector currently performs I'd like to see the status of the solar pump in percent. However, getPumpeStatusSolar reports only 0 or 1, if it's off or on respectively. And getPumpeDrehzahlSolar always reports 0%, no matter what. I am using this XML: https://github.com/openv/vcontrold/blob/master/xml/300/vito.xml .
On the unit's display the 'Drehzahl Solarpumpe' is depicted in percent. So there seems to be the value stored but I'm not sure how to retrieve it.

Thanks in advance!
Juergen

bertmelis, 2019-05-23:
I'm going to follow this topic as I didn't find it myself.

My solar system has a flow meter with pulse output which I currently monitor. (and effectively detected a failure: there was an air bubble which caused the pump to stall)

philverh, 2019-06-14:
have you tried modifying the your vito.xml as follows 
`<command name="getPumpeDrehzahlSolar" protocmd="getaddr">`
`  <addr>6552</addr>`
`  <len>2</len>`
  `<unit>PR2</unit>`
 ` <description>Ermittle die Drehzahl der Internen Pumpe</description>`
 `</command>`
In the current state of the vito.xml you define the unit for the drehzahl value as "RT", which is the same as the status... 

I have not looked into the detailed syntax of the vito.xml file. I have defined just what I needed for my device, and not tried to come up with variable definitions based on device type. 

There is actually an open proposal to reorg the .xml definitions and make them easier to maintain.

stedon81, 2021-02-20:
Has anyone succeeded in finding the right address for the pump speed of the solar system?
Address 6552 indeed only returns a 1 or 0 for on and off, respectively.

speters, 2021-02-20:
> Has anyone succeeded in finding the right address for the pump speed of the solar system?
> Address 6552 indeed only returns a 1 or 0 for on and off, respectively.

I can not find any hints in the configuration files from the [Vitosoft demo](https://connectivity.viessmann.com/de-de/mp_rt/vitosoft.html) that the solar pump rpm is reported.
Follwing the address scheme of the heating circuit pump, you could give 0x6563 (2 bytes, reporting as percentage) a try. But my guess is that the solar pump rpm is not reported. Maybe you have a chance via KM bus.