openv/openv #448: setaddr hysterese kuehlgrenze

promy, 2019-06-01:
Im trying to change the hysteresis for the natural cooling on my heatpump (vitocal 300 using the P300 protocol )
This is parameter 7004.

in vito.xml i have:
---------
<command name='setDeltaKoeling' protocmd='setaddr'>
<addr>7004</addr>
<len>2</len>
<unit>UT</unit>
<description>Temperatuurdelta koeling INSTELLEN</description>
</command>
-----------

I changed vcontrold.xml for the fault in the SETADRR
---------
<macro name='SETADDR'>
<command>SEND 00 02</command>
</macro>
------------
However i'm not succeeding to change the parameter: this is the output on the Rpi console:(the same as in the log)
--------
 vclient -h 127.0.0.1:3002 -c 'setDeltaKoeling 10'
SRV ERR: >FRAMER: addr was still active FE06
>FRAMER: Command send
>FRAMER: unexpected length 2 02
Fehler recv, Abbruch
Fehler beim ausfuehren von setDeltaKoeling 10
setDeltaKoeling 10:
server error
---------

The strange thing is that the parameter IS changed on the heatpump, but, no matter what i try to set it to (10 or  4 or  3) it is always set to 12.5

Any idea's on what i'm doing wrong?

philverh, 2019-06-03:
@promy Interesting question, as I have the same equipment running (VitoCal 300). 
From what I could read, the default for the hysteresis is 4.0°, with limits at 0.0° and 20.0° and a step of 0.1°K. The expected value is a signed (???) integer, on 2 bytes. At first sight, the definition in your vito.xml looks OK, but I have something totally different in vcontrold.xml

`  <command name="setaddr">
        <send>SEND 00 02 $addr $hexlen BYTES;RECV 1 SR</send>
   </command>`

Could you publish your vito.xml and vcontrold.xml files (attach them in a reply) to check?  

I'll do some experimentation. 

Goede namiddag.

promy, 2019-06-03:
I found the problem: the vcontrold and vclient version was to old. I tested it with a different Pi running the latest version and it works.

@philverh : in my vcontrold.xml i have a similar line:

```
<command name="setaddr">
		     <send>SETADDR $addr $hexlen;SEND BYTES $unit;RECV 1 SR</send>
		</command>
```
combined with the definition of SETADDR i copied in the previous post it's exactely the same

philverh, 2019-06-04:
Yep, the setaddr command together with the SETADDR macro gives the same result as what I quoted. And I tested the address 7004 on my heatpump. It works. I successfully read and set different values, and verified by reading them back. 

The default value for parameter 7004 is 4°K. 
My understanding is : if the outside temperature exceeds the requested inside temp (for the circuit on which the cooling is defined) plus the hysteresis value, then the cooling is activated. As an example, if I have 20°C set on the thermostat inside, and have 4°K set for parameter 7004, then the cooling should start (after some defined time) if the outside temperature exceeds 24°C, and try to bring the inside temp back to 20°C.
 
I'm using the same information source as you... One of the demo files lists all the values available for device type 2046. The others define the values and the R or RW status. I don't have more, and I believe there is nothing more... I'll have a look at the files you attached, and compare with mine. We may have differences based on the installation options.

promy, 2019-06-25:
@philverh : did you find some time to compare the files? I'm still looking for some of the stuff in the NC box:
- position of the 3way valve (type "met mengklep")
- position of the valve marked with 2times open and 2 times closed (inside the NC box, the one with the round top)