openv/openv #450: Vctrld send ERROR: ERR: command unknown

XerXes777, 2019-06-11:
Ich bekomme immer wieder diese Meldung bei iobroker im Log.
Und meine Daten bei den Objekten stimmen dann nicht mehr (siehe Bild):

Was mache ich falsch ?

[image]

[viesman_adapter.log](https://github.com/misanorot/ioBroker.viessmann/files/3275703/viesman_adapter.log)

[vcontrold.log](https://github.com/misanorot/ioBroker.viessmann/files/3275705/vcontrold.log)

[vcontrold.xml.txt](https://github.com/misanorot/ioBroker.viessmann/files/3275734/vcontrold.xml.txt)

[vito.xml.txt](https://github.com/misanorot/ioBroker.viessmann/files/3275736/vito.xml.txt)

philverh, 2019-06-11:
Hi, assuming your viessmann_adapter.log file is only an extract from the whole log, it seems the time stamps of the errors match the sending of a "heartbeat" command.
And there is no "heartbeat" command defined in the vito.xml file... which generates an error. 
Try defining a heartbeat as a read of address 0x779C (one byte, integer)

XerXes777, 2019-06-12:
hi,
thank for your answer.
sorry but i dont know what the "heartbeat" command is.
i dont use this command. how i can find out, where the command is coming from?

philverh, 2019-06-13:
A heartbeat is a way to ensure the "other" party is still alive, by sending it a dummy command and watching for a reply (like "ok") to come back. If the reply is not coming back in time, you can conclude the other party was somehow disconnected. 
I guess iobroker requires this to ensure the communication with vcontrold is still active.  I briefly looked at the parameters your Viessmann boiler has, and as I said above there seems to be a heartbeat value stored at the address 0x779C. That is why I suggested reading the value, to satisfy iobroker, ut then you may need to instruct iobroker to accept whatever value is returned by your heating installation. 
I don't know what this heartbeat is used for on the viessmann side. (I have a heatpump which does not have this parameter.)
Or maybe you can remove the heartbeat from the iobroker config ? 

Sorry, I don't have more info, as I don't use this same type of heater or iobroker.