openv/openv #451: Kommunikationsprobleme Optolink USB

patricf82, 2019-06-20:
Guten Morgen,

ich habe einen Optolink USB und eine Vito 204D Steuerung.
Wenn ich unter Windows das Programm VitoGraph starte bekomme ich schön die Momentantemperaturen übermittelt.
Mit dem Program VitoTest bekomme ich auch eine Verbindung um Parameter auszulesen oder zu senden (sowohl mit 300er als auch mit KW-Protokoll). 

Dabei steht in VitoTest "Try Type2" / "online Type2"

Wenn ich allerdings ein anderes Terminalprogramm (mehrere getestet, z.B.  Putty) verwende, bekomme ich keine Verbindung. Die Schnittstelle wird geöffnet (4800,8,E,2) aber ich bekomme weder beim Senden noch periodische eine Antwort. Die LEDs am Optolink blinken periodisch für den Empfang sowie beim Senden.

Was mache ich falsch?
Ich möchte gerne eine eigene VB Anwendung mit Visualstudio bauen, da es hier für Windows offenbar noch keine brauchbare Oberfläche für Wärmepumpen gibt.

Vielen Dank.

speters, 2019-06-20:
> Wenn ich allerdings ein anderes Terminalprogramm (mehrere getestet, z.B. Putty) verwende, bekomme ich keine Verbindung. Die Schnittstelle wird geöffnet (4800,8,E,2) aber ich bekomme weder beim Senden noch periodische eine Antwort.

Ein Terminalprogramm stellt "lesbare" Schriftzeichen dar und versteht zudem je nach Terminalemulation unterschiedliche Steuerzeichen (z.B. zur Cursorpositionierung),

Das [Protokoll](https://github.com/openv/openv/wiki/Protokoll-300), welches über die Optolink-Schnittstelle geht, ist aber ein binäres Protokoll, welches alle Bytes von 0x00 bis 0xff verwendet.
So ist z.B. das ENQ-Zeichen 0x05, welches periodisch gesendet wird, nicht vom Terminalprogramm darstellbar. Ebenso werden von Dir über die Schnittstelle gesendete Zeichen nicht dargestellt, es sei denn, Du schaltest lokales Echoing an.

Kurzum: Ein Terminalprogramm ist das falsche Programm, um low-level mit der Optolink-Schnittstelle zu kommunizieren.
Du benötigst entweder eine Programmbibliothek, einen Dämon wie vcontrold oder musst ein eigenes Interface stricken.

patricf82, 2019-06-21:
> Das Protokoll, welches über die Optolink-Schnittstelle geht, ist aber ein binäres Protokoll, welches alle Bytes von 0x00 bis 0xff verwendet.

Ah so, vielen Dank!. Habe es jetzt mit "HTerm" hinbekommen. Das kann Hex/dec/bin ....
Jetzt muss ich es nur noch schaffen mittels Visual Studio die Schnittstelle ordentlich anzusprechen.
Ich werde mir ein Nullmodem Kabel bauen und dann erstmal mittels zwei Rechnern testen was auf er "anderen  Seite" ankommt.