openv/openv #452: Ist es möglich einen USB Optolink zu beziehen

martinschm, 2019-07-12:
Hi,
ich würde gerne meine Heizung auslesen, bin aber was das Löten angeht nicht versiert.

Gibt es eine Möglichkeit einen USB Optolink fertig zu beziehen ?
Die in der Anleitung (https://github.com/openv/openv/wiki/Bauanleitung-USB) genannte Adresse habe ich bereits angemailt. Leider kam noch keine Antwort, ich vermute der Autor ist stark beschäftigt und kann aktuell nicht helfen.

Daher suche ich alternative Quellen.

Ciao
Martin

bertmelis, 2019-07-12:
I have one but don't use it anymore (have the esp8266 version now).

If you want you can have mine (at the expenses of my costs).

martinschm, 2019-07-14:
Hi Bert,
I am interested. Will contact you directly.
Best
Martin

bertmelis, 2019-07-14:
Can you find my e-mail of do you need some hints?

It involves my username on GH and a popular service from Google...

bertmelis, 2019-07-30:
Still interested in the USB-adapter?
[image]
[image]

martinschm, 2019-07-30:
Yes for sure.

martinschm, 2019-07-30:
Hi Bert,
I messaged you via LinkedIn and also send my email address.
Best
Martin

martinschm, 2019-08-14:
Thanks to @bertmelis I now have an optolink adapter

closed 2019-08-14 by martinschm